Klipper3d/klipper #351: Graphics display in second ramps mcu

mvturnho, 2018-05-18:
Hi,

I first want to compliment you with this elegant 3d printer firmware. I have been experimenting with it for two weeks now and I am impressed by its design and posibilities.

Now I added a second ramps mcu and use its pins for the graphics LCD. This works just fine. But the display on the main mcu keeps displaying garbage.

I tried adding a second display in the config for the main mcu but klipper uses the first display defined in the printer.cfg would it be possible to use two lcd's or disable the LCD on the main mcu completely?

Also where would I start to add support for a colour spi tft display ili9341. I looked in the graphics LCD code and you drive the display directly by its pins in the mcu. How can I use the spi code already available in the avr gpio code.

KevinOConnor, 2018-05-20:
On Fri, May 18, 2018 at 01:27:59PM -0700, Michiel van Turnhout wrote:
> Hi,
>
> I first want to compliment you with this elegant 3d printer firmware. I have been experimenting with it for two weeks now and I am impressed by its design and posibilities.
>
> Now I added a [...]

I'm not sure what you mean by "displaying garbage".  Some displays
show an odd pattern if they haven't been initialized.  If you're
seeing a randomly changing screen pattern, it may be due to the
control lines floating, but that would be surprising.  If you think
this is a Klipper defect, then you should attach the Klipper log file
as described at:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

> I tried adding a second display in the config for the main mcu but klipper uses the first display defined in the printer.cfg would it be possible to use two lcd's or disable the LCD on the main mcu completely?

Klipper currently only supports one display.  Could you just
disconnect the display you don't want to use?

> Also where would I start to add support for a colour spi tft display ili9341. I looked in the graphics LCD code and you drive the display directly by its pins in the mcu. How can I use the spi code already available in the avr gpio code.

You're probably better off connecting such a display directly to the
Raspberry Pi.  There are various Linux projects already written that
should be able to drive the display.

If you really want to drive it from the micro-controller, then you'd
need to update klippy/extras/display.py so that it knew how to issue
the appropriate spi_send messages.  The klippy/extras/ad5206.py code
has an example of this.

-Kevin

mvturnho, 2018-05-21:
Hi thanks for your answer. The garbage consists of displaying the fan icon and unrecognizable text. So I assumed the display in the main mcu is still partially updated.
I have purchased a tft display for the raspberry and will start there with some code based on pygame and a connection to octoprint. 

I assume that the mcu statistics are only available from the log. Would be nice to have this on the tft display as well. 

Best regards and again thanks for this great firmware.

closed 2018-05-21 by mvturnho