Klipper3d/klipper #353: Probe command is going crazy with autolevel sensor.

skuznetsov, 2018-05-22:
I have Anet A8 "official" autolevel sensor.
In order not to crash my z stepper into the heatbed I have to add into the probe activate_gcode statement and execute gcode like this: 
G28 X0 Y0
G1 X110 Y110

I found that when it goes down during the probe it goes strange: it starts to move towards X/Y endstops instead of going down.

I solved it by commenting lines 69 and 70 in klippy/extras/probe.py and replacing them with
pos = [ None, None, self.z_position]

Otherwise I suppose that they are moving towards 0 X/Y coords.
Probably it didn't picked up correctly last coordinates from G1 X110 Y110 and still thinks that it is 0,0

Also I found that in the probe.py:build_config() it tries to add z stepper to the steppers that is already there so that Z endstop object ends up with two copies of the the same stepper instance in the self.steppers array. So it is easily solvable with simple check inside of add_stepper where you check before adding stepper into the list if that object is already there.

skuznetsov, 2018-05-22:
Also I found that when I issue G1 X15 Y15 to go to the Bottom left position of the heatbed it goes up to the initial Z position it was probing from not the found one.Looks like it is not storing Z position from the probe.

skuznetsov, 2018-05-22:
Thank you Kevin!

It is weird as I did homing_override before and it didn't work the way I want so I tried it through the probe activate_gcode option.

So I removed all probe gcode from probe and modified my homing override like this:
[homing_override]
#set_position_x: 110
#set_position_y: 110
set_position_z: 0
gcode:
    G90   ; Uncomment these 2 lines to blindly lift the Z 2mm at start
    G1 Z7 F600
    G28 X0 Y0
    G1 X110 Y110 F3600
    ; PROBE
    G28 Z0

And it started to work like a charm.

My question for now is what is the purpose of activate/deactivate_gcode in probe section then?

closed 2018-05-22 by skuznetsov

derdigge, 2018-05-27:
@skuznetsov 
Can You Share your whole config?
I want to share pins for endstop ant autolevelsensor.

skuznetsov, 2018-05-27:
@derdigge here is my config: 
[printer.zip](https://github.com/KevinOConnor/klipper/files/2043096/printer.zip)

Please note that my bed tilt coordinates are little modified as my autolevel sensor is connected  as per this image:
[image]

derdigge, 2018-05-28:
Thanks. I´ll try this next week.