Klipper3d/klipper #360: Using endstop pn for autoleveling not possible.

derdigge, 2018-05-27:
Dear all.

Ich do have a Sensor for autoleveling. In the Past i used the endstop connection simmilar to this one:
http://1.bp.blogspot.com/-6VcowyDvQmI/VRc83GpH2uI/AAAAAAAAGrE/Fhod5g2fKfU/s1600/induction%2Bdistanc%2Bsensor%2Band%2B3d%2Bprinting%2Bbed%2Bleveling%2B2.jpeg

So Endstop and Autolevel are sharing the pin. I liked the idea to have a additional "security stop".
So i configured Klipper like:
[probe]
pin: ^!PC4
z_offset: 0.0
speed: 5.0

It is the same pin like Z endstop uses. But klipper startup fails with log:
`=======================
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 234, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 212, in _read_config
    m.add_printer_objects(self, config)
  File "/home/pi/klipper/klippy/toolhead.py", line 441, in add_printer_objects
    printer.add_object('toolhead', ToolHead(printer, config))
  File "/home/pi/klipper/klippy/toolhead.py", line 236, in __init__
    self, printer, config)
  File "/home/pi/klipper/klippy/cartesian.py", line 16, in __init__
    for n in ['x', 'y', 'z']]
  File "/home/pi/klipper/klippy/stepper.py", line 246, in LookupMultiHomingStepper
    return PrinterHomingStepper(printer, config)
  File "/home/pi/klipper/klippy/stepper.py", line 109, in __init__
    self.mcu_endstop = ppins.setup_pin('endstop', config.get('endstop_pin'))
  File "/home/pi/klipper/klippy/pins.py", line 241, in setup_pin
    pin_params = self.lookup_pin(pin_type, pin_desc)
  File "/home/pi/klipper/klippy/pins.py", line 231, in lookup_pin
    raise error("pin %s used multiple times in config" % (pin,))
error: pin PC4 used multiple times in config
`

Full config:
'# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^!PC2
position_endstop: 0
position_max: 330
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^!PC3
position_endstop: 0
position_max: 330
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PD6
step_distance: .0025
endstop_pin: ^!PC4
position_endstop: 0.5
position_max: 400

[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .0111
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
pid_Kp: 19.680543
pid_Ki: 0.762812
pid_Kd: 126.939503
min_temp: 0
max_temp: 275
#max_power: 1.0	#	0.0 - 1.0
#	PID_CALIBRATE HEATER=extruder TARGET=195
#	1st - Autotune: final: Kp=19.680543 Ki=0.758402 Kd=127.677523
#	2nd	- Autotune: final: Kp=19.680543 Ki=0.762812 Kd=126.939503
#	3rd - 

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
control: watermark
#pid_Kp: 64.222092
#pid_Ki: 0.877351
#pid_Kd: 1175.264286
min_temp: 0
max_temp: 85
#	PID_CALIBRATE HEATER=heater_bed TARGET=60
#	1st - Autotune: final: Kp=64.222092 Ki=0.849499 Kd=1213.797541
#	2nd	- Autotune: final: Kp=64.222092 Ki=0.877351 Kd=1175.264286
#	3rd - 

[verify_heater heater_bed]
# adjust for personal bed setup, this prevents stock heated bed from issuing
# false positive heating errors due to slow temperature increase
check_gain_time: 100
hysteresis: 1

[fan]
pin: PB4
#max_power: 1.0	#	0.0 - 1.0

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 100
max_accel: 1000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA1
sclk_pin: PC0
sid_pin: PA3

[probe]
pin: ^!PC4
#   Probe detection pin. This parameter must be provided.
z_offset: 0.0
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
speed: 5.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.

# buttons are:
# PD2, PD3: encoder
# PA5: click
'

How to achive this? i have no free pins anywhere on this melziboard.

oderwat, 2018-05-27:
Please add some ``` at the top and bottom of your code examples.

```
like this
      example
```

Will make it easier to follow!

derdigge, 2018-05-27:
done, thanks

Nume1977, 2018-05-27:
By the log it seems you have the same pin declared twice:
´´´´endstop_pin: ^!PC4
and
´´´´[probe]
pin: ^!PC4

Please double check

derdigge, 2018-05-28:
Yes sure thats how it worked in Marlin.

There are two. one microswitch for security (3-5mm lower than the probe) and the probe itself. Both are connected to the same endstop pins(no other pins available.)

Nume1977, 2018-05-28:
What controller board do you have? Aren't there Z+ / Z- / X+ / X- / Y+ / Y- on it?
I use Z+ and Z-, one is for the endstop the other for Z probe signal

derdigge, 2018-05-28:
IT is a sanguino board. Just Google for tronxy x3s mainboard. There is no other pin free, thats why. I want to avoid a board change.

oderwat, 2018-05-28:
Using Klipper I think you have a lot of other options because it does not use encoder / sd card .. and the buzzer pin seems to be available too.

derdigge, 2018-05-28:
@cmock

I used theese pins to flash the Bootloader. How do i find out the pinnumbers? I like your idea!

oderwat, 2018-05-28:
If you search your Chip you can just read it from its pinout

http://www.instructables.com/id/ATmega-DIP40-in-Arduino-IDE-164/

oderwat, 2018-05-28:
[image]

jakep82, 2018-05-29:
From example-extras.cfg 

> The probe
> # section also creates a virtual probe:z_virtual_endstop pin. One may
> # set the stepper_z endstop_pin to this virtual pin on cartesian style
> # printers that use the probe in place of a z endstop.

You need to change the endstop pin in your Z stepper section to probe:z_virtual_endstop

KevinOConnor, 2018-05-29:
As mentioned above, one can use the probe:z_virtual_endstop mechanism or move the endstop to its own pin.

-Kevin

closed 2018-06-09 by KevinOConnor