Klipper3d/klipper #361: Pressure advance useless for curves / circles?

oderwat, 2018-05-27:
Am I right, that pressure advance will have no effect when you print a tube or cylinder?

StefanMeGit, 2018-05-27:
if the its not printed in vase mode, pressure advance will also work on tube or cylinders... it works on every speed or speed down... doesn't matter what object or curve or whatever.

oderwat, 2018-05-27:
So it does not work on single extrusions/lines but on a continuous extrusion speed calculated as one dimensional vector. If that speed changes it jumps in. Is that about correct?

StefanMeGit, 2018-05-28:
Thats correct... thats the main function of pressure_advanced.

oderwat, 2018-05-30:
So if I understand @KevinOConnor and the docs right, PA will only look at the E speeds and not at all at the printer head speeds. But the E speeds are calculated by Klipper based on the given extrusion lengths and travel distances (and accel/max speed) calculated from that. That is close to what I imagined on how pressure advance may work. I just overlooked that the Extruder speed actually is directly coupled to what I thought of being the "one dimensional" speed vector calculated from the movements. Thats me trying to grasp stuff without programming it myself :)

shoreliner, 2018-05-31:
@oderwat Klipper processes standard sliced gcode and makes adjustments to the way sudden changes in acceleration are handled in order to form a more stabilized motion throughout the layer; this leads to a number of  great results while printing. For example, the dreaded Zig-Zag is replaced by a trapezoidal wall infill method. In fact, the trapezoidal method is really how the acceleration control is "implemented" even though it's jut Gcode at that point.   Further, the extruder axis kinematics are handled separately than the X/Y/Z in Klipper.  Also, remember that the real power in Klipper is that we can offload complex kinematics like bezier curves to a separate, more powerful microcontroller.    If you really want to understand, the Kinematics document is extremely well written.  And if you REALLY want to understand(i'm pretty sure Kevin has linked one or both of these)... take a look at these articles.)  I worked in industrial controls for many years, so, keeping 20 ton machines from going into a literal death spiral has always been important for me. ;)  You probably know this one from Physics... https://en.wikipedia.org/wiki/Hooke%27s_law   -- but this one is  a bit more advanced, and better describes what you're dealing with in terms of extrusion kinematics.  https://en.wikipedia.org/wiki/Hagen%E2%80%93Poiseuille_equation

And just to reinforce what I am saying here:  yes, PA is looking at your e-velocity.  Lookahead is concerned with your printhead's acceleration.  HOWEVER I think the part that may be missing, is this... and please @KevinOConnor correct me if I'm wrong... and I'm probably wrong on a few things.  Pressure advance has its own lookahead method as well.  This prevents globbing on things like corners when printing at high speeds with pressure advance.

oderwat, 2018-06-03:
What parameter do I need to change to define the mm/s speed of the retraction part for the pressure advance execution? 

My current tests of different pressure advance values do a visible and audible effect to the extruder but there is hardly any change in the print with many models I tried. For what I am currently testing it seems as if the retraction and "de-retraction" occurs with such a high speed and "just where the speed changes" that there is barely any effect on the actual print result. This is what I noticed before (and hence created this issue). It "does stuff" but it has no effect besides kicking my extruder extra hard to do some kinda 0,05 second long retraction/de-retraction on extrusion speed changes (look ahead is disabled btw).

I am currently coding a "gcwiz" dubbed application. This was used for special the test print below. It can dynamically test pressure advance and many other stuff. For pressure advance my first test is this one perimeter wall with two sections of different speeds. Basically what the linear advance test for Marlin does but stacking. Every 10 layers the pressure advance parameter gets changed. I was of the impression that there should be an effect on the "bulge" at the location where the speed changes. That it would be possible to have it smoothed because of pressure advance.

The following image shows a one one perimeter wall (here laying on the side). The lowest ten layers where printed with a linear advance setting of 0,00. Then it is increasing every ten layers by 0,05. There are total 200 layers, so it goes 20 times +0,05 which ends at 0.95 pressure advance.

[image]

Here some output which shows what I do (for layer 100 in this fragment)

```
Switching to Layer 180 (35.900000 mm)
G1 Z35.900000
ok
G92 E0
ok
Setting Pressure Advance to 0.900000
SET_PRESSURE_ADVANCE ADVANCE=0.900000 ADVANCE_LOOKAHEAD_TIME=0.000000
ok
G1 X60 Y110 F600 E1.663007
// pressure_advance: 0.900000
// pressure_advance_lookahead_time: 0.000000
ok
G1 X10 Y110 F1200 E3.326014
ok
Switching to Layer 181 (36.100000 mm)
G1 Z36.100000
ok
G92 E0
ok
G1 X60 Y110 F1200 E1.663007
ok
G1 X110 Y110 F600 E3.326014
ok
```

As I wrote, the effect is very noticeable on the extruder steppe but there is nothing really happening in the printed model!

I just started a test where it will adjust pressure advance 20 x 0,25 steps so it will end with 5 mm .. lets see it that does anything noticeable.

P.S.: I actually started to write me some Python code which lets me easily code dynamic tests where I can change a lot of parameters programatically which are hard to dynamically change in a slicer and would need a lot of GCode post-processing if at all possible. This (alpha) app directly prints to Klipper without "anything" in between.

Examples of what I can do with it: Repeatable easy printer stress testing. Printing a cube with 4 one perimeter sides where all four sides use a different extrusion multiplier. Modifying temperature, speed and extrusion multiplier at the same time for a stringing test. Printing mathematical shapes. Basically programmable GCode "on the fly" creator. It's great fun after I solves some unexpected problems :)

oderwat, 2018-06-03:
I must miss something! Even with values up to 5 mm the print does not change and the extruder just "clicks" shortly at the speed changes.

mmone, 2018-06-03:
Did you try the ["official" pressure advance tuning](https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md) tips?

oderwat, 2018-06-03:
@mmone Sure. With no effect either. Do you have tried it and got any effects?

mmone, 2018-06-03:
No but I'm about to. Did you set the parameters in the config file or only via GCODE? If I read the code correctly, if the pressure advance parameters are not present in the config file modifying them via GCODE won't work.

oderwat, 2018-06-03:
@mmone The documentation says that pressure advance need to be "in the config" but can be set to 0.0 to not use it. I read that as "if the line is not there, Klipper does not enable the feature". I did a long series of tests and read a lot about the feature and how it is supposed to work. I set it in gcode and in config and it also obviously does change how klipper works. It just works strange and has no noticeable effect for the printed object. I actually wonder if there is a  bug in the code. I hope @KevinOConnor will find some time to read https://github.com/KevinOConnor/klipper/issues/361#issuecomment-394129007 and gives his thoughts about it.

mmone, 2018-06-03:
I'm running a few testprints and will post pictures.

KevinOConnor, 2018-06-03:
On Sat, Jun 02, 2018 at 06:48:24PM -0700, Hans Raaf wrote:
> What parameter do I need to change to define the mm/s speed of the retraction part for the pressure advance execution? 

Pressure advance is configured in units of "mm per mm/s".  It's
configured by either issuing a SET_PRESSURE_ADVANCE command or by
setting the pressure_advance value in the config file.

I recommend you follow the guide at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md

In particular, start tuning with pressure_advance=0.05 and slowly work
your way up to 0.20 if needed.  If you don't see a noticeable
improvement at 0.20 then pressure advance is unlikely to improve
results.  (Technically, it's possible to go above 0.20 if one
dramatically reduces max_accel, but the basic steps should be followed
first.)

I don't doubt there are other ways to tune pressure_advance, but I
recommend following the standard guidelines first. If those guidelines
aren't working for you, please attach the full klipper log file (as
described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md )
along with some pictures of the standard testing square at PA=0.0,
PA=0.05, etc.

-Kevin

mmone, 2018-06-03:
- [/docs/prints/square.stl](https://github.com/KevinOConnor/klipper/blob/master/docs/prints/square.stl) sliced with Cura
- coreXY kinematics
- direct extruder
- pressure advance parameters set in config file
- slicer max speed 100mm/s

from top to bottom with lookahead time = 0.0
pressure advance: 0.0, 0.05, 0.1, 0.15 0.2

[image]

oderwat, 2018-06-03:
@mmone so what is your conclusion?

oderwat, 2018-06-03:
@KevinOConnor Would you please do me a favour and read more closely what I posted? 

> Pressure advance is configured in units of "mm per mm/s".

OK. So how is 0,02 mm at all related to mm/s ? What mm/s? If the 0,02 is a length and related to some speed one needs the speed. If the 0,02 is the speed (mm/s), where does the length come from.

But I did not even ask about the meaning of the parameter. I asked about how to change the speed the extruder uses for the retraction part and if that is supposed to be more than just some millisecond long click. On my system it basically just "clicks" slightly and there seems to be no difference in that with any pressure advance value I set.

I also did the standard guidelines days ago and "convinced" myself that "0,3" is a good value. But actually I could not really see any difference at all. Changing pressure advance does not really changed anything in my printing results. That is why I wrote code to make the tests easier and automatic. I also thought the one perimeter test needs to make changes visible. It does not. Why?

If your answer is "if it does not help from 0,05 to 0,2.. it will not help you" then this is a bit weak. I have made similar tests with linear_advance in Marlin and those do work in an easy to reproduce way.

oderwat, 2018-06-03:
@mmone So you think it does change something. From the picture you sent I can't hardly see some change. When I compare that with the pictures from the guid https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md there looks everything just the same in you picture.

BTW: It should also not be bound to printing with 100mm/s. The pressure in the nozzle builds up at any speed. So maybe I am thinking of that feature in a whole wrong way and it does not even do what I thought it should do.

mmone, 2018-06-03:
The improvement is indeed minor but it is there the pictures lie somewhat as you see the elephant-foot looking straight down. First layer was a little bit to thin. Also the PLA I used has a very low viscosity. A higher viscosity would probably create a more pronounced effect.

I would say the pictures in the docs show a signifikant improvement. The corner goes from having a buldge to the two walls join exactly with a radius at nozzle size to the corner being clearly starved with a small chamfer.

I would describe the feature as following. If you accelerate the head you need to spit out plastic at a faster rate. To do that you need to take into account that when you try to extrude faster the temperature of the melt sinks thus the viscosity of the melt increases and we need to build up pressure in advance to extrude the required amount of material in the future. We are basically  compensating for the impedance of the melting process as the hotend can't dump energy into the melt at an infinite rate.

oderwat, 2018-06-03:
@mmone I think I do understand what it is...Pressure advance is about the pressure which you need to put the filament through the nozzle. You need to "press a bit" until there is enough pressure build up that it comes out of the nozzle. So from "pressing" to "flowing" there is some time where nothing or "to few" filament comes out. For this to compensate pressure advance extrudes more filament in the begging of the move and stops before ending. The parameter for this needs to consider the springiness of the tube/filament and is dependent on the heat and viscosity of the used filament.

mmone, 2018-06-03:
Yes basically. Thinking more about it the model of [electrical inductance](https://en.wikipedia.org/wiki/Inductance) is probably a very good analog for whats happening.
One probably obvious addition to above, we need to compensate for these effects when accelerating as well as decelerating. That again applies as well when thinking of the melt as current flowing through an inductor.

KevinOConnor, 2018-06-03:
On Sun, Jun 03, 2018 at 09:31:25AM -0700, Hans Raaf wrote:
> @mmone I think I do understand what it is...Pressure advance is about the pressure which you need to put the filament through the nozzle. You need to "press a bit" until there is enough pressure [...]

Indeed.  And the pressure_advance setting configures how much extra
filament to push in at the start of a move in order to build that
pressure.  So, for example, a printing move at 100mm/s might result in
an extruder velocity of 5mm/s.  In this example, if pa=0.05 and the
move started from 0mm/s then Klipper would push in .25mm of filament
during acceleration to account for pressure (.05 * 5 == .25).

If you're not seeing good results, then post the log and pictures as
previously described, and we'll take a look.  Keep in mind, we know
nothing about your printer (type, filament, software version,
micro-controller, etc.) - without the log and pictures we have no way
to meaningfully assist.

-Kevin

hg42, 2018-06-03:
well, if you have an Ender this doesn't mean you couldn't have additional printers.

A bowden should usually benefit from pressure advance, but only if you see problems at all. If you print very slow, the result could be perfect either way.

AI am not sure about a Titan extruder. The gear basically compensates the weaker motor (I think).
I got big improvements when I changed from a direct extruder to a geared extruder with big gears and a ratio of about 5. My extruder also needs a high pressure on the filament to not skip in any situation.
"Clicking" doesn't sound good.

BlackStump, 2018-06-04:
I purchased a ender-3 recently and used your config @oderwat  as it was no alterations ....what I found was I needed 6mm of retraction to stop stringing on the print ..to me that seems alot for a not so long bowden tube. I tried PA and to no real effect as it seemed the extruder acceleration/velocity was slow even tho I had fast 100/ms in the slicer for retraction.
I will get to the point I needed to add this to the config to speed up retractions
max_extrude_only_velocity: 120
max_extrude_only_accel: 3000
I suspect that will have an effect on PA as well , tho not tested since making those changes but I can reduced retraction ....I am down to 4mm and can probably reduce it even more.
On a side note the leadscrew was binding for me straight out of the box. you need to be able to twist the leadscrew with your fingers easily to raise or lower the X axis

cheers

oderwat, 2018-06-04:
The original firmware for the printer has 25 mm/s max speed for the extruder. I use a retraction length of 3 mm with 25 mm/s and have absolutely no stringing going on. After I changed geometry for the z-lead-screw my z axis is so smooth, when the motors are offline it even moves down itself when I push it down with a finger.

Just to make that clear: I have pretty much no problems with my prints. No binding, banding, layer shifts, clogging, adhesion problems or whatever. I just want to optimise a bit how seams and corners look by utilising PA.

BlackStump, 2018-06-04:
fair enough.

shoreliner, 2018-06-04:
I'm having a hard time following at this point.  Is the issue that you
aren't able to get pressure advance to cause any change in your prints?  I
just want to kind of clarify and clear to air here because I do not
understand at this point... I also have multiple ender 3s, and other bowden
driven printers... while they have required higher PA settings than direct
setups to print at higher speeds, by following the guide for PA and going
to a higher value for bowden I was able to see results immediately.  You
are asking about acceleration, which at this point you're simply getting
into one of klipper's key functions.  If you check the Kinematics document
you'll find this:

snip->

It is also important to limit acceleration so that the stepper motors do
not skip or put excessive stress on the machine. Klipper limits the torque
on each stepper by virtue of limiting the acceleration of the print head.
Enforcing acceleration at the print head naturally also limits the torque
of the steppers that move the print head (the inverse is not always true).

Klipper implements constant acceleration. The key formula for constant
acceleration is:

velocity(time) = start_velocity + accel*time

<=pins

Still, since I really don't know what you are asking... I have no idea
what the implications would be of changing this, or how much
alteration would be required as i've never looked at the structure of
the movement hierarchy.  Moving right along....

If you want to change your extruder's acceleration kinematics, you can
find that formula in the same document and find a way to affect the
code.  You seem to understand the linkage... while you may have found
something in your tests, which I'm guessing you are doing using
something like simulavr?  (This would also be helpful info, by the
way.)

You must understand how complex pressure systems are, how difficult
they are to model, and that people devote their entire lives to
studying this phenomenon.  mmone demonstrated quite well how pressure
advance changes print quality... I've been able to tweak it same on
all of my printers as well.  So...

All I can guess at this point is that your machine is not mechanically
tuned properly if you aren't seeing any changes from PA.  How about
some photos of the square?  I would love to see PA worked on, but....
do you have any idea what to do to improve it?  I mean at this point i
really wouldn't know where to begin, but if you do... we're all kinda
in this together.  I know this has been basically

just me rambling but honestly I think you are really ahead of the
curve as far as solving this problem... what do you propose?

~David

hg42, 2018-06-04:
@oderwat  I revisited the test you posted above (that with the first photo).
I admit I have not looked close enough to the gcode.

I assume the code is more or less for one layer.
A layer starts at X110,
speed 1200 to X60,
speed 600 to X10, 
spped 600 to X60,
speed1200 to X110.
For each line you reset E.

This looks like a working test.

Now, we don't know anything about timing and the linearity of your extruder or hotend.

First test: use 1200 on both sides, this should be seamless.

Second test: Do the same with 600 on both sides. Do both walls look the same?
You already see this on the test object in the photo. Both sides should look identical, same transparency, same thickness.
On the photo the rights side looks generally darker.
The right side seems to be more transparent.
This would mean less extrusion on the right side.
However, the photo is very unsharp, and the camera doesn't look on the seam symmetrically. I would like to see two macro images of the seam at the bottom and the top. It should look exactly vertical, the background should be the same dark homogenious material.

Third test: use 300/600 instead of 600/1200 and compare both.
Fourth test: use 1200/2400 and compare to both speed combinations above.
May be the extruder/hotend combination is near a limit and test 4 makes this more visible.

However, the speeds are slow, so this shouldnkt be a problem for your ectruder/hotend combination.
The photo shows some irregularities. There are darker layers (lower extrusion?), and you have some white spots (more extrusion).
What about temperature? May be it's to low? perhaps there is some cold draft from a window or door etc.

About the clicking. If the hobbed bolt digs well enough into the filament, skipping usually generates grit that you can find in or around your extruder. You should clean it before a test, to remove older dust.

oderwat, 2018-06-04:
@shoreliner I can't imagine how to be more clear about what I asks. I will print some squares for you later.

@hg42 Thank you so much for your time! I am already following your recommendations. Will post when I have done all the tests. There is for sure some air flow going on. I could close my enclosure for the tests. But I was not focused on getting nice layers for this test. What I thought is important, is to see the seam and how it changes (or not at all). Besides that. Printing 200 layers of a 0,4 mm line on top of each other is challenging. It will also change the look when I look at it and breath :)

It is one line printed vertically and it is hard to take meaningful photos with the iPhone. I may need to use a different camera. The seam is pretty identical from bottom to top for all of the PA values. And you are right: One side looks darker. I use the exact same amount of extrusion on both sides / speeds.

For the following tests I use 209 layers, starting with PA 0.0 and increasing this every 10 layers by 0.05. So it goes from 0.0 to 1.0. BTW: The printer reports the changed values. I also changed my code so my commentary outputs are all starting with "; ".  I may implement, that I can export the GCode instead of sending it to the printer. This way you or somebody else could run the same test. Which would be very interesting.

Printed with 1200/1200 as expected there is no visible seam at all. Both sides are also 100% identical in thickness.

Printed with 600/600 also no visible seam at all. 

The wall from 600/600 looks thicker when laying side by side to the 1200/1200 print. 

But measuring it with a calliper shows they are pretty much the same thickness (with a grain of salt, because you can't hardly measure this reliable). But when I lay both prints side by side with the top layer facing, my finger can not feel a difference in thickness where they meet! I am pretty sure the different appearance has to do with the material reacting to different temperature/cooling. I may change filament later and redo some tests.

With the 300/600 print I already notice that there is nearly no (noticeable) retraction happening at low PA values. The extruder just changes speed in the middle of the line. For PA 0.4 and higher there is something happening for the extruder. But it happens super quick as if it does a retraction of 1-2 steps. But there are no noticeable changes in the print over the full length. It has a pretty much even seam again.

(TBC ... need to work for money)

oderwat, 2018-06-04:
An (happy) update:

1. Getting those strange differences between the slower and faster part I compared my self generated GCode with S3D. This made me kinda happy because I calculated the same extrusion lengths for the same parameters! So my test program should work. I love my little gczip app :)

2. I changed the speeds for the test between the left and right part of the wall to 20 mm/s vs 100 mm/s. This gave me some noticeable changes for different PA values. But it also "broke" my titan (clone) extruder. Actually I could not get the tube staying connected so I (temporarily?) changed my extruder back to the original Ender 3 extruder setup. Then I re-run the same test and watch this:

[image]

So basically on one hand side there must be something wrong with that titan usage but there is also not a lot of effect when testing with low speed changes. From here on I will continue experimenting.

I already rewrote my gcwipz such that it can emit the gcode for something like the test wall on the picture. I am planning to create a more complicated shape for further tests. 

What is seen in the image is what I was expecting from the beginning! Life is good!

Thank you for all your patience 👍 esp. @hg42 because if his post I came to the idea to make more extrem changes in speed for this test.

hg42, 2018-06-04:
from my point of view, the titan extruder is designed to replace a standard direct drive extruder on something like a delta with a much lighter variant (reducing moving mass).

The motor is replaced by a much shorter version (pizza box!) and the lower torque is compensated by a gear box. So at the end the torque is similar.

For a bowden setup with a fixed extruder, I would choose a geared extruder (e.g. with 8mm hobbed bolt) and a standard nema 17. Where "standard" doesn't mean what cheap printers usually have.

Many printers come with weak 40Ncm motors, I would replace them with another model (~60Ncm).

When I did that for the direct extruder of my delta, I got much better prints. A geared extruder also works with the weaker model (5:1 = 5 x torque). Other benefits of the gear ratio are the motors running at higher speeds and the higher resolution.

oderwat, 2018-06-04:
@hg42 I used the titan because for its possibility to transport flexible filaments more consistent. I am missing the possibilities my Anet A8 gave me with it's direct extruder. Using it with the bowden was trying to do a shortcut for medium TPU (not expecting simething like filaflex to work).  I also already have a sandwich stepper (pizza box :) and plan to build an direct extruder hot end and make it a exchange able part of my printer. I am not sure if I can use this stepper on the drivers of the 24V driven CR-10 board and maybe try it with a second board first and learn about using multiple MCU with Klipper at the same time.

I also have some bearings and mk8 + mk9 filament transport "gears". I don't know the correct names. I mean the ones with the concave area which transports the filament while a normal bearing is pressing against it. These are planned to be used with my current or another Nema17 (medium length) I have for experiments with those kind of geared extruders. I did not yet decide which of the designs I wanna try but have some in closer selection.

One other thing I had to debug was why my X stepper (and only this) made a buzzing noise when the head bed is active. It turned out that the MOSFET part on the board causes interferences in the driver for this stepper. When I change X and Y stepper. The sound moves to the other stepper. So I added a SSR into the formula and with the MOSFET of the board only driving the SSR, the noise is gone. Even when I use the same PSU for the bed heating still! I also used another PSU but that was actually not needed. There is still a very slight interference when the hot end is heated.

For testing it that effect is special to my board or a problem other CR-10 boards have, I will order an extra CR-10 board just to try that out. With that I also can try different steppers without risking to blow my motor drivers. I do have multiple other boards here (Rumba / Ramps) and plan to get a detla or smoothie or both in the next months.

I also still have some diamond hot ends here and may try to add two more extruders for a three filament mixing setup. I was also thinking about how to write software which can help with "mixing" colors by calculating the mix "in advance" based on extra input. This is part of why I am experimenting with generating GCode. Much to explore :)

For PA I just added a way to easily "zoom in" to a PA range such that one can get the best value. This picture is for:

```
; gcode made by gcwiz (c) 2018 Hans Raaf / GitHub: OderWat
G28
G1 Z5 F1500
; printing 10 sections for pa from 0.2000 to 0.6000 in steps of 0.0400
; first layer is 0.100 mm, other layers are 0.200 mm, total height is 40.000 mm
; changing pressure_advance every 4.000 mm (every 20 layers of a total of 200 layers)
```

[image]

Next I am going to add a 45 and 90 degree corner with and without speed change to this test routine and parametrise the remaining stuff (I already have layer height, nozzle, flow multiplier and more as variables which calculates the other stuff).

shoreliner, 2018-06-04:
I am unsure if this was even discussed previously, but either way I missed
it.  Are you using S3D for all of your slicing?  There have been issues
reported in the past in which S3d generated gcode was causing "invalid
speed" errors to be thrown while being processed by klippy.   You can find
this particular issue here:
https://github.com/KevinOConnor/klipper/issues/173  --  Now, I've also seen
people have absolutely no problem printing at high speeds with Klipper
printing S3D sliced models.  Anyways, perhaps when you slice your squares
try not only different PA settings, but different slicers.  I'm not sure if
I'd use a proprietary slicer to perform anything I'd consider experimental,
but it can't hurt.  Maybe throw in Ideamaker for the hell of it.

oderwat, 2018-06-04:
@shoreliner I wrote my own program to create these tests and they drive the printer directly or generate actual GCode. I don't care much about the squares actually. I think they need to long to print. 

My original problem is also already solved. The speeds for my test code where to low and my titan extruder sucked.

Besides that I am using S3D for my normal prints and got no errors with Klipper.  I can imagine that you can generate GCode which will let Klipper error out. I actually had that happen with some hand generated GCode I used in a start script for over a year with Marlin. But Marlin just puts constrains onto the print, while Klipper stops with an error. I am not sure if I like the way Klipper handles this, but from a programmers point of view I think it is OK. I live the "warnings are bad and errors need to be solved" style of programming for my whole life.

oderwat, 2018-06-04:
Before I close this issue I want to show what it actually all was about in the end. And why I didn't rest until I got my problems solved to continue working on my original idea:

Here three pictures of what my pressure advance test routine in my upcoming `gcwiz` tool produces, when I tell it to create a test for a 40 mm height wall with the range from 0.0 to 0.6 as pressure advance values. With the additional parameters of 0,4 mm nozzle, 0,4 mm line width, 1.0 extrusion multiplier, hot end on 210C, bed on 50C, 90% cooling fan (starting at the second layer), 0.2mm layer height with a first layer of 0,1 mm.

[image]
[image]
[image]

My app can also create a classic GCode file which can be printed with any host. The file generated for this test is [here](https://gist.github.com/oderwat/5ad1efe0145dea8b593f707664d952da)

A preview of how the GCode is previewd in S3D is here:

[image]

Sadly one **can't print this GCode from S3D**. It seems it stops at the Klipper specific commands. I didn't search for a workaround and don't know if something like that exists.

**Using the GCode in OctoPrint does work** and creates the same result as when `gcwiz` is directly driving the printer over the Klipper printer port (or USB Serial for me).

Printing the whole test "model" needs about 23 minutes, uses about 1.42m filament and needs no human interaction at all. It is also very easy to reason about and in my opinion the quickest way to find a good pressure advance value without even knowing what to do exactly.

One cut down the height of the test in half and still got a good result in about 12 minutes. Compare that to the manual work printing "squares", setting PA values and interpreting the results.

I may design a test for the lookahead value in the future. Not sure what is needed to test it yet.

shoreliner, 2018-06-05:
You know, I'm trying to offer suggestions to you here that are in line with what Kevin has asked for as far as PA tweaking, and for some reason I sense you feel I am patronizing you or something.  Let me assure you, I am not.  I simply had a hard time understand the actual problem you were having; the photos you posted recently along with the description of gcwiz was very helpful.  I only respond here to help.

That said,

We use that calibration square stl because it is a common model that we can use to relate, not just for ourselves, but for others in the community. Could it be an improved process?  I'm sure, this is an ongoing project.  As I previously had mentioned, S3D and Klipper have been previously documented as having some issues... if you check the link I put above, you'll see S3d generated some quite odd commands in some scenarios that caused those errors to be thrown.

Additionally, as It seems like you have quite a lot of experience in this realm, so I would love to see what you develop for lookahead var testing.  I'd only like to assist you in this. I just hope that you have read the documents we've linked.  You will notice in fact that one of the To-Do items is to finish/further develop the simulation system:

"Testing features

    Complete the host based simulator. It's possible to compile the micro-controller for a "host simulator", but that simulator doesn't do anything currently. It would be useful to expand the code to support more error checks, kinematic simulations, and improved logging"

I can be of particular assistance here with the mathematics in question if you'd like to tackle this item.

oderwat, 2018-06-05:
@shoreliner I don't feel like you are patronising me. I just think you didn't read closely what I wrote or I am not able to write it understandable enough.

You ask me in your current post to check the S3D issue #173 you posted yesterday. I already did this yesterday and even contributed to this thread. I could give a suggestion for a work around to the actual problem.

I also think that the problem from #173 is actually a Klipper problem and not a S3D one, because Klipper should try to be "as good as other firmware" even if that means it create "warnings" and interpret stuff. I already know that @KevinOConnor has another opinion and that is fine. It is his project. But I think he is to narrow minded in that respect. I am long time developer myself and understand why he thinks like that. I would have a hard time too. But in the end I may had convinced myself that I would better accept "bad" input, generate a warning and continue printing. Because that is what a user expects. I usually make software for users to use and not to render two programs (S3D and Klipper) unusable.

I don't want to use the calibration square, because it is wasting material, lasts a long time, needs a manual change of the parameters and the interpretation of the result is hard for me. But of course you can do what you want to test your printer.

For comparing results with other firmware I would prefer more complex models. I hate all these cubes, squares and benchy people print. And always there is the next person which asks how to change the steps because his 20x20x20 cube is 1 mm off. People need clear models where you can directly see if something is good or bad.

I don't feel like I have a lot of experience and I am do bad errors all the time. I have read all the docs multiple times but I eventually will read the code. Code is my friend :)

oderwat, 2018-06-05:
BTW: My question is still not answered. How to change the timing of the [red circled](https://github.com/KevinOConnor/klipper/issues/361#issuecomment-394177237) part in the graph I copied from the documentation. I mean like "where in the code" or "what parameter" needs to be changed to get different acceleration for this part. I understood that the "max" of it is defined by PA. I asked what defines the "duration" for this part.

hg42, 2018-06-05:
> gcode generator

@oderwat I like your approach to the problem by using a gcode generator. I did this myself before, when solving other problems and I like how I get much more control and complete independence from slicers.

I think using variation of parameters over geometry (e.g. height) is a good, fast and reproducible way for calibrating (see temperature towers).
The only problem, why this is still a tool for experts is because you usually cannot use a slicer for this (without using advanced features like specifying parameter changes for layers) and directly distributing gcode to users doesn't work well, because the init sequences, position limits, speeds, accelerations etc. would not fit to everyone.

> red circled part

(without looking at the code)
Because the volume per length is a fixed relation extruder acceleration is always bound to the (xyz-)movement acceleration.
The PA algorithm only changes the extra volume per length depending on the extruder speed:
`extra_filament = pressure_advance_coefficient * extruder_velocity`
So the acceleration is fully defined by these formulas.

> pictures

I am not sure what I am supposed to see there. Apparently,
* even with PA=0 the print looks good.
* you are getting a seam when printing a line with a sudden speed change in the middle and this doesn't change much (or at all?) with PA (besides making everything worse above 0.3). I agree, this indicates a non perfect pressure algorithm (the formula may indeed be too simple for this case), but I think, the test doesn't represent a realistic case, because no one will ever suddenly change the speed in the middle of a line. Instead the algorithm aims at corners and ends.

oderwat, 2018-06-05:
@hg42 

1. "Everybody" will be able to use `gcwiz` for this and other stuff, as long as he can run the code and has either a gcode or serial target. You don't need a slicer. Just run the program or use the GCode somebody generated. The one I posted will actually work for a lot of printers. One may just change the temperature. 

2. If that was not clear: I am going to release `gcwiz`. I may also release a black box Klipper RasPi unit and/or prepared image, which works by putting it between the printer and a computer without any special software setup needed by the user, as long as the printer is in an online database. But thats "future" and depends on how stable I get everything. Currently it works already pretty well.

3. This can't be the complete formula as it is just linear and the graph isn't showing a linear curve. The graph does explicitly say "extruder velocity" on the y axis. So it does push more for some time in the beginning and at the end it de-accelerates and even crosses the x axis which means it has a negative velocity. The extra amount of filament is not added linear over the whole time. I actually wonder over the "extra_filament" too. There should not be any extra filament as it stops pushing before the end (much like coasting). It just changes the distribution of the pressure. Everything else actually would not make sense to me at all. I don't think that I have understood that wrong. But I will look at the code when I find time for that.

4. Pictures are hard. You could just print one of those tests yourself and see what the seam really does. Check this picture:

[image]

oderwat, 2018-06-05:
I actually wonder if I need to do a little modification for the gcode as the start and stop point for the mid part and the edges are actually the same coordinates currently. I think it would be more correct if it starts and stops halve the nozzle size before the actual coordinates. this would need a move without extrusion for positioning between the stop point and the next starting point. I think I saw something like that in S3D gcodes. This would then affect the seam considerably and the current "flat" point would be earlier. It would also probably make the seam go away or much smaller for the slower prints I did before.

shoreliner, 2018-06-05:
You know, after looking ahead at that prior S3D thread and being taken
aback with the way some of the entitlement issues some individuals have...
I wanted to respond with something that contained vastly more words than
this.  However, instead... man, this is a fairly small open source project
that for most of us is a fresh change in architecture; it also produces
great prints with a properly configured printer and slicer.  S3D is closed
in every way; there's no developer API, there's no good information on
S3D's gcode generation routines, there's never been so much as a whisper of
divulging any helpful information to firmware developers.  Just the
opposite in fact, S3D has went ONLY to printer OEMs in an effort to embed
their software in rebranded versions... and I hear you saying, "But you
could just generate a warning instead of throwing an error....".    Of
course you could, but given the experimental nature of this project and the
fact that it controls a machine that spews hot plastic, I think its safe to
say that we don't want to handle issues that could result in damage to
someone's printer, hurting themselves, etc when the alternative is that
they have to re-slice the print or use another slicer.  Anyways.  I wish
you luck, but I don't have anything positive I can offer here at this point.

~David

oderwat, 2018-06-05:
@shoreliner I don't know what this has to do with what I am trying to do here. Why are you bringing up the S3D thread here? I lover Klipper for it's great print results and try to contribute stuff. But everything I do is put down as it seems? I tell you what: S3D is a great working slicer and gave me very good results over the years. I don't regret the money I spent there. I do start to regret that I am contributing money, my time and enthusiasm here. I was running around telling everybody and his cat to try out Klipper for some time now. I guess I need to stop that.

There is still not an answer to what I was asking. No response to what I am demonstrating. All I got is you preaching to me about S3D and what they didn't do for firmware developers? I am myself struggling with them. I just today wrote again with their support for getting more possibilities to use Klipper with their software. Do they care? Maybe not. Do I care.. maybe not anymore.

But one thing I do wonder: What do you wish me luck for?

oderwat, 2018-06-05:
@shoreliner Just one more thing: If you have nothing positive to offer... maybe just stay away from the keyboard. I am really upset about your way you handle me. This is ridiculous!

shoreliner, 2018-06-05:
Just one more thing Hans:  *shrug*

BlackStump, 2018-06-06:
@oderwat  I think your PA test is great, easy to use much better and easier to see then printing squares and could not agree more that squares are just a waste of plastic.
Any other tests that you formulate that are simple and easy to determine results I for one would be happy to try.
keep up the good work.

BlackStump, 2018-06-06:
@oderwat  I have already mentioned I have a un-altered ender-3. I have printed your PA test as is and the bulge smooths out around 0.6 for me which is not what you are seeing in your picture so I altered you gcode and changed the PA to 0.5 steps instead of 0.3 so I max out at 1.0  
I wanted to see when too much was too much and for me that is around 0.7 so it seems to me hotends and bowden tubes must have a effect on the results. I will at some stage change over to a E3D hotend and retest to compare against the short little stock hotend.
sorry my camera is not good enough to show the effect on the print.

KevinOConnor, 2018-06-06:
On Tue, Jun 05, 2018 at 03:50:00PM -0700, Hans Raaf wrote:
> There is still not an answer to what I was asking. No response to what I am demonstrating.

Hi Hans,

Your enthusiasm for Klipper is appreciated.

It is not clear to me which question you feel is unanswered.  I
believe both @hg42 and I addressed your acceleration question earlier
in this thread (in brief, extruder acceleration is directly tied to
move acceleration, move acceleration is controlled by max_accel).

It's also not clear to me what you are trying to demonstrate.  You've
posted a number of pictures, but I personally have no basis of
comparison for those pictures.  So, if by chance, you are looking for
a response from me, then know that I do not have any feedback to
provide.

If you'd like my feedback, I recommend you run through the standard
pressure advance tuning steps, attach a klipper log file from at least
one print, and post some pictures of the standard tuning squares.
That will give myself (and others) a basis of comparison with which to
evaluate alternative tuning processes.  If you do not wish to take
these steps, then that is fine.  I understand that you have your own
goals and only you know how best to allocate your time and resources.
Please keep in mind though, that I too must allocate my time and
resources.  If you decide it is not worthwhile to follow these steps,
please don't expect my feedback, or feel that you need my feedback.

Cheers,
-Kevin

oderwat, 2018-06-06:
@KevinOConnor 

For my open question:

Your graph shows extruder velocity for PA ... I don't see that this is coupled to the X/Y speed. The nozzle would "jump" at the start and slow down till and then move backwards in the end. So how can the extruder speed (including the PA) be a direct tie to X/Y speeds?

[image] .. I am asking *what defines the acceleration of the part in the red circle*? What defines how long it accelerates and de-accelerate? I want to change the length of it (slightly). BTW: I expected it to be a "max accel" value for the extruder (for pa?). Maybe the simple answer I look for is: `max_extrude_only_accel` ... But shouldn't this parameter be different for a "retraction" and a normal "extrusion"? I can accelerate faster and to a higher speed for retracts than for extruding plastic.  Anyway. Not knowing how it works is the reason for me to ask.

For the PA testing and my test procedure there is probably a major misunderstanding going on:

I don't want/need you to tell me what is a good PA value. I did in fact printed multiple rectangles in the past. Why should I print this model over an over and what will be the possible feedback? I already found that an overall 0.3 PA is good for my prints (full prints not tests). I use Klipper since many days for normal printing now. But over USB2Serial, with S3D which works still fine btw..

You say you have no comparison for the pictures. Why would a comparison be needed? What to compare against? The comparison to do is "inside" the picture. PA 0.0 at the bottom PA 0.6 at the top. Picture shows the effect. I don't understand what this should be compared too? Do you want me to print the 20 squares with the same PA values as they are used in the picture? From this wish I have a feeling that you don't really know what these pictures depict. Or I am really to dumb to understand what you are telling me.

Here are three hollow square prints from bottom to top: 0.0 / 0.35 / 0.6. Speed was 100 mm/s with 0,2 mm layer height and 0,4 mm nozzle and line width at 1.0 extrusion multiplier (I usually would use 0,95 with this filament). The filament is BDP (Greentec / from Extrudr in Swisserland. It is pretty good but a bit costly about $60 per 1kg). Thats the same I used for all the tests here and is usually used in most of my prints. I disabled outline/short layer speeds being changed. So it is really 100 mm/s and not 50% or whatever the slicer(setting) normally would use. I guess most people, which will try this test, will end up with slower speeds, because they don't know/compensate for the slicer doing "sane" things to the outlines/perimeters.

[image]

This now out of the way, back to what I wanted to demonstrate and (optionally) get feedback for:

What I tried to develop is an alternative method which uses less time and is more practical and better to reason about. My program does the testing and parameter changes automatic. The pictures show what happens when PA is gradually increased and proves that PA does act and lets one decide which value is best. This is directly coupled to the thickness of the seam and the under extrusion visible with higher PA values. What I wanted is to discuss this method and what may be flawed with it. @BlackStump actually did and I am happy to have at least this person evaluating my idea. He also mentioned that his printer, even being basically the same as mine, has different PA settings. I run my test multiple times and it has a perfectly repeatable result. I guess thats the same for him. This again proves that my method does reflect the minimal differences we may have with out printers. But it could also be filament and temperature related. Which again gives a lot of value to my test, because it is much faster than to run multiple manual tests with the squares for each filament and temperature change.

What I (maybe thought I could) expect is that you say: "Maybe you are onto something.. let me try it for one of my printers, where I know how PA reacts very well. Please generate me some GCode with a temperature of 190C / 50C and make it fit my 123x240 sized bed, such that I can run a test of what it does myself."

If that seems good I could setup a website where people can get their GCode much like the one for linear advance. I could contribute my code to your Klipper scripts, publish my GCWiz or whatever. I can do most of that without your. Maybe running a 25 minute print to test my routine is to much for you. Or you possibly know a reason why my test is bad designed in the first place.

oderwat, 2018-06-06:
Oh... I forgot that my last printed squares where just two outlines for saving filament. This is not exactly following the guide. So I am doing a PA 0,4 mm with klippy.log and attach it when it is finished. This will then be the last print with this filament. So if thats not good enough I need to re print a whole bunch of squares again.

oderwat, 2018-06-06:
For what I mean with "real 100 mm/s" see this preview: 
[image] ... in comparison to this one:
[image]
... but I think you all know about that when testing the PA as described in the guide.

oderwat, 2018-06-06:
There we go. Just for the record: Printing one of those squares with 100% infill at a "constant" 100 mm/s lasts 36 minutes on my printer. This is as said before with PA 0.4

[image]
[image]

I think I could print one more ;-)

[image]

Here ist the klippy.log from firmware start / reset to finish this print and M112 shutting down:

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2077269/klippy.log.zip)

KevinOConnor, 2018-06-07:
On Wed, Jun 06, 2018 at 03:52:44PM +0000, Hans Raaf wrote:
> Your graph shows extruder velocity for PA ... I don't see that this is coupled to the X/Y speed. The nozzle would "jump" at the start and slow down till and then move backwards in the end. So how [...]

Lets take an example with a printing move at 100mm/s that results in
an extruder velocity of 5mm/s. In this example, if pa=0.05 and the
move started from 0mm/s then Klipper would push in .25mm of filament
during acceleration to account for pressure.  If max_accel=500 then
the move at 100mm/s would take .2 seconds to accelerate from 0 to
100mm/s.  So, with pa=0.05 the extruder would push in an extra .25mm
in .2 seconds, which is an additional constant extruder velocity of
1.25mm/s.

In the graph you point out, there are two areas with acceleration -
the velocity slope on the left and the instantaneous velocity change
that you have circled.  The velocity slope on the left uses constant
acceleration and it is unchanged with any PA setting - it's directly
tied to the move acceleration.  The instantaneous velocity change is
infinite acceleration and it is not configurable.  The magnitude of
the instantaneous velocity change is determined by the
pressure_advance config setting and the ultimate cruising velocity of
the move.

[...]
> What I tried to develop is an alternative method which uses less time and is more practical and better to reason about. My program does the testing and parameter changes automatic. The pictures show [...]

There are some advantages to the square test:

- It can be generated with any number of slicers.  A g-code generater
  needs, at a minimum, the filament diameter, the extrusion width, and
  the layer height.  These parameters are different on different
  printers so there is no universal default for them.  People tend to
  have a slicer of choice which they've already configured (and set
  other useful features like bed size, temperatures, homing/nozzle
  wipe sequences, etc).  A good number of people will be reluctant to
  configure another g-code generator.  Using the square avoids this
  issue.

- It tends to work well even if the bed is not fully level.  If
  there's extraneous signals in the test then it's easy for new users
  to misinterpret the results which could lead to significant lost
  time.

- Using a print object with multiple outer walls tends to magnify the
  pressure issues (inner walls blob causing outer walls to blob
  further).  This makes it easier to identify a good PA value.

- The square is structurally sturdy, so it can be kept for comparisons
  with future/past runs.

I'm sure it is possible to improve the PA test.  It certainly has some
flaws:

- It does take longer to print and it uses more filament.

- The corners are susceptible to warping which can
  complicate/invalidate the test.

I did try a variety of tests when coming up with the PA guide.  I
settled on the square test because I felt it was the most robust test
I tried - to wit, it's better to spend an extra 15 minutes on a test
then to risk spending 10 hours debugging the test.

In particular, in my past single wall tests, I found that I got a
different (worse) PA value.  I'm not fully sure why - I think it may
have been the single wall corner being too fragile.  In your tests,
the square seems to clearly indicate 0.40 is your optimal PA value -
that is also very close to your single wall test - but I can't really
tell from the pictures if it is exact.

Be aware, it seems you were using a max_accel=500 - that isn't a very
common setting on most printers.  A low max_accel value like that does
make it easier to reach higher PA settings in general.

So - my feedback is to "have at it".  If you want to work on new
methods to identify pressure_advance then great!

> What I (maybe thought I could) expect is that you say: "Maybe you are onto something.. let me try it for one of my printers, where I know how PA reacts very well. Please generate me some GCode with [...]

No, alas, I'm backlogged, and likely will be for the next several
months.

-Kevin

oderwat, 2018-06-07:
@KevinOConnor Thank you for your response and time! I think it will help me a lot.

About the graphic: OK. I think I now understand how it works and that the graphic is correctly depicting what happens. The instant speed change is the "back to the normal extrusion value" vertical line at the end. The slope is directly bound to the acceleration which is bound to the acceleration of the printhead. So to make it slower I would need to set an even lower acceleration. This kinda leaves open what happens at the de-acceleration and why it crosses the time line and goes into the negative. But I think it helped me already to understand it. I have a feeling that it could be better do "decouple" the E speed from the printer head speed and use another curve instead the linear correlation. But I am not here to change that :)

I think for the most of the "pro square" reasons there is a very easy solution.

1. I could just read printer.cfg or there could be a way to query Klipper for the needed values. Actually my `gcwiz` code can easily be started on the RasPi. It could even be a OctoPrint plugin. Still, the Filament size and the build plate area is very common knowledge for any 3D printer user. The layer height can be bound to the nozzle width as can the line width. I don't think that this will overwhelm a 3D printer user. Especially when he goes lengths to use Klipper :) 
2. I can easily extend my test with a section of dual and tripple wall thickness for seam and corner.
3. What my test prints was sturdy enough to keep but it also can be reproduce very fast.
4. It is not spending 15 minutes more to print say 5-10 squares. It is 3-5 Hours vs. 25 mins for my broad test and maybe another 25 mins for a coarse version. The squares are also prone to make errors by forgetting to `RESTART` and because one need to edit the config file multiple times and keep track of what square was which print. All of that is not needed for my test.
5. In the cons you forgot that different slicer settings will have a big impact: Retraction, Coasting at the line end, different speeds for outline than basic speed. Different line widths for outlines and normal lines. I am sure I forgot stuff. With my test those are all under control and "standardised".

My `max_accel` setting is what is the original firmware value, as is 25 mm/s for max z velocity. I am sure both could be doubled but I kinda forgot about it. Thanks for the heads up on that.

To check the "best" PA of the picture one can just measure the length of the full print, the length of the "best" value and "0,6 / full_mm * best_mm". I am sure that you know that though :)

About the single wall test. You may have noticed that I don't test the corners but the speed change at a straight part of the wall. something a slicer can't 100% replicate (I tried with different processes in S3D, the GCode is always polluted with extra commands at the seam). The seam seems (lol) to be a very good indicator. It also makes a lot of sense that the optimal PA creates the smallest seam. What I wrote in my last comment is, that I suspect that the delta of the speed change has some influence to. Maybe this can be made better.. maybe it is good enough as it is.

hg42, 2018-06-07:
@oderwat 
> This kinda leaves open what happens at the de-acceleration and why it crosses the time line and goes into the negative

`extra_filament = pressure_advance_coefficient * extruder_velocity`

so, for linear acceleration, the `extra_filament` (a **length**!) will also increase linearly, which means the additional `extra_extruder_speed` (= `extra_filament / time`) is constant.

This is the constant speed offset in the curve.

At the start of the movement the offset is positive (to build up the pressure) and at the end the offset is applied negative (to release the pressure).
Which leads to a retracting filament at the end.

hg42, 2018-06-07:
btw. I also had some hard time to see, where the time constant for the third curve can found in the simple formula:
[image]

I think it's kind of inverted logic:
because the form of the curve is always similar and the speed offset and the raising time are dependent, the PA coefficient is only value to be calibrated to match the given time constant.

oderwat, 2018-06-07:
I actually have to confess that talking about math / physics in english seems to be very hard to me :(

What I currently think is that the curve for the extra pressure should look like an exponential function. And this would be the mirror to the logarithmic behaviour that is depicted in "actual filament".

Part of the topic of S-cuves which I only skimmed over, seems to address that in a way, because the movement of the head will not be linear anymore. Hence the extruder acceleration won't either.

In the end there is still the assumption that the pressure effect is parallel to the acceleration. But in reality I think the pressure would need to build up slightly before the acceleration and stop at a point where the "remaining pressure" can be picked up by the next movement. Instead of assuming it to go to zero (and retract). I think this is why there is the lookahead. Which I not really understood yet.

This also goes a bit above my head and intend. I wanted to find a way to make PA better to grasp and to demonstrate with some reproducible and easy to follow procedure. I actually think that I found it and I got responses outside of this issue, which are very encouraging to follow this route further.

hg42, 2018-06-07:
you have to clearly distinguish which value you are looking at in the curves.

Constant acceleration means linear speed ramp and quadratic position ramp.

> S-curves

don't help here because jerk, vibration etc. is not a problem here

> assumption that the PA algorithm works good enough wile being parallel to the head acceleration

it always has to be in parallel, because for each part of the movement you need to lay down an exact amount of filament (basically length * height_of_line * width_of_line). If you move faster, you also have to extrude faster by the same factor.

Unfortunately, if you move the extruder some mm consuming a volume of filament, the nozzle doesn't extrude the same volume, because of the springy behavior of some parts in the system (bowden, filament, melted filament). You need a specific pressure to achieve this. And the pressure is dependent on the speed. For every speed change you need to adjust the pressure, which is done by pushing more or less extra_filament into the tube.
I think PA is mainly designed to compensate for the melted filament behavior (pressure, compression, flow, etc.). But because the behavior of the other parts is also kind of springy, they are automatically included in the compensation with a proper calibration.

hg42, 2018-06-07:
a calibration "tower" is definitely a good way, but with the mentioned problems.

But if you restrict it to Klipper this can work, you already mentioned some of the possible solutions (e.g. reading parameters from the config).

oderwat, 2018-06-10:
I think I just continue my `gcwiz` coding in my own repository. Thanks for all the input and clarifications.

closed 2018-06-10 by oderwat