Klipper3d/klipper #362: M204 -> SET_VELOCITY_LIMIT

ChiliApple, 2018-05-30:
Hy,
how does the M204 go in SET_VELOCITY_LIMIT?

Did I understand correctly that the M204 GCode command must be changed by a PostScript to the SET_VELOCITY_LIMIT command?
does anyone have a script or an example?
i use Slic3r PE

thx, Udo

oderwat, 2018-05-30:
I did not discover the possibilities with that, but I made a post processing script for Slic3r in the past. It was made to allow that I can add notices into the preferences which then add a pause at layer or z-height. This can be used for filament change or placing hardware into the print. It also adds M117 commands as I use them in S3D to show Layer numbers in the display.

This was made for myself and is not "production" quality and in the current way only works on OS X easily. It is written in PHP and you need an PHP interpreter on Windows to use it.

https://github.com/oderwat/slic3r-nanny

KevinOConnor, 2018-06-15:
Given that M204 is produced by several common slicers, I think it makes sense to add this to Klipper (commit efe63d5e).

-Kevin

closed 2018-06-26 by KevinOConnor

CameronLamont, 2018-09-01:
Just now having issues using Slic3r PE 1.41 - apparently the S parameter is no longer standard for Marlin but it was in the past.  After testing the same model on 1.40 with Klipper I suddenly have an error basically saying there is no 'S' parameter in the M204 commands it now writes up front.

e.g. 
Error on 'N3 M204 P1500 R1500 T1500*84': missing S

Seems this latest version has the following at the very start now - I can't see a config option for it

M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec

I've found this in my searching https://github.com/slic3r/Slic3r/issues/2745 where the S was retired in lieu of separate retraction, printing and travel speeds.  Also https://www.reprap.org/wiki/G-code#M204:_Set_default_acceleration

I've had a quick look at the klipper code in toolhead.py; not sure just yet how to make the 'S' parameter optional and equivalent to 'P' (as per Marlin)

Also to recognise the P, R T variables? Guessing they should be equivalent to the acceleration parameters from printer.cfg

cfelicio, 2018-10-18:
Just change Slic3r to generate gcode for Reprap instead of Marlin, and you should get output with S values