Klipper3d/klipper #363: [Feature request] Print resume on power fail or other problem

Nume1977, 2018-05-31:
In case of power failure it would be interesting to also have this feature.

In fact it is easy to implement, and two things are required:

a) klipper must create a file in the octoprint directory while a print is going on
b) a new function must be created to handle the print resume

On a),
Create a file named RESUME_original_filename.gcode and keep updating it's contents with the last printed line number, i.e. "RESUME 12346"
In case of a power failure or any other problem, this is the file that must be printed to resume.
The file is deleted upon print completion.

On b) 
Create a klipper function called i.e. "RESUME <line number>" that once called will load "_original_filename.gcode", parse temperature settings, do a G28 and restart printing the file from <line number> + 1

Off course this is the most simple way of resuming, and the stopped print must be removed from the bed before continuing, has it will not resume on the existing print itself (on delta it can be done, but not on cartesians, because G28 will crash the print head on the print).

Regards,
Rui

hg42, 2018-05-31:
some thoughts:

I have done resumes a lot of times manually.
From my experience the situation from which to resume is changing.
Repeating situations are usually solved and don't exist anymore.

With good adhesion, the part is usually still in place. Because the toolhead is stopped and you need some time to move it away from the part, there is usually some blob around the nozzle caused by it's renaining heat
The motors might have skipped.

I don't think automatic homing in any way would work all the time. This should be done manually. It's easy enough. The blob must be removed, because the toolhead will crash into it otherwise.

So, the main problem is storing the current position.
The other problem is setting the positions of axes and extruders to the correct values before sending the first move after the failure. This is because gcode usually uses ansolute positioning.

I think, a resume would usually be a function of the host software. But because of buffering, the firmware is the only one knowing the exact position. So, the firmware might be a better place for resume functionality.

gcode files can be very different. E.g. it could be using relative moves or some mixed forms. It can also use non-standard commands like Klipper's text commands.
The fimware is the only one knowing the results.
I guess the most secure way to handle all possibilities would be to completely play the gcode file up to the resume position, but without executing normal moves (but executing temperature and eventually homing commands, you would have to manually move the toolhead at a secure position before).

The resume would use a move to resume_z+delta, then resume_x/y, then resume_z to not collide with the part and set all extruders to the current position.

Storing the position all the time will probably quickly wear out flash storage (e.g. sd).

Nume1977, 2018-05-31:
@hg42 some good points there! I just went with the creality "resume function" that was adopted by marlin a few commits back. Of course there are better ways, but in practice one could start with a simple approach and evolve from there.

oderwat, 2018-05-31:
I don't think it is worth the trouble and getting an UPS for shorter outages goes long way.

hg42, 2018-05-31:
well, I thought more of "other problems" :-)
There are quite a lot of reasons why resuming a print could be useful.
The probability increases with the total print time and how experimental you are.
Functional parts for house and garden etc. often don't need perfect appearance or get a finish anyways.

oderwat, 2018-05-31:
I don't think it is a question about how useful it is. Sure it would be useful if it would work "automatically". I continued some prints with S3D but never because of a power outage.

I just measured the height it had printed so far, then started printing from there by setting this height as starting height in the slicer. This works well if you can home the printer manually and continue printing after cleanup.

If you can't home or removed the object. Just measure it and shove it into the buildplate (-Z) for that height.

KevinOConnor, 2018-06-15:
I agree that it would be useful to be able to recover from a failed print.  Though, it may be better to do this in OctoPrint (or similar) than in Klipper.  In any case, I'm open to someone submitting code to implement this.  I don't have any immediate plans to implement this myself.

-Kevin

hg42, 2018-06-22:
summary::
* it's often possible to resume manually by
   * measuring the height
      * sometimes difficult or even  impossible
      * often inaccurate
   *  ignoring the position inside the last printed plane
      * works if certain conditions are met
         * low layer heights
         * low viscosity (e.g. high temperatures)
      * may fail otherwise or give bad results
* remembering position of failure (xyze) improves the result
* because of queuing and buffering the firmware is the only one knowing the last move
* storing the complete position on each move may quickly wear out flash memory
* the last position could be reported to the host
   * on each finished layer
   * on motor off
   * on shutdowns
* resuming automatically wouldn't be safe in most cases

KevinOConnor, 2018-06-30:
It doesn't appear this is actively being worked on, so I'm going to close it for now.

closed 2018-06-30 by KevinOConnor

gonzoga, 2019-11-28:
I know this feature was mentioned a bit ago. I would be interested in working on this feature... if you don't mind.