Klipper3d/klipper #368: connect the BL Touch....problem and question

Rivega42, 2018-06-02:
Good afternoon. I have a  problem on  ...
I connect the BL Touch( to ramps1.4) with the instructions
"https://github.com/KevinOConnor/klipper/blob/master/config/sample-bltouch.cfg"

"Recv: // Unable to translate pin name: config_end_stop oid = 14 pin = z_virtual_endstop pull_up = 1 stepper_count = 1"

Tell me please how to understand?

In the standard configuration, the table (the Z axis) is lowered to the optical sensor (the coreXY printer) and everything is OK, but I want to use the BL Touch, which means the table should go up. Please tell me how to configure it.

Rivega42, 2018-06-02:
"endstop_pin:  probe:z_virtual_endstop"
It's right?

MrKekson, 2018-06-03:
Dont have the probe, but the 11. line from extra-config contain this: # Z height probe. One may define this section to enable Z height
# probing hardware. When this section is enabled, PROBE and
# QUERY_PROBE extended g-code commands become available. The probe
# section also creates a virtual probe:z_virtual_endstop pin. One may
# set the stepper_z endstop_pin to this virtual pin on cartesian style
# printers that use the probe in place of a z endstop.
#[probe]
#pin: ar15
#   Probe detection pin. This parameter must be provided.
#z_offset:
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
#speed: 5.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.
#activate_gcode:
#   A list of G-Code commands (one per line) to execute prior to each
#   probe attempt. This may be useful if the probe needs to be
#   activated in some way. The default is to not run any special
#   G-Code commands on activation.
#deactivate_gcode:
#   A list of G-Code commands (one per line) to execute after each
#   probe attempt completes. The default is to not run any special
#   G-Code commands on deactivation.

So i think you need to set up the probe, and then set the virtual end-stop. But if i'm right, you don't have to, you can use both, the probe and the end-stop, got no experience on that part tho.

KevinOConnor, 2018-06-03:
If you are having issues with Klipper, you need to attach the full klipper log file as described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Rivega42, 2018-06-04:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2069219/klippy.zip)

closed 2018-06-04 by Rivega42