Klipper3d/klipper #369: MCU 'mcu' shutdown: Timer too close when starting a print

marcuswu, 2018-06-03:
It looks to me like it is sending the command to heat the print bed and then immediately sending the command to heat the extruder without waiting:

```
Send: N6 M190 S60*88
Recv: T0:23.7 /0.0 B:22.9 /60.0
Recv: ok
Send: N7 M104 S205 T0*33
Recv: // MCU 'mcu' shutdown: Timer too close
```

It's totally possible I have something misconfigured, but I'm not sure what if anything I changed since the last time I printed (it's been a while).
[klippy.log](https://github.com/KevinOConnor/klipper/files/2065955/klippy.log)

oderwat, 2018-06-03:
I also got an error related to timing. When I set pressure advance to 5.0 I immediately get the message that the timer is in the past. I don't have the correctly worded message (and my printer runs) but found it a strange thing to happen.

KevinOConnor, 2018-06-03:
On Sun, Jun 03, 2018 at 05:59:13AM -0700, Marcus Wu wrote:
> It looks to me like it is sending the command to heat the print bed and then immediately sending the command to heat the extruder without waiting:

Alas, you've hit a bug in Klipper.  The bug is specific to the
handling of the Replicape's global enable pin.  Specifically, Klipper
raced with the delayed fan disablement (M106 S255 ; G4 P500 ; M106 S0)
and the following bed heater enable (M190 S60).

I need to think about how to best fix this.  In the meantime, you
should be able to avoid this by either removing the fan toggle or
enabling the bed heater prior to toggling the fan.

-Kevin

marcuswu, 2018-06-03:
> In the meantime, you should be able to avoid this by either removing the fan toggle or enabling the bed heater prior to toggling the fan.

This workaround did the trick. I'm up and printing again. Thanks!

Grimpster, 2018-07-02:
I know it's a bit old.
But I've seen to hit this "bug" or limitation as well. However it started happening at random parts of the same print.

The only value I've been messing around with is the PA from .25 to 1.1
But I think I might have hit the actual limit to the ramps1.4/2560 - The fix seemed to be reduce speed overall on the printer, especially on the extruder.

But this seems contradicting as it happened different places on the same print.

KevinOConnor, 2018-07-04:
This issue is limited to a specific gcode sequence on the Replicape.  If you're seeing a "timer too close" error on a ramps it is not related to this.  A "timer too close" is typically the result of comms issues, but you can open a new issue with the klipper log file and we'll take a look.

-Kevin

ckvsoft, 2023-01-14:
I think i have a similar problem.
Since the update to v0.11.0-66-g4671cf2d, the same message with firmware stop comes if I want to cancel a print that is still waiting for the temperature of the heating bed/extruder in the warm-up phase.

PR #5984 (2023-01-17): replicape: Only call note_pwm_enable() in main thread
The pca9685_pwm set_pwm() code can be called from a background thread (when the pin controls a heater).  Propagate updates to the main thread for enable tracking.

This should fix #369.  I don't have the replicape hardware to actually test this.

-Kevin
closed 2023-01-17 by KevinOConnor