Klipper3d/klipper #370: Output_pin should use scale from gcode

mvturnho, 2018-06-03:
Hi,

I added a pwm output_pin for my led strip and ramps board fan and want to use the scale parameter in the printer.cfg

I use a scale value of 255. But when I send the gcode to set the value for the pin to 255 the printer responds that the value should be between 0 and 1

I tried adding a control to octoprint but it seems that octoprint can't handle the float value between 0 and 1

could we make the gcode accept the value configured in the scale parameter for this pwm pin?

mvturnho, 2018-06-04:
some extra info:

in my printer.cfg

```
[output_pin headled]
pin = aux1:ar9
pwm = True
cycle_time = 0.0100
hardware_pwm = True
value = 0.05
scale = 255
shutdown_value = 1
```

from the octorpint terminal I send the folowing gcode

```
RESTART
SET_PIN PIN=headled VALUE=200
```

Then my terminal log says
```
Send: SET_PIN PIN=headled VALUE=200
Recv: !! Error on 'SET_PIN PIN=headled VALUE=200': VALUE must have maximum of 1.0
Recv: ok
```

KevinOConnor, 2018-06-05:
Yes - it looks like that is a bug in Klipper.  It looks like maxval=self.scale is the correct fix.  Commit 183e0dec.  Thanks.

-Kevin

mvturnho, 2018-06-05:
Okay thanks. I pulled the update and it works okay now.
I close the issue.

closed 2018-06-05 by mvturnho