Klipper3d/klipper #373: Failed during Make flash

ejg3855, 2018-06-05:
So I had some issues with a previous sd card. I've been trying to reflash klipper but keep getting a failure.
- Install octopi and setup, done.
- cd /~klipper, done
- make menuconfig, done,set
- make, done

the it gets confusing
- sudo service klipper stop, done
- make flash FLASH_DEVICE=/devttyACM0, fails.

Am I supposed to run these commands to get this setup, I am a pi novice so following the guide to a T for me has worked in the past. I had klipper up and working but cannot for the life of me remember what I did or didn't do. Any advice would be great.

Hardware is basic Pi3, Ramps 1.4

`
###### Installing system start configuration...

###### Launching Klipper host software...
[ ok ] Restarting klipper (via systemctl): klipper.service.
pi@octopi:~ $ cd ~/klipper/
pi@octopi:~/klipper $ make menuconfig
scripts/kconfig/conf --olddefconfig /home/pi/klipper/src/Kconfig
#
# configuration written to /home/pi/klipper/.config
#
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

pi@octopi:~/klipper $ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.6.0-130-ga748cf3-20180605_003252-octopi
  Linking out/klipper.elf
  Creating hex file out/klipper.elf.hex
pi@octopi:~/klipper $ sudo service klipper stop
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/ttyACM0
  Flashing out/klipper.elf.hex to /dev/ttyACM0 via avrdude
avrdude: ser_open(): can't open device "/dev/ttyACM0": No such file or directory
ioctl("TIOCMGET"): Invalid argument
ioctl("TIOCMGET"): Invalid argument
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

src/avr/Makefile:27: recipe for target 'flash' failed
make: *** [flash] Error 1`

oderwat, 2018-06-05:
Try executing `ls /dev/serial/by-id/` and pick the port of your printer. I don't think `/dev/ttyACM0` is what you need.

ejg3855, 2018-06-05:
does the ramps need to be connected? could it be that simple?

oderwat, 2018-06-05:
Yes. I can't flash the software to the board if it is not connected.

ejg3855, 2018-06-05:
if i shut down the pi (move it physically) do I need to sequence through the commands again or can I just go back to the make flash part?

ejg3855, 2018-06-05:
pi@octopi:~/klipper $  ls: cannot access '/dev/serial/by-id/': No such file or directory
-bash: ls:: command not found

ejg3855, 2018-06-05:
I'm starting to think my MEGA died, windows doesn't recognize it.

KevinOConnor, 2018-06-15:
This looks like a system issue.

closed 2018-06-15 by KevinOConnor