Klipper3d/klipper #375: Internal error on command:"M107"

forerunnert, 2018-06-05:
Hi Kevin,
I installed klipper today after running Repetier for a couple of years. 
The printer seems fully functional now entering manual commands or using Octopi controls but I'm running into problems when executing gcode on it.

The issue I'm seeing is "Internal error on command:"M107"". The gcode sequence is:
;FLAVOR:Marlin
;TIME:1190
;Filament used: 1.04859m
;Layer height: 0.15
;Generated with Cura_SteamEngine 3.3.1
M104 S230
M109 S230
M82 ;absolute extrusion mode
G21        		;metric values
G90        		;absolute positioning
G28 X0 Y0  		;move to endstops
G92 E0            ;zero the extruded length
G1 X0 Y0 Z100
G1 E6 ;insert filament (was retracted last print)
G92 E0            ;zero the extruded length again
G92 E0
G1 F12000 E-5.35
;LAYER_COUNT:20
;LAYER:0
M107
G0 F5000 X17.491 Y-28.24 Z0.3

If I remove the M107 command from the gcode I get the same error on the next G1 command, and so on.

Any idea what is going on?
Thanks for the help!

[klippy.log](https://github.com/KevinOConnor/klipper/files/2073470/klippy.log)

KevinOConnor, 2018-06-05:
On Tue, Jun 05, 2018 at 06:05:34PM +0000, forerunnert wrote:
> I installed klipper today after running Repetier for a couple of years.
> The printer seems fully functional now entering manual commands or using Octopi controls but I'm running into problems when [...]

Your extruder step_distance does not look correct.  As currently
configured, each step is nominally moving the extruder ~2 nanometers.

The above retraction is a request to retract 5.35mm at 200mm/s which
would require over 90 million steps per second with the given step
distance.  That's multiple steps per micro-controller clock tick -
which flagged an internal sanity check.

So, double check your extruder step_distance.

-Kevin

forerunnert, 2018-06-05:
In Repetier I bound speeds by the firmware. I thought I caught them all in Cura but I missed the extruder..
The extruder step_distance was a decimal error: it was 459.47 steps/mm so 0.002176 step_distance.

Printing as we speak, thanks a lot!

closed 2018-06-06 by forerunnert