Klipper3d/klipper #379: Delta: Allow setting printable radius to a value different from "delta_radius"

ruevs, 2018-06-07:
Hi first of all - great job on this firmware! Clean architecture, nice documentation, worked from the first attempt! On a delta the steppers move noticeably smoother than with Repetier and much, much smoother than with Marlin.

After configuring a delta printer (https://github.com/KevinOConnor/klipper/pull/378) I noticed that the `delta_radius` option, which sets the geometry of the printer, is the only "radius" that can be configured.

In delta.py in the `check_move` function there is rather sensible code that checks whether the end position of the move is in the printable area:

    self.max_xy2 = min(radius, min_arm_length - radius,
                       ratio_to_dist(4. * SLOW_RATIO) - radius)**2

Where `radius = delta_radius`
This code presumes that the configured geometry is the only thing limiting the printable radius. However this is not always the case. For example for the Anycubic Kossel that I configured `delta_radius: 99.8` but the actual printable radius is only 90mm. In my opinion there should be a `printable_radius` configuration option that should be included in the above `min` statement.

hg42, 2018-06-07:
note, deltas can also print outside the tower radius!
Some have a square bed.
Some have a hexagonal or octagonal bed.
And there might be even more possibilities.

KevinOConnor, 2018-06-07:
On Thu, Jun 07, 2018 at 05:31:47AM -0700, ruevs wrote:
> Hi first of all - great job on this firmware! Clean architecture, nice documentation, worked from the first attempt! On a delta the steppers move noticeably smoother than with Repetier and much, [...]

Thanks.  If you're comfortable making code changes, feel free to add
an additional configuration parameter (and corresponding
documentation) and submit a pull request.

-Kevin

KevinOConnor, 2018-06-07:
On Thu, Jun 07, 2018 at 06:34:32AM -0700, Harald wrote:
> note, deltas can also print outside the tower radius!

Indeed.  Klipper currently uses a very simplistic check though - the
valid region is modeled as a cylinder with a cone on top.

I haven't seen an indication this is a problem in practice, but if
someone wants to implement a more complete check then that is fine.

-Kevin

hg42, 2018-06-07:
> I haven't seen an indication this is a problem in practice

if my delta would still run, it would suffer from the limit I think. It had a big square bed (300x300) and two corners were outside the tower radius.
I think specifying a print_radius would be enough.

Its a general problem, most checks of this kind are simplistic and for many cases too simplistic.
In general it would need a 3D model and it also depends on the toolhead (which can be changed and often is asymmetric).
E.g. my delta had some restrictions above 300 mm (I attached an extruder there).

Nobody wants to check a complex model all the time.
It's more a task for the slicer or host software, because the print model could be intersected with the restriction model.

So, simply switching off the restriction for such special geometries and only for such big objects is the easiest way to go.

KevinOConnor, 2018-06-30:
It doesn't appear anyone is actively working on this, so I'm going to close this for now.

closed 2018-06-30 by KevinOConnor