Klipper3d/klipper #381: Error using endstop pin on second MCU

aegelsky, 2018-06-07:
Dear Kevin

I'm building quite a monster with your Klipper:
https://pp.userapi.com/c834401/v834401964/1639bf/JYtQ_EY-3X4.jpg
it has build volume 1200х2200х2000mm, so I had to use 6! Z motors

having trouble configurating them now, because first Z motor with it's endstop is on main MCU
and all others - on second (zboard), attaching config and log,
it says
Reported error: Endstop and stepper must be on the same mcu
when I use Z motor from main MCU as main and all stepper_zN without endstops
not sure why it should be forbidden to use endstop from another MCU (could you answer this?) 

so I made second (dummy for now) endstop on main MCU and used it's Z motor in config as stepper_z1,
main Z motor is on second MCU and all other steppers uses (should) this main's endstop (on second MCU)
here is picture, not sure it explains something though:
https://pp.userapi.com/c845016/v845016928/73fe3/EJ7tWTm4eaw.jpg

now error is 
Reported error: Internal error on command:"G28"

any ideas how could I do this?
it's a test version, in final every motor will have endstop for it, but I don't see why this would not work,
because your config says it should (apart from first question why I should not use endstop from another MCU)
line 1445 in log says something, but I don't get it, not too good with python, sorry 

sftp://192.168.1.128/home/pi/klippy.log
sftp://192.168.1.128/home/pi/printer.cfg

wizhippo, 2018-06-07:
@aegelsky  How big is your nozzle going to be?  Can't wait to see one of your multi-week prints.

KevinOConnor, 2018-06-07:
On Thu, Jun 07, 2018 at 03:13:49PM +0000, aegelsky wrote:
> I'm building quite a monster with your Klipper:
> https://pp.userapi.com/c834401/v834401964/1639bf/JYtQ_EY-3X4.jpg
> it has build volume 1200х2200х2000mm, so I had to use 6! Z motors
>
> having trouble [...]

During a homing operation the micro-controller internally reacts to
the endstop signal and halts stepper movement.  In order for this
process to work correctly, the micro-controller must have the signal
locally - it can't be a signal received on another micro-controller.

FYI, it should be possible to wire the endstop pin in parallel to both
micro-controllers so that both micro-controllers receive the signal
when the endstop triggers.

> now error is 
> Reported error: Internal error on command:"G28"
> 
> any ideas how could I do this?

Unfortunately, you've encoutered a problem in Klipper.  It has to do
with the way homing is handled (all steps taken during a homing
operation must fit in the micro-controller ram) and how step
compression works.  In particular, the splitting of the Z between two
micro-controllers, combined with the extremely large Z axis is causing
the Z movement of the single stepper on the main mcu to not fit within
the mcu ram.

I don't have a good fix for this.  I suspect you could workaround the
issue for now by reducing your Z position_max from 2000 to 500.

-Kevin

hg42, 2018-06-07:
if you don't want full control over each Z-stepper you could connect pairs of steppers in series, which reduces the number of Z-axis drivers to three.

ear0wax, 2018-06-08:
Just an idea but you could use a stepper expansion board to add a 6th stepper to that one mcu, that way you would have all 6 steppers with their own driver and you can just wire the Z end stop to the one mcu

aegelsky, 2018-06-29:
@KevinOConnor Just info, moved all to second board and it kinda works with 6 motors (4+2 on Z slot).
Got bunch of new problems, will update soon.
Most annoying is ADC shutdown, though wiring is OK and Marlin worked smooth on the same board.
And when Klipper finally power ups, temperatures are constant and consistant.
Any way to suppress/ignore this error?

@hg42 those are the most powerful/current consuming motor from nema17, 1.8A, they don't work in series and 1.8A is nearly the limit for "dupont" 2.54 connector of driver.
Had only option to use two smaller motors in the back, they are in series in Z slot of RAMPS, works.
Also A4988 sucks here, only drv88xx did the job.
Going to print car parts from my designs and buildings outter/inner "barocco/art deco" decals, I'm in love with sweet arts :) will keep you posted if you like.

@wizhippo it's Volcano E3D, now it's 0.8, have up to 1.2mm or even 1.6.
will keep you posted if you like too.

Thanks for support, all you guys, really appreciate it, it's third year I'm making it and I'm kinda on my limit of patience :)

hg42, 2018-06-30:
just an idea:
for such a big printer, you might consider to use a hangprinter, which seems to be much easier to build.
Though I don't have any experiences with such a printer, e.g. according to speed, resolution etc.

closed 2018-07-07 by KevinOConnor

KevinOConnor, 2019-07-10:
FYI, the underlying issue should now be fixed (PR #1766).

-Kevin

thepingas, 2019-10-01:
With this change, is running an endstop switch on a different MCU than that which controls the axis stepper is now a supported mode? Or do you mean that the groundwork for future implementation has been established?

I tried setting up such a configuration and Klipper throws an error:
"Error: Endstop and stepper must be on the same mcu".

I am using the present master branch.

I may just be confused ( I frequently am).