Klipper3d/klipper #383: MCU Shutdown due to Internal error on command:"G1"

TCL987, 2018-06-08:
My printer just shutdown mid-print due to an error processing a G1 command in the middle of a layer. There doesn't appear to be anything unique about the G1 commands dumped into the crash log so I am at a loss as to what caused the error.

[KlipperCrash.zip](https://github.com/KevinOConnor/klipper/files/2086444/KlipperCrash.zip)
[image]
[gcode_file.zip](https://github.com/KevinOConnor/klipper/files/2086460/gcode_file.zip)

KevinOConnor, 2018-06-09:
Alas, you've hit a bug in Klipper.  It's an obscure rounding issue and loss of precision on some "double" floating point math.  I've been able to put together a workaround for it, and I've uploaded it to a work-corexy-20180608 branch:

cd ~/klipper ; git fetch ; git checkout origin/work-corexy-20180608 ; sudo service klipper restart

I need to test this further before committing to the main branch.

-Kevin

KevinOConnor, 2018-06-20:
FYI - this issue should now be fixed on the master branch (primarily as of commit fc4a9e75).

-Kevin

closed 2018-06-30 by KevinOConnor

drelich, 2019-09-23:
Hate to re-open the issue but I just got the Internal error on command:"G1" error on my Ender 3. I was printing an xyz calibration cube when this happened. I should probably note that I am testing the new s-curve feature I installed earlier today.

robustini, 2019-12-29:
Same problem here #2334.