Klipper3d/klipper #385: FLASH did not work after update

StefanMeGit, 2018-06-10:
I updated my pi tot he newest version as I did many times before, but I get this error...

```
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/ttyUSB0
Makefile:78: recipe for target 'out/src/sched.o.ctr' failed
make: *** [out/src/sched.o.ctr] Illegal instruction
```
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2087864/klippy.log.zip)

cmock, 2018-06-11:
I think I also had that recently and solved it with

    make clean
    make

KevinOConnor, 2018-06-11:
An "Illegal instruction" is usually an indication of a corrupt sd card.  But, first check with a full rebuild as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

StefanMeGit, 2018-06-11:
Thats the funny thing, I already did it like in the description. I order a new SD Card before I make more tests about this. I ll give feedback as soon as possible

StefanMeGit, 2018-06-16:
After a an complete new install of Octoprint, id worked on the third try...

closed 2018-06-16 by StefanMeGit