Klipper3d/klipper #388: [Request] Porting guide for new MCUs

FHeilmann, 2018-06-11:
Hello,
I am currently attempting to port klipper to the Duet 2 generation of boards and there are some things which I am not sure about.

I looked at the code for the existing MCUs and found that I am unable to deduce which functions are required for a given MCU-port to work. For example, gpio_peripheral is defined for the sam3x8e and lpc176x but, as far as I can tell, only used internally by these specific ports.

I think having a template implementation of a new MCU, including code stubs for all functions that need to be implemented (while also hinting at the existing, MCU agnostic functions that may be used like irq_save/irq_restore for arm) and descriptions about what exactly each function is supposed to accomplish would be very beneficial for people like me who attempt to port klipper to new MCUs.

While looking at the sam3x8e code and porting to sam4e8e I noticed that utilizing a sdk such as the atmel software framework could potentially enable the porting of klipper to many architectures at very little extra cost. However the sam3 port utilizes low level code and only relies on the CMSIS layer. Is the usage of an SDK frowned upon or can I use it?

Thanks for your time and I hope to be able to submit a PR soon.

mvturnho, 2018-06-11:
Hi, 

I am also starting a port to the esp32 to drive some peripherals. It would be verry nice to have such documentation / template project.

Teslastellar, 2018-06-11:
Hello,

Thank you all for your time and effort.
I was wondering about the ESP32 and the Duet and would be very interested
in the results of your work. I'm also trying to go through the code to see
if I can add closed-loop control (not the kind offered by step encoders but
real positional awareness and PID corrections). Just getting started but
this would greatly simplify my work too.
Would be interested to know if anyone has tried anything similar.

Regards,
Seyed

KevinOConnor, 2018-06-11:
On Mon, Jun 11, 2018 at 12:41:08AM -0700, Florian Heilmann wrote:
> Hello,
> I am currently attempting to port klipper to the Duet 2 generation of boards and there are some things which I am not sure about.
>
> I looked at the code for the existing MCUs and found that I [...]

Yes, gpio_peripheral() is just an internal helper (it is used to set
the pin "function" on the particular board).

My advice for bringing up a new port would be:

- first bring up serial and the timer - see commit 970831ee as an
  example.  This is the hardest part of a port - once one can
  communicate with the board everything else becomes dramatically
  easier.  Link in the src/generic/ c files to take advantage of the
  generic implementations of architecture specific calls where
  possible.

- then bring up basic gpio in/out - see commit c78b9076 as an
  example.

- then bring up additional peripherals - for example commit 65613aed,
  c812a40a, and c381d03a.

> I think having a template implementation of a new MCU, including code stubs for all functions that need to be implemented (while also hinting at the existing, MCU agnostic functions that may be used [...]

The src/simulator/ directory was intended to do this.  It should have
the minimum calls necessary to bring up a new architecture.  It is not
well documented though.

> While looking at the sam3x8e code and porting to sam4e8e I noticed that utilizing a sdk such as the atmel software framework could potentially enable the porting of klipper to many architectures at [...]

It's fine to use an SDK as long as the SDK is GPLv3 compatible.  For
example, the stm32f1 code uses the "STM low level" interface code.

-Kevin

FHeilmann, 2018-06-11:
Thank you for your input, that's really helpful. Is the Sam3x8e port considered stable? I may use it to gain some insight on what I have to do.

Cheers

KevinOConnor, 2018-06-11:
On Mon, Jun 11, 2018 at 01:38:36PM +0000, Florian Heilmann wrote:
> Thank you for your input, that's really helpful. Is the Sam3x8e port considered stable? I may use it to gain some insight on what I have to do.

The sam3x8e is in the Arduino Due, and that seems to be a pretty
popular board to use with Klipper.

-Kevin

dim08, 2018-06-12:
How about ESP32, it seems it can run MicroPython. Wonder if its resources might be enough to run the server side ?

FHeilmann, 2018-06-12:
> How about ESP32, it seems it can run MicroPython. Wonder if its resources might be enough to run the server side ?

Micropython does not have access to the full python API. While I don't see why not you want the server side to be as powerful as possible so using a microcontroller for the server side kind of goes against what klipper is trying to achieve.

dim08, 2018-06-12:
> While I don't see why not you want the server side to be as powerful as possible.

Power consumption, price, PCB footprint … Python is not optimal in terms of speed, probably klipper should be rewritten in C ?

FHeilmann, 2018-06-12:
> Power consumption, price, PCB footprint … Python is not optimal in terms of speed, probably klipper should be rewritten in C ?

Not to sound rude, but I think you misunderstand the purpose of klipper. The idea is to have a more powerful planner than what is possible on current MCUs. So klipper outsources as much computation as possible to the server (klippy). 

Using a weak (compared to an rpi) MCU to run the server re-introduces the problems that klipper aimed to solve in the first place. Even porting klippy to C is not going to solve that. Moreover, you'd need a third device to make the klippy server accessible again, since running a host software together with klippy on a low cost MCU is certainly not possible.

If you care about footprint and power consumption, using only the 3D printer board with a suitable firmware such as Marlin or (in case of duet) RRF is your best bet.

dim08, 2018-06-12:
Currently porting RRF to 100 pin SAM4Es, but some issues – not all peripherals are presented as well as experiencing some peculiarities. The timers work on some pins not on others, can not configure the JTAG pins as GPIO … You are from Germany ? Do you get parts from INELTEK ? Any issues with them ?

KevinOConnor, 2018-07-11:
FYI, I added some porting information to docs/Code_Overview.md (commit f60b0093).

@FHeilmann - how is the SAM4 port coming along?

-Kevin

FHeilmann, 2018-07-13:
@KevinOConnor Serial and Timers are working (at least I have comms and some stats about the clocksync are making it to klippy), currently working on GPIO and setting up a JTAG debug environment.

KevinOConnor, 2018-07-17:
Okay, thanks for the update.  I usually debug gpio by using console.py (see docs/Debugging.md) to send set_digital_out commands on led pins.  If one can control the leds, than it generally means gpios are working and the problem is with configuration of the stepper drivers.  Let me know if you need any assistance.

I'm going to close this issue as I think the main item (porting docs) has been addressed.

closed 2018-07-17 by KevinOConnor