Klipper3d/klipper #389: Firmware halt on Z_TILT_ADJUST when no G28 was issued.

mvturnho, 2018-06-11:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2089971/klippy.log)

[printer.zip](https://github.com/KevinOConnor/klipper/files/2089973/printer.zip)

Hi, 
I added 4 steppers to test the multi z axis bed leveling, and it works nicely.
There is only one problem; when I issue  Z_TILT_ADJUST before I done G28 (initialize home) the firmware crashes. I have to do FIRMWARE_RESTART.

I sugest to not emergency stop the firmware but to just return a message that we need homing before we can issue Z_TILT_ADJUST.

My log and printer config are attached.

KevinOConnor, 2018-06-12:
Thanks.  I think it should be fixed now (commit 3819ad29).

-Kevin

mvturnho, 2018-06-12:
Hi Kevin,

tested it just now and works just fine.

```
Send: Z_TILT_ADJUST
Recv: !! Must home axis first: -30.000 10.000 10.000 [0.000]
Recv: ok
Send: M105
```

gr. Michiel

closed 2018-06-12 by mvturnho