Klipper3d/klipper #390: console.py hangs

yschroeder, 2018-06-11:
I am new to klipper and trying to get it running on the Blue Pill STM32F103 (for a project that will not be a 3D printer). I got the code running on the MCU and connected it via an FTDI to the Raspberry Pi (pins A9 and A10).

Now I am trying to use the debug console `console.py` to check if everything works (toggle the on-board LED for example). The board responds and I can toggle the LED, however, after some time the console hangs and the MCU does not accept commands anymore:

```
pi@raspberrypi:~/klipper $ ~/klippy-env/bin/python ./klippy/console.py /dev/ttyUSB0 250000

  This is a debugging console for the Klipper micro-controller.
  In addition to mcu commands, the following artificial commands are
  available:
    PINS  : Load pin name aliases (eg, "PINS arduino")
    DELAY : Send a command at a clock time (eg, "DELAY 9999 get_uptime")
    FLOOD : Send a command many times (eg, "FLOOD 22 .01 get_uptime")
    SUPPRESS : Suppress a response message (eg, "SUPPRESS stats")
    SET   : Create a local variable (eg, "SET myvar 123.4")
    STATS : Report serial statistics
    LIST  : List available mcu commands, local commands, and local variables
    HELP  : Show this text
  All commands also support evaluation by enclosing an expression in { }.
  For example, "reset_step_clock oid=4 clock={clock + freq}".  In addition
  to user defined variables (via the SET command) the following builtin
  variables may be used in expressions:
    clock : The current mcu clock time (as estimated by the host)
    freq  : The mcu clock frequency

==================== attempting to connect ====================
INFO:root:Starting serial connect
DEBUG:root:Starting stk500v2 leave programmer sequence
DEBUG:root:Got '<6' from stk500v2
DEBUG:root:Unknown message 43 (len 14) while identifying
Loaded 52 commands (v0.6.0-133-ge90edc0-20180608_070954-raspberrypi / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU config: RECEIVE_WINDOW=192 ADC_MAX=4095 CLOCK_FREQ=8000000 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 MCU=stm32f103 STATS_SUMSQ_BASE=256
DEBUG:root:new minimum rtt 270959.771: hrtt=0.002339 freq=8000000
DEBUG:root:new minimum rtt 270960.027: hrtt=0.001496 freq=8000027
====================       connected       ====================
006.333: stats count=64 sum=3726 sumsq=1974
011.338: stats count=55 sum=2162 sumsq=792
016.328: stats count=55 sum=2163 sumsq=794
021.334: stats count=55 sum=2162 sumsq=792
026.323: stats count=55 sum=2163 sumsq=794
DEBUG:root:new minimum rtt 270986.786: hrtt=0.001488 freq=8000372
DEBUG:root:new minimum rtt 270987.772: hrtt=0.001311 freq=8000374
DEBUG:root:new minimum rtt 270988.757: hrtt=0.001224 freq=8000374
031.328: stats count=55 sum=2165 sumsq=792
DEBUG:root:new minimum rtt 270989.743: hrtt=0.000540 freq=8000374
DEBUG:root:new minimum rtt 270990.728: hrtt=0.000488 freq=8000374
DEBUG:root:new minimum rtt 270991.714: hrtt=0.000445 freq=8000373
DEBUG:root:new minimum rtt 270992.699: hrtt=0.000359 freq=8000373
DEBUG:root:new minimum rtt 270993.684: hrtt=0.000296 freq=8000373
036.333: stats count=55 sum=2172 sumsq=921
DEBUG:root:new minimum rtt 270994.670: hrtt=0.000184 freq=8000373
041.323: stats count=56 sum=2335 sumsq=1025
046.327: stats count=55 sum=2152 sumsq=779
051.332: stats count=55 sum=2159 sumsq=790
056.322: stats count=55 sum=2145 sumsq=769
061.328: stats count=55 sum=2162 sumsq=792
066.335: stats count=55 sum=2169 sumsq=803
```

The time after which is hangs is more or less random between 10 seconds and a minute or so.
Is it supposed to be this way? Am doing something wrong?

KevinOConnor, 2018-06-11:
On Mon, Jun 11, 2018 at 07:27:18AM -0700, Yannic Schröder wrote:
> I am new to klipper and trying to get it running on the Blue Pill STM32F103 (for a project that will not be a 3D printer). I got the code running on the MCU and connected it via an FTDI to the [...]
[...]
> The time after which is hangs is more or less random between 10 seconds and a minute or so.
> Is it supposed to be this way? Am doing something wrong?

No, that is not normal.  It's possible the mcu is resetting, or the
serial connection got lost.  I don't have either problem on my "blue
pill" test board.

Some things you could try:

- run the STATS command and see if the retransmits or "invalid bytes"
  counter is increasing

- run the get_uptime command before the failure (the high field
  contains the top 32bits of a 64bit counter).  After a failure, try
  reconnecting and see if get_uptime got reset (indicating an mcu
  reset) or is continuing to increase (indicating a serial issue).

Also, you made sure to use 3.3v signaling on the serial pins, right?

-Kevin

yschroeder, 2018-06-12:
I am using 3.3v logic level for the FTDI.

I tested both your suggestions and the log output is the following:

```
pi@raspberrypi:~/klipper $ ~/klippy-env/bin/python ./klippy/console.py /dev/ttyUSB0 250000

  This is a debugging console for the Klipper micro-controller.
  In addition to mcu commands, the following artificial commands are
  available:
    PINS  : Load pin name aliases (eg, "PINS arduino")
    DELAY : Send a command at a clock time (eg, "DELAY 9999 get_uptime")
    FLOOD : Send a command many times (eg, "FLOOD 22 .01 get_uptime")
    SUPPRESS : Suppress a response message (eg, "SUPPRESS stats")
    SET   : Create a local variable (eg, "SET myvar 123.4")
    STATS : Report serial statistics
    LIST  : List available mcu commands, local commands, and local variables
    HELP  : Show this text
  All commands also support evaluation by enclosing an expression in { }.
  For example, "reset_step_clock oid=4 clock={clock + freq}".  In addition
  to user defined variables (via the SET command) the following builtin
  variables may be used in expressions:
    clock : The current mcu clock time (as estimated by the host)
    freq  : The mcu clock frequency

==================== attempting to connect ====================
INFO:root:Starting serial connect
DEBUG:root:Starting stk500v2 leave programmer sequence
DEBUG:root:Got '' from stk500v2
getLoaded 52 commands (v0.6.0-133-ge90edc0-20180608_070954-raspberrypi / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU config: RECEIVE_WINDOW=192 ADC_MAX=4095 CLOCK_FREQ=8000000 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 MCU=stm32f103 STATS_SUMSQ_BASE=256
DEBUG:root:new minimum rtt 330390.796: hrtt=0.001343 freq=8000000
_====================       connected       ====================
up002.286: stats count=104 sum=15271 sumsq=15086
time
002.826: uptime high=111 clock=3463842330
DEBUG:root:new minimum rtt 330392.194: hrtt=0.000998 freq=8000354
DEBUG:root:new minimum rtt 330393.179: hrtt=0.000921 freq=8000371
DEBUG:root:new minimum rtt 330394.164: hrtt=0.000876 freq=8000454
007.274: stats count=56 sum=2351 sumsq=934
012.278: stats count=55 sum=2171 sumsq=805
017.282: stats count=55 sum=2163 sumsq=793
022.286: stats count=55 sum=2171 sumsq=805
027.273: stats count=55 sum=2162 sumsq=792
get_uptime
029.927: uptime high=111 clock=3680620686
032.277: stats count=56 sum=2349 sumsq=928
037.280: stats count=55 sum=2170 sumsq=804
DEBUG:root:new minimum rtt 330426.681: hrtt=0.000609 freq=8000444
DEBUG:root:new minimum rtt 330427.666: hrtt=0.000535 freq=8000444
DEBUG:root:new minimum rtt 330428.651: hrtt=0.000427 freq=8000444
get_uptime
041.282: uptime high=111 clock=3771505326
042.284: stats count=57 sum=2514 sumsq=1149
047.271: stats count=55 sum=2171 sumsq=804
get_uptime
051.004: uptime high=111 clock=3849202756
052.275: stats count=56 sum=2357 sumsq=941
057.279: stats count=55 sum=2170 sumsq=803
062.283: stats count=55 sum=2172 sumsq=806
get_uptime
064.539: uptime high=111 clock=3957560428
067.272: stats count=56 sum=2322 sumsq=892
g072.276: stats count=55 sum=2164 sumsq=794
et_uptime
073.929: uptime high=111 clock=4032632534
077.282: stats count=56 sum=2356 sumsq=940
082.271: stats count=55 sum=2161 sumsq=791
get_uptime
085.113: uptime high=111 clock=4122111180
[... 81 lines cut ...]
044.448: stats count=55 sum=2164 sumsq=794
049.451: stats count=55 sum=2163 sumsq=793
STATS
bytes_write=791 bytes_read=3367 bytes_retransmit=9 bytes_invalid=49 send_seq=124 receive_seq=124 retransmit_seq=2 srtt=0.007 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=8000437
054.439: stats count=55 sum=2160 sumsq=791
059.445: stats count=55 sum=2170 sumsq=803
get_uptime
061.191: uptime high=112 clock=1138378924
DEBUG:root:new minimum rtt 330641.101: hrtt=0.000235 freq=8000437
064.448: stats count=56 sum=2340 sumsq=916
DEBUG:root:new minimum rtt 330642.086: hrtt=0.000179 freq=8000437
STATS
bytes_write=923 bytes_read=3658 bytes_retransmit=100 bytes_invalid=49 send_seq=146 receive_seq=140 retransmit_seq=145 srtt=0.010 rttvar=0.007 rto=4.940 ready_bytes=0 stalled_bytes=0 freq=8000436
STATS
bytes_write=977 bytes_read=3658 bytes_retransmit=707 bytes_invalid=49 send_seq=155 receive_seq=140 retransmit_seq=155 srtt=0.010 rttvar=0.007 rto=5.000 ready_bytes=0 stalled_bytes=29 freq=8000436
STATS
bytes_write=977 bytes_read=3658 bytes_retransmit=798 bytes_invalid=49 send_seq=155 receive_seq=140 retransmit_seq=155 srtt=0.010 rttvar=0.007 rto=5.000 ready_bytes=0 stalled_bytes=32 freq=8000436
^C
```

It starts with `uptime high=111` (I connected it yesterday and did not reset it on purpose). Later uptime changes to `112`. After about 147 seconds it freezes. The `STATS` command shows that the retransmit counter increases. I restarted the console and `get_uptime` returns `112` so the MCU did not reset. After about 64 seconds it freezes again and the retransmit counter starts to increase again.

So I suspect a serial communication error. I will test to connect it directly to the UART header of the Raspberry Pi and see if that changes anything.

By the way: Could there be any other program consuming bytes from the serial port so the console gets only half the data? It should not be the normal `klipper` service, that one is stopped.

yschroeder, 2018-06-12:
Ok, the problem seems to be my chinese (fake?) FTDI device. When connecting to the hardware UART of the Pi it works for at least 2300 seconds without any retransmission or invalid bytes.

I also tested a genuine FTDI TTL-232R-3V3 cable and that also works as well as the hardware UART (for 600 seconds now).

So it is not a problem of klipper. Sorry for bothering you with this.

One lesson we can learn from this is, that the console could output a log message (log level warning or error) when it encounters serial communication errors.

KevinOConnor, 2018-06-17:
Okay.  Sounds like we can close this then.  The console will report errors - but a retransmit isn't considered an "error".  Feel free to submit improvements to the console.py code.

closed 2018-06-17 by KevinOConnor