Klipper3d/klipper #391: BED_TILT_CALIBRATE always delivers x/y adjust 0.0 with BLTouch Sensor?

oderwat, 2018-06-11:
I recently added a BLTouch (clone) and added a bed_tilt section with 9 probing points. When I issue BED_TILT_CALIBRATE with a surface I tilted artificial I just get an x/y adjust of 0.0. klippy.log shows something around 3.2704931388989546e-08 for y which actually would be lower than 0.000000. Similar for X where the negative sign in front of the -0.000000 relates to the klippy.log value of -2.4975840774510105e-08. 

[image]

[image]

[klippy-bed_tilt_2018-06-11-15-57.log.zip](https://github.com/KevinOConnor/klipper/files/2090267/klippy-bed_tilt_2018-06-11-15-57.log.zip)

I also got a lot of timeouts while probing ("Failed to home probe: Timeout during endstop homing") sometimes on the first but also when it probed multiple previous points. What causes this? When I repeat the "BED_TILT_CALIBRATE" command after this I get "... NEXT already registered" errors. I need to do a "RESTART" in between to make it work.

KevinOConnor, 2018-06-11:
On Mon, Jun 11, 2018 at 07:32:37AM -0700, Hans Raaf wrote:
> I recently added a BLTouch (clone) and added a bed_tilt section with 9 probing points. When I issue BED_TILT_CALIBRATE with a surface I tilted artificial I just get an x/y adjust of 0.0. klippy.log [...]

That's odd.  It's showing that each probe point stopped at exactly
2.4mm (see the "orig: ...." in the log).  The e-08 stuff is just an
artifact of using coordinate descent to find the solution - it's
basically saying the bed is level.

Could you reproduce with the pristine code and the bltouch sequence
found in config/sample-bltouch.cfg and send that log?

> I also got a lot of timeouts while probing ("Failed to home probe: Timeout during endstop homing") sometimes on the first but also when it probed multiple previous points. What causes this? When I [...]

A "timeout during endstop homing" means the nozzle moved all the way
to the z position_min without the probe triggering.  It could mean
that the position_min needs to be adjusted, but more likely it means
the probe failed to trigger.

-Kevin

oderwat, 2018-06-12:
I actually did change to the exact code which is used in the example while I was trying to find out what cases the timeout (because my code does a lot of waits). The only change in my klippy is the axes param, which should not change anything. I will try to get "as pristine" as possible though. 

But I expect the BLTouch with 60° endstop mode may not really work very well. If the BLTouch is used normally, you see the pin "snap back" as soon it is touching the bed. As if you put a hand on a hot surface. With the endstop mode it actually stays deployed the whole time and triggers a while after it touches the bed. I think in this mode, it snaps back when the magnetic force pulls it back instead of when the hall sensor registers a movement at the tip.

About the timeout: Isn't it normal, that the nozzle can move below z_min before the BLTouch/Sensor triggers? if you home at the middle of a tilted bed you will be higher than with a flat bed and this defines "0". So if you go near the edge which is lower than the middle, the probe would need to go down further than z_min until it triggers, because the bed is tilted down and hence the distance is greater.

I am pretty sure the probe does not fail to trigger.. it may only not "yet" have triggered?

oderwat, 2018-06-12:
I created you two  videos which show how different the BLTouch unit reacts in 10° and 60° mode. I am not sure if that is related but at least you get an impression of what I talk about :)

[bltouch-mode-videos.zip](https://github.com/KevinOConnor/klipper/files/2092122/bltouch-mode-videos.zip)

KevinOConnor, 2018-06-12:
On Mon, Jun 11, 2018 at 05:10:12PM -0700, Hans Raaf wrote:
> I actually did change to the exact code which is used in the example while I was trying to find out what cases the timeout (because my code does a lot of waits). The only change in my klippy is the [...]

Okay.

> About the timeout: Isn't it normal, that the nozzle can move below z_min before the BLTouch/Sensor triggers? if you home at the middle of a tilted bed you will be higher than with a flat bed and [...]

Yes, specifying a negative position_min is common when using bed_tilt.

> I am pretty sure the probe does not fail to trigger.. it may only not "yet" have triggered?

Since you have a 2.4mm probe distance, the bed would have to have a
severe tilt for that to occur.  Generally with a z probe distance that
high, a negative postition_min isn't necessary.

-Kevin

KevinOConnor, 2018-06-12:
On Mon, Jun 11, 2018 at 05:22:08PM -0700, Hans Raaf wrote:
> I created you two  videos which show how different the BLTouch unit reacts in 10° and 60° mode. I am not sure if that is related but at least you get an impression of what I talk about :)

It maybe necessary to create a custom bltouch "extras" module.

As a quick test, you could modify probe.py:PrinterEndstopWrapper to
define home_start() and enforce a maximum "rest_time" of 1ms.  Then it
should be possible to use angle=10 mode?

-Kevin

oderwat, 2018-06-12:
OK. I did not expect to need to set the "global" z_min to a negative value. I would have expected that something like that only is being used by probing and would suggest to add such a parameter for probing.

By using -5 I didn't get timeouts anymore and could try the setup from the picture included. It also give 0/0 correction values. So there is for sure something going wrong here! I am very curious about that but sadly have no time to dig deeper till later today.

[image]

oderwat, 2018-06-12:
OK. I added this to the probe endstop wrapper:

```
    def __init__(...)
       ...
       #self.home_start = self.mcu_endstop.home_start
       ...

    def home_start(self, print_time, sample_time, sample_count, rest_time):
        return self.mcu_endstop.home_start(print_time, 0.001, sample_count, rest_time)
```

I also changed my probe/homing sections to this:

```
[probe]
pin: ^PC4
z_offset: 2.55
activate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=10
deactivate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=90
    G1 Z5 F600

[homing_override]
axes: z
set_position_z: 5
gcode:
    G90
    G1 Z10 F600
    G28 X0 Y0
    G1 X150 Y115 F7200
    G28 Z0
    G1 X0 Y0 F7200
```

This basically works for homing and probing but still has a problem. The pin goes down for probing. Then jumps up on trigger. But then it comes down again before it is pulled up. As you may saw in my video it will "naturally" come down again a short time after triggering in 10° mode. My probe deactivate code should (servo 90° ) told it to go up / stay up. So there seems to be some delay between it triggering and calling the `home_finalize` or executing the servo command. Enough time that the pin briefly hits the bed a second time before it is pulled up. On Marlin it stays up after triggering.

It also did not change anything about BED_TILT_CALIBRATE reports a perfectly flat bed even if it is for sure not!

oderwat, 2018-06-12:
I think it would be helpful if the "PROBE" command would report the z position it triggered at, if the z-axis motors are enabled and the heads z position is known. I tried to implement this but I think I only can catch the position after probing which is being reset to the same spot it seems.

KevinOConnor, 2018-06-12:
On Tue, Jun 12, 2018 at 08:56:32AM -0700, Hans Raaf wrote:
> I also changed my probe/homing sections to this:
> 
> ```
> [probe]
> pin: ^PC4
> z_offset: 2.55
> activate_gcode:
>     SET_SERVO SERVO=bltouch ANGLE=10
> deactivate_gcode:
>     SET_SERVO SERVO=bltouch ANGLE=90
>     G1 Z5 F600

Don't place movement commands in the activate/deactivate_gcode fields
- it will definitely mess things up.

-Kevin

oderwat, 2018-06-12:
@KevinOConnor without that it will always end up in an error state. It triggers, goes up, comes down again but not to the full extend (because z did not move up, what my code prevented) and tries to do 90° which will fail with an error, because the device thinks it was blocked in the step before.

oderwat, 2018-06-12:
When I remove the movement it also does not influences what happens at all. The delay before 90° is executed seems to happen before the movement is being executed.

oderwat, 2018-06-12:
OK. I got it kinda work using it like this:

```
[probe]
pin: ^PC4
z_offset: 2.55
activate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=10

deactivate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=90

[homing_override]
axes: z
set_position_z: 5
gcode:
    SET_SERVO SERVO=bltouch ANGLE=160
    G4 P200
    G90
    G1 Z10 F600
    G28 X0 Y0
    G1 X150 Y115 F7200
    G28 Z0
    G1 Z5
    SET_SERVO SERVO=bltouch ANGLE=90
    G1 X0 Y0 F7200
```

This way I also get results for `BED_TILT_CALIBRATE`!

It seems that the "deactivate_gcode" is applied after the probe moved up again when probing. I think it would need to be executed as soon as the triggering was detected. Can you hint me what to change to try that?

oderwat, 2018-06-12:
Do I understand it right that the whole homing is executed as an atomic operation (command queue) so that the BLTouch Servo Pin retraction (90°) would need to be integrated into this commands? Or even better both, the extraction and retraction should be embedded into this command queue?

oderwat, 2018-06-12:
After playing around with klippy I think that it is not possible to do the servo 90° quickly enough after the endstop triggered from klippy. I may need to be done directly on the MCU without any round trip delays. I don't know what causes so much delay but you wrote something about 100ms delays between sending the servo commands.

KevinOConnor, 2018-06-12:
On Tue, Jun 12, 2018 at 10:27:17AM -0700, Hans Raaf wrote:
> OK. I got it kinda work using it like this:
> 
> ```
> [probe]
> pin: ^PC4
> z_offset: 2.55
> activate_gcode:
>     SET_SERVO SERVO=bltouch ANGLE=10
> 
> deactivate_gcode:
>     SET_SERVO SERVO=bltouch ANGLE=90

There should be a "G4 P100" after both SET_SERVO statements.  (The
sample-bltouch.cfg is missing a delay in deactivate_gcode as well.)
The servo doesn't move instantly.  The delay is needed to give the
servo time to take the action prior to moving the head.

On Tue, Jun 12, 2018 at 11:26:56AM -0700, Hans Raaf wrote:
> I made you another movie of what happens when using the above config. I am not sure if I did descibe it right.

Okay, thanks.

On Tue, Jun 12, 2018 at 02:10:44PM -0700, Hans Raaf wrote:
> After playing around with klippy I think that it is not possible to do the servo 90° quickly enough after the endstop triggered from klippy. I may need to be done directly on the MCU without any [...]

It's not clear to me what you are reporting.  Is a well timed ANGLE=90
needed to ensure correct operation, is it just annoying for the pin to
bounce a second time, or does the pin not retract at all if it is not
timed correctly?

-Kevin

KevinOConnor, 2018-06-12:
On Tue, Jun 12, 2018 at 03:27:02PM -0700, Hans Raaf wrote:
> But this actually cases the problem! When it is automatically deployed after the initial triggering, the head may not have moved up far enough up that the pin can go all the way out again (as it moves some mm before it triggers). If this happens and another command is send, it goes into the error state (also in the original diagram).

I don't understand.  After the bltouch triggers, the head should
immiedietly stop - no further movement at all.  The next command
should be an ANGLE=90 - no other command to conflict with it.  Only
after ANGLE=90 is sent, and given time to take effect, should the head
start moving.

The "G4 P100" commands are needed in the activate/deactivate_gcode to
make the above happen.  It's definitely wrong to not have them.

So, my question is, if the ANGLE=90 takes several hundred milliseconds
to complete, and as a result the "continuous trigger testing" tests
the bed multiple times, what goes wrong?  Is it visually annoying,
does it causes the bltouch to fail, or does the bltouch not retract at
all?

-Kevin

oderwat, 2018-06-12:
Sure the head stops. But before the 90° gets executed there is a to long delay and the BLTouch already deployed the pin **itself** again and then the 90° may fail because the pin could not deployed fully (which is why it triggers the alarm, being blocked).

The `G4 100` does not change anything. And a `G4 2000` neither.

I already described what goes wrong: The pin gets deployed by the BLTouch while the head is at a position where the pin can't be fully deployed again. This then causes the alarm. I made another video. It shows what happens when you issue a 10° .. move the head down till it triggers and then do nothing or even a 90° without moving the head up in between, so that the pin can fully extend.

[bltouch-10-mode-error-trigger.zip](https://github.com/KevinOConnor/klipper/files/2096097/bltouch-10-mode-error-trigger.zip)

oderwat, 2018-06-13:
Today I flashed to Marlin and tried what happens there. With Marlin the pin gets dragged up "most of the time" but also not always. While I had Marlin installed I also run a repeating probe accuracy test and this showed a devastating result for this "factory new" Smart BLTouch clone. I had it trigger in a range of >0.5 mm, which is absolutely horrible.

So I went back to Klipper and added the probing z-output (PR #396) and wrote a little gcode macro to ease repeated probing with Klipper. As already expected it also had completely horrible results!

This lead me to ditch this new BLTouch clone and replace it with one older version which I bought last year. This one does *not* expose the "pin down" problem. It just works as expected. It also has a much more desirable repeatability with a range of < 0.06mm for the triggers.

I am not sure how I should feel about it. This "nicely build" new smart BLTouch clone was more expensive than the older ones and it is either defective or just total crap. It seems as if that device itself introduces the delay / malfunction. 

The device I use now does even faster put the pin back down in 10° but it actually reacts to the 90° command as expected. Everything seems to run smoothly with it now.

I am very sorry for wasting time... I hope that at least the small code contributions I made are helpful.

oderwat, 2018-06-13:
Lol ... It's not true. It still does the extra pin down when working in the 10° mode.

KevinOConnor, 2018-06-14:
On Tue, Jun 12, 2018 at 04:15:02PM -0700, Hans Raaf wrote:
> Sure the head stops. But before the 90° gets executed there is a to long delay and the BLTouch already deployed the pin **itself** again and then the 90° may fail because the pin could not deployed fully (which is why it triggers the alarm, being blocked).

I do understand this.  What I do not understand is what specifically
happens when it "fails".

The reason I ask is that it is relatively simple to update the code to
make a servo change within a few milliseconds of a signal with 99.999%
confidence.  If a late 90° command results in a soft failure (eg, it's
just visibly annoying, the alarm can be cleared, etc.) then this would
be the preferred approach.  However, if a late 90° command, places the
bltouch in a non recoverable state, then a different (more complex)
development approach would be needed.

-Kevin

oderwat, 2018-06-14:
It can be worked around by extra reseting the device and doing carefully designed moves "around" it's usage. It also can be worked around by using the "endstop test" 60° mode.

Basically with a working BLTouch you can always recover (alarm state lets it say it is 'triggered') by making sure the probe is high enough and issuing a 160° with a follow up 90° (after which, if it worked, is 'not triggering' anymore).

The problem I see is that you can't really use the alarm state this way. This translates into: Currently you can move the head when the BLTouch is in alarm (the endstop is triggered). This is inherently unsafe, because there is some reason why it is in the alarm state.

The better way would be avoiding the alarm all together and not relying on extra movements for "freeing" the probe when it drops down after "10° / trigger" because of the to late 90°. With the "quick" 90° the probe stays up from the beginning and there is no "cleaning" up needed.

Having it "clean" implemented , allows the alarm to be used. You can "safely" assume, that a 90° which results in a (still) triggered state is an alarm. 

One could add a recover routine which does automatically sends a 160° and repeat the 90° / trigger test. 

If it still is triggered, it means the probe is blocked, not working properly (not adjusted) or even defective. Basically anytime the end stop is triggered outside the "10 ms" signal while being in 10° state will then be an error, which should stop Klipper from sending moves to the printer. 

I may also (finish) implementing it, when I get some head-start on what to do, as you don't have a probe and also other important stuff to work on. I may be able to do it alone, but I need to look more into the debugging stuff.

What I actually want to test is "hacking the 90°" into the MCU code, such that I can see if it really works if it is set as soon as the probe triggers. I also wonder if the endstops are handled by interrupts or polling. The code does not look very complex but I don't understand (yet) how it works.

KevinOConnor, 2018-06-19:
It looks like the main issue raised here (bed_tiilt_calibrate reports zero tilt) was addressed.  Lets track the secondary concern (bltouch probe sequence not correct) in issue #387.

closed 2018-06-19 by KevinOConnor

oderwat, 2018-06-22:
@hg42 From my tests and past experience It retracts in Marlin every time. With my worst clone it is still about 8 out of 10 times. But why not continue discussing in #387?

hg42, 2018-06-22:
all the information I refer to is here...it seemed unnatural to post the comment in the other issue.

does Marlin retract faster than probing?

oderwat, 2018-06-22:
When the BLTouch triggered, it pulls the pin up. It just needs to stay there. It has nothing to do with how fast the head is moving. Do you have a BLTouch? Did you watch my videos, which I posted here?

P.S.: This issue indeed is solved, closed and the conclusions are in #387.

hg42, 2018-06-23:
hmm, I think I read all your posts and saw all your videos...

> It just needs to stay there

that's why I asked for a "single trigger mode" (in contrast to the continuous mode 10°).
So 60° mode is the solution? I am asking, because I think you started with the 60° mode and I don't understand why you switched to 10° mode.

> It is designed for having the head moving up directly after triggering
...
> then one could do all the probing (when moving the head up quickly after the probe triggered)
...
> It currently kinda "works" because the head is moving up

My reception of all that was, that the pin now goes down again, **while** the head is already moving up but not retracted enough, despite your 3rd video showing the pin being pushed down before the head moves up...

And no, I don't have a BLTouch. Does it mean, I am not allowed to be interested in it and how it works?

So I shut up now...