Klipper3d/klipper #392: Random error "MCU 'mcu'shutdown: Move queue empty"

ich777, 2018-06-12:
Hello,

since the last few days i got a strange error randomly, anyone experienced this?
Print speed is 80mm/s and non printing moves 200mm/s

In the Octoprint control windows it says:
"MCU 'mcu' shutdown: Move queue empty Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software. Printer is shutdown "
and:
"MCU 'mcu' shutdown: Move queue empty "

Attached also the klippy.log

Regards, Christoph

[klippy.log](https://github.com/KevinOConnor/klipper/files/2092898/klippy.log)

KevinOConnor, 2018-06-12:
This is a result of the host losing clock synchronization with the micro-controller.  It can be seen by running graphstats.py -f :
[image]

The spike on the right side of the graph causes the Klipper software to malfunction.

Some notes:
- I noticed you are using an x86 machine for the host.  That's not a well tested path.  You should make sure that the ntp configuration is stable (or not running).
- If this is a recurring issue, you could try making the CLOCK_MONOTONIC_RAW change described in issue #175 to see if that helps

-Kevin

ich777, 2018-06-12:
Thank you, as always, for the quick answer.

I will give CLOCK_MONOTONIC_RAW a chance and let NTP turned on, for what i've found:

```
CLOCK_MONOTONIC never experiences discontinuities due to NTP time adjustments, but it does change in frequency as NTP learns what error exists between the local oscillator and the upstream servers.

CLOCK_MONOTONIC_RAW is simply the local oscillator, not disciplined by NTP. This could be very useful if you want to implement some other time synchronization algorithm against a clock which is not fighting you due to NTP. While ntpd (the reference implementation of NTP protocol and the most widespread NTP daemon) is reputed to be "gentle" with time adjustments, it's more accurate to say it's gentle with the absolute time. It's willing to slew the clock by 500ppm which is pretty dramatic if you're in a position to measure your clock frequency against some other standard.
```
Source (2nd answer): https://stackoverflow.com/questions/14270300/what-is-the-difference-between-clock-monotonic-clock-monotonic-raw

ich777, 2018-06-13:
@KevinOConnor thank you for the hint with CLOCK_MONOTONIC_RAW, it solves the problem so far, printed already one print with 8,5 hours and one now with 4 hours and got no problem.
NTP is still enabled.

Regards, Christoph

KevinOConnor, 2018-06-15:
Okay, thanks.  I think the Klipper software could have done a better job here.  I'll have to give this some further thought.

-Kevin

ich777, 2018-06-15:
@KevinOConnor One last question bevore i close the issue...
If i update klipper (like in the FAQ described) do i have to change CLOCK_MONOTONIC every time to CLOCK_MONOTONIC_RAW?

KevinOConnor, 2018-06-17:
Sometimes git will auto-merge, but it may require manual work.  In any case, I think this should be fixed in Klipper - so lets leave this open until a fix is ready.

KevinOConnor, 2018-07-09:
I have changed the main code to use the CLOCK_MONOTONIC_RAW clock (commit e1833e02).  This should fix this issue.

-Kevin

closed 2018-07-17 by KevinOConnor