Klipper3d/klipper #393: Resume after Pause

ich777, 2018-06-12:
Hello,

i've tried the pause function in Octoprint with Klipper, one thing that i've noticed is if you are not quick enough to change the filament and the motors turn off you can't resume the print (set the motor deactivation time to 600 as workaround).
But in my configuration that's not always possible to do it in that time for example when i am not at home and the filament sensor triggers the pause and i get home after for let's say for about an hour or two, i also can't do a new home command because i use a probe and the homing_override (so that the probe is in the middle of the bed).
Is there another workaround for this so that the steppers stay always on after the G0 gcode is sent or the steppers turn off and the printer remembers the position and let me continue my print?

Regards, Christoph

oderwat, 2018-06-12:
Isn't this controlled by the `motor_off_time` in the printer section of the configuration. It is 600 seconds (=10 minutes) by default.

See: https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L296

ich777, 2018-06-13:
@oderwat Yes, i write that above, but should i set my stepper off time to for example 10000? Is there no better way so that Klipper detects the pause state of Octoprint and set the timeout to unlimited so that i can set the motor_off_time to 120?

Regards, Christoph

oderwat, 2018-06-13:
I don't think that a pause can be distinguished from simply not getting anymore data. This would be different if either OctoPrint would use something like G4 Pxxxx for pausing. But that would need to be refreshed every X seconds and needs to be implemented in OctoPrint or if Klipper would support the `M0 show text and wait for button` command as Marlin does for example.

ich777, 2018-06-13:
Why would that have to be checked every X minute? Sorry but i think you don't understand what i mean, if the filament sensor detects no filament or you simply press Pause in the main screen of Octoprint changes the status that can be read by Klipper and if you click on Continue the status also changes btw. this works like you would pause the print in Marlin on the printer with the only difference that you do it on the Octoprint webpage.

Look here for more info about the status: [Click](http://docs.octoprint.org/en/master/events/index.html#printing)

oderwat, 2018-06-13:
Why do you think that Klipper can read the status of OctoPrint? Klipper has no knowledge of OctoPrint. What I suggested is that OctoPrint does send Pause commands such that Klipper knows that the host is still alive.

ich777, 2018-06-13:
@oderwat Since i'm not a programmer that answer helps me not in any. Not everybody can read the code perfectly, i understand most of it but not everything. ;)

I've asked if there is a way or a plan in the future to set the motor_off_time to unlimited if a pause is triggered and that was my idea to do it with the statuses that where reported from the Octoprint API.

closed 2018-06-24 by ich777