Klipper3d/klipper #395: Different acceleration on print

Gi0cas, 2018-06-13:
Hi, in Marlin i used a lot the command M204, so i can specify different acceleration for external perimeter, retraction, bridge, etc... I use SLIC3R and there is this opportunity to set different acceleration values in the settings.
When i print with Klipper, in the terminal i see that Klipper not recognize M204 command...and i want to know if it is something that can be implemented or the developer is not interested about this function.
I think it's very useful, because you can have strong value on infill or internal perimeter and sweet and calm values on external perimeter for example...what do you think?

oderwat, 2018-06-13:
I think you want to read this:  https://github.com/KevinOConnor/klipper/issues/282

Gi0cas, 2018-06-13:
Oh no, i searched but involontary skipped this issue... can you please explain better the mean of this commands? 
SET_VELOCITY_LIMIT [VELOCITY=<value>] [ACCEL=<value>] [**ACCEL_TO_DECEL**=<value>] [**JUNCTION_DEVIATION**=<value>]
I think that ACCEL is the acceleration i want...
- what is "ACCEL_TO_DECEL" ?
- what is "JUNCTION_DEVIATION" ?

Gi0cas, 2018-06-13:
Thanks a lot. Now i'm writing a script for SLIC3R to insert the command for the various type of acceleration. Thanks again!

closed 2018-06-13 by Gi0cas