Klipper3d/klipper #397: Introduce the concepts of FDM into Klipper

megablue, 2018-06-14:
As of right now, most FDM firmwares have no idea what they are doing, they do not have the concept of FDM except for the basic of flow control while moving the axes.

I am purposing to introduce some FDM concepts into Klipper so that more parameters can be adjusted on the fly.

Introduce commands to let Klipper know what it is doing, says
- something like SET_FDM_STATUS_PRINT_START and SET_FDM_STATUS_PRINT_END, so that Klipper will know a print is started/ended instead of just gcode blindly.
- status commands like _PERIMETER, _INNER_WALL, _INFILL, _ZHOP, _FIRST_LAYER, _VASE_MODE, _TRAVEL
- allows acceleration, deceleration, junction deviation, max speed, min speed, retraction, temp, fan speed to be tight to specific STATUS and overwrites the slicer gcode when specified and possible. 
- on top of that negative G0/G1 E(xtrude) command should be parsed and overwritten when a print is started.
- positive and negative extrude moves could have a different acceleration/deceleration (if pressure advanced isn't enabled)
- Also please a different acceleration/deceleration/junction deviation/max-min speed for non-extrude moves
- The parse could also say, repeat the entire perimeters/infill/walls/layers since it knows what it is doing and it can repeat the said STATUS when a user wants it to.

It is all very vagues i know... but i hope someone would bring in better/solid ideas.

KevinOConnor, 2018-06-19:
Thanks.  I don't think that github issues are the best place to propose and/or track high level items like this though.  My fear is that these types of issues stay open indefinitely, because there is always more to be done.  The mailing list and/or the irc channel may be alternative places to discuss these types of items.

-Kevin

drphil3d, 2019-03-24:
thats an excellent idea