Klipper3d/klipper #399: Feature Request: Allow aliasing of Extended GCode Commands & Temperature based switches on other than fans

AxMod3DPrint, 2018-06-15:
Our Printers use M80/M81 for power on/power off, with Klipper this is achieved by setting pin ar12 on an arduino as ps_on via SET_PIN PIN=ps_on VALUE=0/1. This is currently macroed via the custom command plugin in Octoprint. We also work via the console for speed, and typing that line is cumbersome, being able to alias that out to M80/M81 would be handy, and I'm sure there would be other use cases for this.

Which brings me on to the 2nd request. Temperature based switching for things other than fans. Our Smoothie based printers arm a temperature switch on print completion via M1100 S1 which then calls the printer to switch off via M81 when the hot end gets to 50c. Again, I'm sure there are other use cases here that a temperature based switch may come in handy apart from Fans

These are just ideas to bat around, and discuss, so I'm not expecting anything.

oderwat, 2018-06-15:
Can't this be done with a gcode macro for `m80` oder `poff`?

```
[gcode_macro m80]
gcode:
  SET_PIN PIN=ps_on VALUE=0
```

Switching off the printer may be something a OctoPrint plugin can do. Maybe a klipper command to run some gcode at a specific time could be used.

cmock, 2018-06-15:
Switching off only after the hotend has cooled down can be done with the Octoprint "PSU Control" plugin, I'm using that with klipper and "SET_PIN" successfully.

AxMod3DPrint, 2018-06-15:
@oderwat - Still learning, as such I'd missed that one, which'll do what we need. 

@cmock  - thanks, last time I'd looked at that plugin was a while back and it could only be done via the RasPi itself over the PS_ON pins. 

So this really closes it for our needs. However, I'll leave this open should any other person think they could use something like this.

KevinOConnor, 2018-06-30:
Sounds like a solution was found.

closed 2018-06-30 by KevinOConnor