Klipper3d/klipper #401: x end stop problem

chizhevskiy, 2018-06-16:


chizhevskiy, 2018-06-16:
I have tried to configure x-endstop on my Tevo Black Widow 2130 customized printer. x end stop is definitely recognized by klipper - if I end stop is closed x is moving right during homing, otherwise x is moving left during homing. I have tried to change direction, position min max everything , but cannot get the end stop work properly. Almost giving up to move back to Marlin :-(

AxMod3DPrint, 2018-06-16:
@chizhevskiy - is your X axis moving in the wrong direction? IE Left is Right, Right is Left?

chizhevskiy, 2018-06-17:
Problem is that x carriage is moving beyond end stop.

mvturnho, 2018-06-17:
You can test your endstop with the code m119 if they are correctly wired or maybe reversed.

http://marlinfw.org/docs/gcode/M119.html

AxMod3DPrint, 2018-06-17:
@chizhevskiy - I would suggest doing what @mvturnho says, check if your endstop is registering correctly, IE, is it closed when it should be open or vice versa, if that is the case, you will need to invert the signal by putting an exclamation mark in front of the pin to reverse the output.

chizhevskiy, 2018-06-22:
few things were setup incorrectly: max current, step size, etc.

closed 2018-06-22 by chizhevskiy