Klipper3d/klipper #403: 'temperature_fan' gives TemperatureFan instance has no attribute 'max_power' when reaching 'target_temp'

mvturnho, 2018-06-17:
Hi,

I configured a temperature_fan for cooling the ramps board when reaching a certain temperature.
For testing I set target_temp to 30 degree. when this target temp is reached I get python errors in klippy.log.
The error is not completely clear to me, or the config is missing the max_power or it should not be there. 
So I added a max_power attribute (becouse it was still commented out) to the config 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2108852/klippy.log)

end retried , but the error remained.

```
Exception in serial callback
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 51, in _bg_thread
    hdl(params)
  File "/home/pi/klipper/klippy/mcu.py", line 410, in _handle_analog_in_state
    self._callback(last_read_time, last_value)
  File "/home/pi/klipper/klippy/extras/thermistor.py", line 74, in adc_callback
    self.temperature_callback(read_time + SAMPLE_COUNT * SAMPLE_TIME, temp)
  File "/home/pi/klipper/klippy/extras/temperature_fan.py", line 51, in temperature_callback
    self.control.temperature_callback(read_time, temp)
  File "/home/pi/klipper/klippy/extras/temperature_fan.py", line 73, in temperature_callback
    self.temperature_fan.set_speed(read_time, self.temperature_fan.max_power)
AttributeError: TemperatureFan instance has no attribute 'max_power'
```

KevinOConnor, 2018-06-17:
FYI, this code was contributed by @wizhippo.

mvturnho, 2018-06-17:
Also, when I use PID for control thera are no default pid values for pid_Kp, pid_Ki and pid_Kd and the startup for klippy fails.
I think we need to add these default values so there is no need to configure them.

wizhippo, 2018-06-18:
@mvturnho as `fan` controls the max power I think the mistake is it should be self.temperature_fan.max_speed not max_power.  It is fan's responsibility to limit max_power.  

Thoughts?

wizhippo, 2018-06-18:
Also line 71 I guess should be min_speed not 0.

wizhippo, 2018-06-18:
Also I think pid values should be set by user in config and not have defaults just like heater does.  The docs can show an example for users to go by but the values are very machine specific.

mvturnho, 2018-06-18:
I agree, when other modules are configured like that so should this one. And the max_power was already in the code if this should be max speed so be it.

wizhippo, 2018-06-18:
I have submitted https://github.com/KevinOConnor/klipper/pull/407 This should fix the issues.

KevinOConnor, 2018-06-27:
#407 was pulled - commit 9519e053.

closed 2018-06-27 by KevinOConnor