Klipper3d/klipper #405: LPC1768 - Re-Arm - Adding Display section causes error

AxMod3DPrint, 2018-06-18:
Re-Arm - LPC1768 designed to replace the Arduino Mega with the same footprint and run smoothieware, untested as yet, but will report back once tested. However, after adding in a section for a st7920 display in the config, the error Unknown command: st7920_send_cmds is thrown. I've tried to dig around to see if I could fix it but to no avail.

Log attached:
[klippy-harlequin.log](https://github.com/KevinOConnor/klipper/files/2109903/klippy-harlequin.log)

KevinOConnor, 2018-06-18:
The lcd support wasn't enabled on the lpc176x code.  I enabled it (commit 4fbecfa1).  Note, though, that it hasn't been tested so I'm unsure if it will work.

In order for the above change to take effect, be sure to pull down the latest code and then do a full "make menuconfig ; make ; make flash" sequence to apply the change.

AxMod3DPrint, 2018-06-18:
Figured that would be the case, just couldn't see how to enable it myself, will check it later.

AxMod3DPrint, 2018-06-19:
OK, tested, not everything is showing, the temperatures and status messages work and there's a flashing cursor on the screen, see image. No errors are being shown in the log, attached.

[klippy-harlequin.log](https://github.com/KevinOConnor/klipper/files/2114182/klippy-harlequin.log)
[image]

KevinOConnor, 2018-06-19:
On Tue, Jun 19, 2018 at 12:03:53AM -0700, AxMod 3D Print wrote:
> OK, tested, not everything is showing, the temperatures and status messages work and there's a flashing cursor on the screen, see image. No errors are being shown in the log, attached.

If you incrase ST7920_DELAY in klippy/extras/display.py to .000040
does the problem go away?

-Kevin

AxMod3DPrint, 2018-06-20:
@KevinOConnor - nope it gets worse and increasing the the delay further actually makes it worse making the display totally garbled. However decreasing it to .000010 it seems to get better, but the cursor is still showing, none of the icons and it can get slightly garbled when heating and will often go blank.  

See video link.. https://youtu.be/EAp2WMiWHhA

KevinOConnor, 2018-06-21:
It definitely looks like a timing issue.  All of the symptoms are consistent with the initialization sequence not being fully received.  Try increasing the ST7920_DELAY, and make sure to run "sudo service klipper restart" after each change to the code.

I've brought my st7920 up on my lpc1768 dev board.  I did need to modify ST7920_DELAY to .000040 to make it work (not sure why, as it works fine under the due - I think the voltage from the board is low, as the screen is also quite dim).

-Kevin

AxMod3DPrint, 2018-06-21:
Ah, I was just restarting from the console and not issuing a full restart.
I'll give it a shot later

As for the dim display, you're correct about the low voltage, the LPC1768
uses a 3.3v logic level over a 5v. The re-arm uses a cable that splits the
5v power line from the ribbon cable on to a 5v shifted pin.

AxMod3DPrint, 2018-06-21:
@KevinOConnor that worked. So, just a minor adjustment to ST7920_DELAY and the display works fine. 

This has been added to pull request #406

KevinOConnor, 2018-06-21:
Right, but I wired the display up to 5V power.  (As far as I know, the sid, sclk, and cs are output only from the mcu so they are okay to drive at 3.3v while the lcd is powered at 5v.)  So, not sure.  Maybe there are some pullups on the lines that are messing with the voltages.

In any case, I changed the code so that the delay can be specified in the config file (commit d94c8561).

KevinOConnor, 2018-06-23:
After looking at this a little closer, I don't think the issue is with the main st7920 delay.  Instead, I think there is a separate delay needed when changing command modes.  Can you retest with the work-st7920-20180622 branch?

cd ~/klipper ; git fetch ; git checkout origin/work-st7920-20180622 ; make ; make flash ; sudo service klipper restart

-Kevin

AxMod3DPrint, 2018-06-26:
Noticed the above has been merged into the master branch at commit f08a0c5e93b92a6efa72d16a3d53968f3e54bf1c issue is resolved

closed 2018-06-26 by AxMod3DPrint