Klipper3d/klipper #409: Dual Z motors, single endstop

stefan85, 2018-06-20:
Yesterday I was trying klipper for the first time. At the moment I am using two steppers and drivers for the Z-axis. But this did not allow me to use a single endstop for both motors.

In another topic, someone said to use Z-Max pin in parallel for the second motor. This would work, but this doesn't seem to be a neat solution. 

Would it be possible to add a function to use dual Z with one endstop?

StefanMeGit, 2018-06-20:
U just have to delete the second endstop. ;)

Just declare the endstop of the main Z-axis

stefan85, 2018-06-21:
Will try that tonight.  I tried that before, but then I had some more issues in my config, so I changed multiple things and never tried it again. Thanks.

derdigge, 2018-06-26:
Maybe this is helpful. Same boat. This is my config for dual Z on MKS gen 1.4:

`[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^!ar18
position_endstop: 0
position_max: 400

[stepper_z1]
step_pin: ar36
dir_pin: !ar34
enable_pin: !ar30
step_distance: .0025`

closed 2018-06-30 by KevinOConnor

technoo10201, 2020-02-16:
> Maybe this is helpful. Same boat. This is my config for dual Z on MKS gen 1.4:
>
> ```
> [stepper_z]
> step_pin: ar46
> dir_pin: !ar48
> enable_pin: !ar62
> step_distance: .0025
> endstop_pin: ^!ar18 [...]

Thanks ! It helped me :)