Klipper3d/klipper #411: Delta, tower steps working but extruder stepper not working

micheleamerica, 2018-06-21:
I've installed Klipper on a custom built Delta printer using an Arduino Due with the RADDS board and although everything else is working as expected (including the LCD) I've been unable to make the extruder stepper motor work.
Previously I had MK4duo (a Marlin derivative) and I've used the pin definitions from my working MK4duo configuration to build the printer.cfg file. I've also confirmed that my MK4duo working configuration is in line with the generic-radds.cfg in what concerns the pins being used.
I've also confirmed that the gcode I'm sending to the printer is well formed and includes movements with extrusion.

Here is the part of the configuration related to the extruder:

    [extruder]
    step_pin: analog7
    dir_pin: analog6
    enable_pin: analog8
    step_distance: 0.01
    nozzle_diameter: 0.300
    filament_diameter: 1.750
    heater_pin: ar13
    sensor_type: ATC Semitec 104GT-2
    sensor_pin: analog0
    control: pid
    pid_Kp: 4.838310
    pid_Ki: 0.173971
    pid_Kd: 33.639560
    min_temp: 0
    max_temp: 250

If other information is required please let me know.

Any help in making this work would be appreciated, otherwise I'll go back to MK4duo and wait some time before trying Klipper again.

Thanks in advance.

KevinOConnor, 2018-06-21:
We really need the Klipper log file in order to assist.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Also, have you run through the extruder checks listed at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

If it still fails, issue an M112 after attempting to extrude and then attach the log as described in the link above.

-Kevin

micheleamerica, 2018-06-22:
Hi,
Thanks for your answer. In fact I was missing the ! on the extruder enable pin. Now the extruder is working.
On the other hand I runned into problems during printing. I'll open another issue with the details and log file.

closed 2018-06-22 by micheleamerica