Klipper3d/klipper #412: Pinout for STM32F1 platform

yschroeder, 2018-06-21:
Documentation on the used pinout for the STM31F1 platform is missing.

As I understand it, this platform was implemented for the `Blue Pill` board. While most pins can be freely assigned (GPIOs) there are some limitations for some pins:

1. UART (already figured out that `PA9` is TX and `A10` is RX)
2. SPI (which SPI is used and on what pins? doesn't look like SPI is available at all in `gpio.c`)
3. ADC
4. Which pins are available as GPIOs?

Documentaiton is needed which pins are configured for what. The STM32F1 has a lot of different options for these signals (in contrast to AVR platform).

KevinOConnor, 2018-06-21:
FYI - @grigorig contributed the STM32F103 code.

grigorig, 2018-06-26:
Good point, there's basically zero documentation for the backend right now. I'll put together some basic documentation.

PR #425 (2018-06-27): stm32f1: add port documentation
Fixes #412.
commit eb020439 (2018-06-27): stm32f1: add port documentation (#425)

Fixes #412.
closed 2018-06-27 by KevinOConnor