Klipper3d/klipper #413: delta_autocalibrate questions

outlookhazy, 2018-06-22:
I'm just now returning to my printer after a year or so and discovered a lot of the things I had been doing externally with my own software are now included in klipper (awesome! thank you for your work!)

This may be covered somewhere I was unable to find, but why aren't the _arm_n_ parameters included in the descent adj_params? Is it a processing power tradeoff, or is that perhaps compensated for with the other adjustments? It's been so long at this point, I don't remember the subtleties of inferring the geometry...

Also, it _looks_ like the delta_calibrate config section supports a manual points list (though undocumented in the example-delta.cfg) is that correct? I'm not a Python developer, so following the weakly-typed references around is foreign to me

KevinOConnor, 2018-06-22:
On Fri, Jun 22, 2018 at 04:15:43PM +0000, Charlie wrote:
> I'm just now returning to my printer after a year or so and discovered a lot of the things I had been doing externally with my own software are now included in klipper (awesome! thank you for your [...]

I found that if arm_a/b/c is adjusted then the parameter search
doesn't converge (or it converges to absurd values).

If we think of the positions on a delta as being determined by three
triangles and each triangle has a base, a height, and a hypotenuse,
then the configured radius/angles control the base, the endstop
positions control the heights, and the arm lengths control the
hypotenuses.  I don't think it would be possible to try and calibrate
all three parts of the triangles simulaneously from just Z height
measurements.  (There could be very many triangles of different sizes
that would result in the bed being level.)

What I've been hoping to do is put together a 9-point calibration test
that involves printing out a calibration object.  If the user can
measure a handful of dimensions on this printed object with a digital
calipers, then I think those could (along with z measurements) be used
to perform a much more detailed delta calibration.

> Also, it _looks_ like the delta_calibrate config section supports a manual points list (though undocumented in the example-delta.cfg) is that correct? I'm not a Python developer, so following the weakly-typed references around is foreign to me

I believe it would work, but I have not tested it.  It's not
documented because picking good delta calibration points isn't
necessarily easy - I figured it was simpler to stick with some
standard probing points.

-Kevin

outlookhazy, 2018-06-22:
I figured it was something like that, I'll have to look at the code on my end and see if the assumptions are similar.

The reason I ask about the points was that in my autocalibration implementation, I was doing a similar routine, except I was iterating it repeatedly (rerunning after applying adjustments) and wanted to avoid the potential of surface imperfection(s) at the points throwing off the math. To that end I had been applying randomized shifts to both the angle and the radius on each subsequent run of the routine  for each point (in addition to varying the number of points etc). This seems to be partly what your scatter values are doing, except those are fixed.

KevinOConnor, 2018-06-22:
On Fri, Jun 22, 2018 at 11:45:18AM -0700, Charlie wrote:
> The reason I ask about the points was that in my autocalibration implementation, I was doing a similar routine, except I was iterating it repeatedly (rerunning after applying adjustments) and wanted [...]

The reason I scatter the points is that I felt it would give a better
calibration of the radius parameter.  An incorrect radius would result
in a bowl or bell shaped curvature of prints - if we don't scatter the
distance between the center of the bed when probing, then potentially
we only get two points to measure radius from (you get the center of
the bed, and then all the rest of the points are an equal distance
from the center of the circle).

I'm sure there are other ways to do this though.

-Kevin

outlookhazy, 2018-06-22:
Yep, rod length was available but looks like I was skipping it too

closed 2018-06-22 by outlookhazy

KevinOConnor, 2018-07-13:
In another issue, @outlookhazy asked:
> I'm preparing to set up a touch sensor and noticed that the probe section doesn't have an x/y offset.
Does the auto-calibrate math you're using still work in such a scenario? My sensor is going to be hanging off the side of my delta effector.

My understanding of delta kinematics is that the probe XY offset should not be used in the calibration.  (The reason is, the triangles formed with the rods control the nozzle location - there's no expectation that the triangles would do anything meaningful at an offset from the nozzle.)  Therefore, in order for the probe to be useful, it should have a static Z offset regardless of where on the bed it is measuring.

This is actually a notable problem for delta printers probes with an XY offset - slight differences in arm length can result in a variable tilt to the effector which is magnified to a probe with an XY offset.  For example, the probe on my delta printer is basically useless - I use manual probing.

> maybe the correct way to handle it would be to add relative moves accounting for the offset into the probe activate/deactivate g-code? Would the routine still consider those probe results as from the "start" point?

No - never perform movement or coordinate modifications in the probe activate/deactivate gcode.  It isn't intended for that and will likely confuse the system.

-Kevin

outlookhazy, 2018-07-13:
@KevinOConnor
I'm approaching calibration with the assumption that the effector travels parallel to the bed. In that case the z-height of the probe is directly related to the nozzle height.
The calibration routine is matching the actual nozzle height against the expected height at the collection of points in the set to calculate parameters.
So if the nozzle probes at [0,0] with the probe offset by [10,5] then the result should be treated as a probe at [10,5] with the probe z-offset added in.
You're right that uneventful rods would skew this, but under different firmware I was able to get something like .001mm height deviation across the bed calibrating with a limit switch probe hanging off the side of the effector

outlookhazy, 2018-07-13:
Actually, I may need to revise that - know it's possible, but I'm on a plane at the moment and can't check exactly what the procedure/logic was.

outlookhazy, 2018-07-16:
@KevinOConnor I still haven't been able to parse through all my old delta calibration code, which was basically a translated version of the javascript on [this page](http://www.escher3d.com/pages/wizards/wizarddelta.php). But I was using repetier at the time, and moving the resulting parameters over to klipper (this was why I requested tower angle offset support back in the day).

While the x/y offset might make the results imperfect, my procedure was iterative and used a _bunch_ of points around the radius in (up to 3 - depending on count) concentric circles.

At first glance, i'm not seeing any specific references to the probe offset in the calibration itself. I'm thinking that the repetier probe routine does/did an offset to compensate for x/y offset for mesh leveling support. I haven't used that firmware in a year, so I don't remember 100%.  In my recollection, the head moved to the commanded point x/y/z+probe offset + probe height, then did a rapid by xoff, yoff, probed that location (a programmable number of times/speed), then did a rapid back to the commanded probe position. This would have been to conceptually "measure relative **bed** height at X/Y position" rather than "measure relative **toolhead** height at X/Y position". I don't think there was a separate probe routine to be used for geometry calibration, which would explain why I still remember that offset.

At any rate, I'm still conflicted regarding whether the X/Y offset would be irrelevant for delta calibration... On one hand, additional moves with faulty geometry wouldn't be accounted for, because you'd be treating the toolhead as being at a location where it isn't. On the other hand, treating the probe location as having the same z-height as the offset sensor willfully ignores any bed tilt/warp.

 I think my previous approach using the mechanical offset only worked because it was iterative. So while there was additional error introduced in the unaccounted-for movement, unless the starting parameters were WAY off, it tended towards correct after a couple runs.