Klipper3d/klipper #414: 2 rookie questions, pins and extruder

ejg3855, 2018-06-23:
I have to blowers for my extrusion area. They are tied together physically. I am trying to define the pin they use my config file says ar9. 

I have Google searched high and low for a ramps pin out guide and I can't find a thing that correlates what I see in klipper config files to a ramps pinout.

Is there a guide for this? Movement is all working mint. 

I am also trying to establish the step distance for my extruder,I have "#" out the min temperature so in theory I can G0 E10 and then meausre and back track to an accurate config number. Klipper is still telling me that my min extruder temp is to low and thus extruder won't move.

AxMod3DPrint, 2018-06-23:
Min temp if you just comment it out, it uses the default of 170, set it to 0 to cold extrude. As for the fan, ar9 is the header between the heatbead and the hot end, There is an example pinout for ramps in the config folder. 

pinout for RAMPS: https://bit.ly/2Kb037T

You can ignore the ar at the front of the pin, ar just signifies it's an arduino digital pin. If you have your original Marlin config, you can get the pins and steps from there.

ejg3855, 2018-06-23:
Still can't get fans to come on with m106.

BlackStump, 2018-06-23:
@ejg3855 
you need to supply the klippy.log for support issues as per
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md
If you are trying to get initial settings for a extruder, a simple option is disconnect the hotend and extrude to free air, this will give you ballpark figures to work from. 
The fan for a ramps board is already defined ar9 the physical location should be easy to find with a google search as there is a plenty of info on the net for ramps
From memory Ramps  might be silk printed as extruder 2/fan

ejg3855, 2018-06-28:
My struggle is reading the klipper pins and then reading the ramps pins. how do I know that ar9 = d9 that is illogical. Or if I am looking at the other analog/servo pins on ramps and then trying to define them in klipper.

hg42, 2018-06-28:
Arduino uses a mapping from arduino io numbers to processor io numbers.
On top of the processor board sits a board like ramps that has a second mapping.
Note, there are other boards that use arduinos and may have a different mapping.

**ar**X is **ar**duino io numbering.

To map these to pins, you need a mapping for your processor board + io board combination, e.g. mega2560+ramps.

You csn often find them in Marlin config files for your combinstion.
In your case there is a table in
https://reprap.org/wiki/RAMPS_1.4#Firmware_and_Pin_Assignments

AxMod3DPrint, 2018-06-29:
@ejg3855 - you should be able to comment and close this

ejg3855, 2018-06-29:
do when I am trying to hunt down a pin, I have to take the pin in klipper look at where ti goes into the mega and then transpose that to its corresponding pin on the ramps?

hg42, 2018-06-30:
this may depend on the pin.
E.g. if you connect the endstop y min, the line
ˋ#define Y_MIN_PIN 14ˋ
tells it's arduino pin 14, so klipper needs ar14

ejg3855, 2018-06-30:
So as an example I'm trying hi Jack the step and direction signal for x/y. It says pin 54/56 for x.

I'm trying to find where that goes to the driver so I can just use a Dupont connector rather than solder. Or can I just leave out the driver and steal the signal  that feeds the driver? Going to use servo motors.

AxMod3DPrint, 2018-06-30:
@ejg3855 just use a male Dupont cable directly into the socket.

hg42, 2018-06-30:
the reprap page I linked above says:
```
#define X_STEP_PIN 54
#define X_DIR_PIN 55
```

If you can reuse a pin for a different purpose depends on the circuit present on the interface board (reg. ramps). In this case step and dir are only digital outputs without additional circuitry, because the driver needs a simple digital signal. But be aware that certain pins have additional parts attached, like pullups for endstops etc..

ejg3855, 2018-06-30:
The motors just need step and direction so from what you're saying my thoughts to just pull the signal as it enters the driver chip should be ok

hg42, 2018-06-30:
I never used servos, but I think they need a single simple digital signal (pwm)?
You should also look for the voltages, 5V vs. 3.3V signals on inputs and outputs.

hg42, 2018-06-30:
sorry, you are using an arduino, which should use 5V...so probably ok

ejg3855, 2018-06-30:
You can configure tecnik motors to take step and direction signals. Makes them easy replacement for steppers. Side benefit is they don't skip steps or lose position and they are silent.

KevinOConnor, 2018-07-07:
It seems you found the necessary pins, so I'm going to close this issue.

closed 2018-07-07 by KevinOConnor