Klipper3d/klipper #418: Detailed debug information for TMC drivers? M122

StefanMeGit, 2018-06-23:
Is there any change for detailed debug info about connected TMC drivers? I have an special setup (Arduino DUE / RAMPS1.5 with some little modifications) and everything is working great, except of the TMC2130 drivers. They are working just fine on the MKS GEN1.4... 

In marlin u can use the M122 for an quick overview about the driver status: http://marlinfw.org/docs/gcode/M122.html

StefanMeGit, 2018-07-01:
Can be closed... DUMP_TMC is now available for debugging. 

Thx! I am now one of your Patreons!

nicozorza, 2018-07-02:
I'm wondering if there is a way to set the sensitivity of the sensorless homing during runtime. I created my own command to set this value, but it seems that my command has no effect. 
My command is something like this:
    ```
def cmd_SET_SGT(self, params):
        gcode = self.printer.lookup_object('gcode')
        sgt = gcode.get_int('VALUE', params, default=0, minval=-64, maxval=63) & 0x7f
        self.add_config_cmd(REG_COOLCONF, sgt << 16)
```
I guess that add_config_cmd() doesn't work on runtime, but I didn't find another way. Did anyone try this?

KevinOConnor, 2018-07-06:
On Mon, Jul 02, 2018 at 04:32:31PM -0700, nicozorza wrote:
> I'm wondering if there is a way to set the sensitivity of the sensorless homing during runtime. I created my own command to set this value, but it seems that my command has no effect.
> My command is [...]

The add_config_cmd() is not valid at run-time.  Use the set_register()
method instead.

-Kevin

nicozorza, 2018-07-06:
That worked, thanks!!

closed 2018-07-07 by KevinOConnor