Klipper3d/klipper #419: Long Print: MCU 'mcu' shutdown: Move queue empty

lmgreco, 2018-06-24:
Hello,

I was printing a large print with over 24 hours print time already executed when I received this error "MCU 'mcu' shutdown: Move queue empty"

I am not a coder, so I am submitting this error log in hopes it helps find a bug with my printer or with the firmware. Also, I do not want to start another long print if this is something I need to fix!

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2131091/klippy.zip)

KevinOConnor, 2018-06-25:
This appears to be the same as issue #392.  This is what your frequence graph looks like:
[image]
During the failure, the host software could no longer make predictions about the micro-controller clock and that eventually led to an error.  The Klipper software should have done better - this is a bug.  There is a work around described in issue #392, but a full fix still needs to be implemented.

-Kevin

KevinOConnor, 2018-06-30:
Okay, I'm going to close this in favor of the existing issue #392.

closed 2018-06-30 by KevinOConnor