Klipper3d/klipper #420: Move exceeds maximum extrusion (shared heater with multiple mcu's)

ich777, 2018-06-24:
Hello agian,

my printer cancel's the print with the error:
"Move exceeds maximum extrusion (731188792651064448.000mm^2 vs 0.640mm^2)"

But i've printed the same exact gcode one time sucessfully. Can somebody explain me why this is happening?

[klippy.log](https://github.com/KevinOConnor/klipper/files/2131099/klippy.log)

KevinOConnor, 2018-06-25:
This typically happens when one changes the g-code coordinate system and then does not change it back (eg, M82/M83/G90/G91/G92).  In particular, your activate/deactivate_gcode sections are issuing G92 commands, but that could conflict with the g-code in a print.  You might want to try going into relative mode (G91), move the filament, and then go to absolute mode (G90) in the activate/deactivate sequence.

-Kevin

ich777, 2018-06-25:
Tested this again with the same Gcode except that i've switched from two mcu's to one, changed nothing else, now it print's just fine but changed nothing.

Tanky for the explenation btw.

KevinOConnor, 2018-06-26:
That's odd.  If you think it is a bug in multi-mcu support, then please issue an M112 immediately after the event and then post the resulting klipper log file.  (We need the extra info that M112 provides.)

-Kevin

KevinOConnor, 2018-07-07:
There's a good chance this is the same root cause found in issue #461.  (It's hard to tell without the g-code that you were printing, but the bed_tilt could have been the cause.)

closed 2018-07-07 by KevinOConnor