Klipper3d/klipper #422: Pulse width for step/dir signals

motorist828, 2018-06-25:
I do not know English well, I hope you understand me.
I have slow drivers tb6560 that are controlled by Arduino Mega 2560
I fail due to very fast signals, after switching from mach3 to klipper
Can I set the minimum signal width?

AxMod3DPrint, 2018-06-26:
It's currently not possible, but if you provide your klippy.log as described in https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md there may be something else that is able to be done to help.

motorist828, 2018-06-26:
I do not have errors in the log
I get a print incline, because the drivers do not keep up with the signal 
this happens at any speed and acceleration
[image]

dragonnn, 2018-06-26:
I think he need the same option as here:
https://duet3d.dozuki.com/Wiki/Using_external_stepper_motor_drivers#Section_Configuring_the_Enable_polarity_and_step_timing
Some stepper driver just doesn't catch up when the pulse length is to short.

motorist828, 2018-06-26:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2137465/klippy.log)

KevinOConnor, 2018-06-26:
It isn't currently possible to configure the step signal width.

It should be possible to change the code to do so if you wish.  For the atmega2560 you would delete the "select NO_UNSTEP_DELAY" line from src/avr/Kconfig and then modify the "#define UNSTEP_TIME timer_from_us(2)" in src/stepper.c to use a greater time.  After these modifications it would be necessary to do a full rebuild: make distclean ; make menuconfig ; sudo service klipper stop ; make flash ; sudo service klipper start

Such a change is untested and would not reach nearly the stepping rates obtainable with an unmodified klipper.

-Kevin

motorist828, 2018-06-27:
Thanks, it helped!

closed 2018-07-07 by KevinOConnor

nick-parker, 2021-06-08:
Sorry to bump a zombie issue, but is there up to date reference material on this anywhere? I have a driver I'm pretty sure is upset with my pulse lengths, and the files referenced above don't seem to have the relevant lines anymore.

KevinOConnor, 2021-06-08:
The pulse width is now configured during "make menuconfig".  Be sure to enable "low level options" and then you should see the option to configure a custom pulse width.

Cheers,
-Kevin

nick-parker, 2021-06-08:
Right, awesome! I've got this test rig moving now, thanks a bunch!