Klipper3d/klipper #424: [REQUEST] M400 gcode command

Gi0cas, 2018-06-26:
Hi, i want to ask if there is the possibility to add support to another gcode command : M400.

In Marlin this command has this feature : "Finishes all current moves and and thus clears the buffer."
In my case it's useful because it force to wait the finish of the movements (especially the homing at the end of the print) than let to go on with other gcode instead of buffer all and consider the print done.

What do you think? It's a doable support?

oderwat, 2018-06-27:
`M400` is supposed to exist already. It is mentioned in the GCode Docs and implemented in code: https://github.com/KevinOConnor/klipper/blob/3e88ffabf1b3c54baa48aca058b98354d4d959bc/klippy/gcode.py#L482

KevinOConnor, 2018-06-30:
It is already implemented.

closed 2018-06-30 by KevinOConnor