Klipper3d/klipper #426: requests  COREXZ

malgangda, 2018-06-27:
Hello, I would like to ask if there is the possibility to provide support to another kinematics (CoreXZ)

I am currently building a 3D printer with an endless belt. Similar to the Blackbelt 3D printer.
For this printer I need a COREXZ kinematics in Klipper.
I have already run Klipper on a standard printer and am thrilled.
Hence this request.

What do you think? It is a feasible support?

oderwat, 2018-06-27:
This is so cool! Is that your own design? I would love to make this one of my next projects! Without having a printer for testing it probably will be pretty hard to help implementing the kinematics though.

malgangda, 2018-06-28:
Hello, the kinematics of CoreXY is the same as in COREXY. The axes Y and Z must be exchanged. So a motor to X and a motor to Z then proceed together.

KevinOConnor, 2018-06-28:
It's certainly possible to implement, but it would require development work.  There's some info on adding new kinematics at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md#adding-new-kinematics

-Kevin

closed 2018-07-07 by KevinOConnor