Klipper3d/klipper #427: Klipper Stopping on FAST moves.  MCU limit or bug?

Maglin, 2018-06-27:
There was a recent change to my Z which increased its resolution considerably.  One the first leveling it stopped on a 100mm/s Z/X/Y move.  This is one MCU running 4 Z steppers and the other MCU running 2 X/Y steppers.  I'm including the log files.  I did it several times and found if I move my Z faster than 50mm/s 
 now it does this.  I issued an M112 after each event so it's in there a few times.

[Klippy.log](https://pastebin.com/fVUCuTZR)

grigorig, 2018-06-27:
You have:
`MCU 'z' shutdown: Rescheduled timer in the past`
This is a typical sign off hitting the MCU step rate limit.

KevinOConnor, 2018-06-28:
As indicated, the requested step rate is too high for an atmega2560 (with your config a Z move of 100mm/s would be 160K steps/s).  I don't think it would make sense to attempt to engineer a software solution - it would be high effort, high risk, and there'd be little reward given there isn't much call to move the Z that fast.

-Kevin

Maglin, 2018-06-29:
Thank you for the reply.  I was just asking.  I would like my Z hop to be as fast as possible.  In a few weeks, the SAM4E will be running klipper.  I'll be using that before too long.

closed 2018-07-07 by KevinOConnor