Klipper3d/klipper #432: OS X Klipper installation wanted?

oderwat, 2018-06-27:
I have Klipper now running on my iMac including flashing the firmware and "everything". I needed to modify the Makefile because the OS X `ln` does not have the `-T` parameter. I also needed to skip `util.clear_hupcl(ser.fileno())` because this causes an error (I am not sure why). 

It works using Homebrew with adding a tap for the avm-gcc and some environment manipulations for getting `make menuconfig` running. But that is all I needed to change. I completed my Anycubic Kossel Plus klipper setup just running everything on the Mac.

Currently I am running it through a Octoprint installation which is also on my Mac but I am still experimenting with getting other slicers/hosts work directly with Klipper.

Is there any interest in adding this information and/or changes to the files somehow into the Klipper repository?

StefanMeGit, 2018-06-28:
I am highly interested in this!

ysoldak, 2018-06-28:
Interesting. Just write up an instruction in docs folder? Like InstallationOSX.md with all gory details so people can repeat the process. Later could integrate it into scripts themselves.

KevinOConnor, 2018-06-28:
I'm open to adding an install script, adding documentation, and/or making the code a little more resilient to different operating systems.

-Kevin

KevinOConnor, 2018-07-12:
Feel free to open a pull request when ready.

closed 2018-07-12 by KevinOConnor

myk3y, 2018-11-27:
So what happened to this?

dieideeistgut, 2018-11-27:
I'm gonna second here that i'd love to try this within OSX - sadly i can't get the perfect (brew-) environment going so some small hints regarding used brew packages/cask may be usefull here. I don't think that a full documentation would be needed though.

oderwat, 2018-11-27:
I am not using it like that anymore but this is what I did the last time I used it (Mai):

Installing Python 2.7 with a virtual env and the following packages.

```
Package    Version
---------- -------
cffi       1.11.5
greenlet   0.4.13
pip        10.0.1
pycparser  2.18
pyserial   3.4
setuptools 39.2.0
wheel      0.31.1
```

I previously also added the following dirty hack to the code. Not sure if that is still needed.

```
diff --git a/klippy/serialhdl.py b/klippy/serialhdl.py
index 0f9e87a..eeb756c 100644
--- a/klippy/serialhdl.py
+++ b/klippy/serialhdl.py
@@ -262,6 +262,7 @@ class SerialBootStrap:

 # Attempt to place an AVR stk500v2 style programmer into normal mode
 def stk500v2_leave(ser, reactor):
+    return
     logging.debug("Starting stk500v2 leave programmer sequence")
     util.clear_hupcl(ser.fileno())
     origbaud = ser.baudrate
```

I can't remember exactly what I installed with homebrew, but I guess something like the following:

```
brew install gnu-sed gawk binutils homebrew/dupes/gperf grep gettext ncurses pkgconfig
brew install avr-gcc
```

For `make menuconfig` to work you may also need a little dirty "hack":

```
brew link gettext --force
brew link ncurses --force
```

This may or may not help. Good luck.

trevjonez, 2018-11-27:
fwiw a single docker image that has octoprint with octoklipper and klipper all together would probably go a long way for this type of thing. I plan to do it at some point as I recently setup an intel nuc that should have the capacity to run many instances of such a container config :)

oderwat, 2018-11-27:
@trevjonez You mean something like https://github.com/sillyfrog/OctoPrint-Klipper-mjpg-Dockerfile ?

trevjonez, 2018-11-28:
yup pretty much exactly that. I am no docker pro but I do know that should at least give a good jumping off point for whatever configuration wants to be done. It is likely also a great way to fix/workaround the issues people have had building arm mcu firmware on ubuntu