Klipper3d/klipper #435: TMC2130 drivers not working with Arduino Due MCU

jakep82, 2018-06-28:
I'm using an OULWare board (http://www.oulware.org/index.php/Printer_Controller) with a Due as the MCU.  I have everything working except my TMC2130 drivers on X and Y.  I've tested these same pin definitions on Cruwaller's fork and they work fine there, but I'm unable to get these drivers to work on the main Klipper branch.  G28 for X and Y simply times out with no movement or indication that the drivers have initialized.  Perhaps I'm doing something wrong?   Log file attached.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2143705/klippy.log)

StefanMeGit, 2018-06-29:
Same for me... has anybody the TMC2130 running with SPI on the DUE with Klipper?

jakep82, 2018-06-29:
My TMC2130 drivers work fine on the Cruwaller fork of Klipper, but I would really like to run the main branch.  I assume it's something simple, but I haven't been able to figure it out.

StefanMeGit, 2018-06-29:
U can acknowledge that the main branch is not working with SPI?

jakep82, 2018-06-29:
If I run the Cruwaller fork my drivers work fine with SPI.  If I run the main branch with the appropriate changes to printer.cfg but the same driver and CS pin definitions, the drivers do not work with SPI.

StefanMeGit, 2018-06-29:
@KevinOConnor do you have any idea? Did it maybe never worked with the DUE?

jakep82, 2018-06-29:
For reference, this is the basic config file that works on the Cruwaller fork.

[oulware.cfg.txt](https://github.com/KevinOConnor/klipper/files/2150767/oulware.cfg.txt)

hg42, 2018-06-30:
@jakep82 could you provide a log of such a non-working session?
Kevin cannot do much without real facts.

What means "do not work"?
There are a lot of possibilities.
No movement at all? 
Or some unexpected behavior?
etc.
To be sure, you know that cruwaller's config differs a lot from Kevin's?

KevinOConnor, 2018-06-30:
On Wed, Jun 27, 2018 at 09:14:30PM -0700, jakep82 wrote:
> I'm using an OULWare board (http://www.oulware.org/index.php/Printer_Controller) with a Due as the MCU.  I have everything working except my TMC2130 drivers on X and Y.  I've tested these same pin [...]

I don't know why that would be.  Nothing has changed recently in the
TMC2130 code nor in the Arduino Due code.  I looked and the sam3x8e
SPI code on the cruwaller repo seems identical to the code on the main
branch.

I put up a test branch that you could try running the new DUMP_TMC
command on:

cd ~/klipper ; git fetch ; git checkout origin/work-tmc2130-debug-20180630 ; sudo service klipper restart

However, this debugging code hasn't itself been tested, so I'm not
sure how helpful it will be.

-Kevin

KevinOConnor, 2018-06-30:
FYI, I've pushed the new DUMP_TMC command to the master branch and I have removed the work-tmc2130-debug-20180630 branch.

-Kevin

jakep82, 2018-06-30:
All it returns is missing stepper.

> Recv: !! Error on 'dump_tmc': missing STEPPER

BlackStump, 2018-07-01:
@jakep82   I notice in your config you use ; to comment out instead of # 
example
[stepper_y]
driver:             tmc2130_y
;driver:             drv8825_y

I dont know if that is correct maybe it is, I only see in configs the use of # to comment a line out

AxMod3DPrint, 2018-07-01:
Just as a note with dump_tmc

 `Recv: !! Error on 'dump_tmc': missing STEPPER` 

Suggests there's a missing parameter, as it's also the same on Arduino Mega.

@KevinOConnor can you give a usage example, please?

Also as a note, if your TMC drivers are not working issue a `RESTART` on the console. That seems to kick mine back into life if they're refusing to move. It may well be there's an SPI issue that underlies this, as I'm having to do that a fair bit, but it's also possible that the Ubuntu ZBox server that that particular printer is connected to is a bit on the warm side so it could be that also on my side. 26°c plus rooms are not good for computers..

KevinOConnor, 2018-07-01:
On Sat, Jun 30, 2018 at 04:52:50PM -0700, jakep82 wrote:
> All it returns is missing stepper.
> 
> > Recv: !! Error on 'dump_tmc': missing STEPPER

There should be a STEPPER parameter with the name of the tmc2130 that
is being queried.  For example: DUMP_TMC STEPPER=stepper_x

There's a little bit of info in docs/G-Codes.md.

-Kevin

jakep82, 2018-07-01:
Results below.  I've tried restart, firmware_restart, changing my power up sequence, etc.  Nothing seems to give me functioning drivers for X and Y.

> Send: DUMP_TMC STEPPER=stepper_x
Recv: // GCONF:        0
Recv: // GSTAT:        0
Recv: // IOIN:  1000000
Recv: // TSTEP:  4000000
Recv: // XDIRECT: 12000000
Recv: // MSCNT: 2d000000
Recv: // MSCURACT: 6a000000
Recv: // CHOPCONF: 6b000000
Recv: // DRV_STATUS: 6c000000
Recv: // PWM_SCALE: 6f000000
Recv: // LOST_STEPS: 71000000
Recv: ok
[...]
Send: DUMP_TMC STEPPER=stepper_y
Recv: // GCONF:        0
Recv: // GSTAT:        0
Recv: // IOIN:  1000000
Recv: // TSTEP:  4000000
Recv: // XDIRECT: 12000000
Recv: // MSCNT: 2d000000
Recv: // MSCURACT: 6a000000
Recv: // CHOPCONF: 6b000000
Recv: // DRV_STATUS: 6c000000
Recv: // PWM_SCALE: 6f000000
Recv: // LOST_STEPS: 71000000
Recv: ok

jakep82, 2018-07-01:
After attempting to home X, the GCONF value changes, but I get not movement on either stepper.

> Send: G28 X0
Recv: !! Failed to home x: Timeout during endstop homing
Recv: ok
[...]
Send: DUMP_TMC STEPPER=stepper_x
Recv: // GCONF:   209609
Recv: // GSTAT:        0
Recv: // IOIN:  1000000
[...]
Recv: // TSTEP:  4000000
Recv: // XDIRECT: 12000000
Recv: // MSCNT: 2d000000
Recv: // MSCURACT: 6a000000
Recv: // CHOPCONF: 6b000000
Recv: // DRV_STATUS: 6c000000
Recv: // PWM_SCALE: 6f000000
Recv: // LOST_STEPS: 71000000
Recv: ok
[...]
Send: DUMP_TMC STEPPER=stepper_y
Recv: // GCONF: 73000000
Recv: // GSTAT:        0
Recv: // IOIN:  1000000
Recv: // TSTEP:  4000000
Recv: // XDIRECT: 12000000
Recv: // MSCNT: 2d000000
Recv: // MSCURACT: 6a000000
Recv: // CHOPCONF: 6b000000
Recv: // DRV_STATUS: 6c000000
Recv: // PWM_SCALE: 6f000000
Recv: // LOST_STEPS: 71000000
Recv: ok

AxMod3DPrint, 2018-07-01:
Random question, have you tried upping the current to the drivers as a test?

jakep82, 2018-07-01:
Changing the current makes no difference.  I've tried setting the run and hold currents to 1 amp.  I've also tried reducing the stealthchop threshold.  The drivers simply don't initialize, and I get zero indication of movement or power to the motors.

AxMod3DPrint, 2018-07-02:
The fact you are getting values back from dump_tmc means they are getting powered and your CS pins are correct you'd be getting 0 for everything. Klipper seems to be doing what it should be. I'd suggest checking your pin out and wiring, if you've not already.

StefanMeGit, 2018-07-02:
I run my DUE now without the TMC2130 and need the printer for some parts. Tomorrow I ok get a new due and will go in with some test to verify jakep82 problems.

My last tests also didn’t power up the motors (no click if I try to move the motors for the fist time) the just like death. No movement at all. On my MKS Board i didn’t had that problem before. They worked as aspected.

KevinOConnor, 2018-07-02:
On Sun, Jul 01, 2018 at 08:10:07AM -0700, jakep82 wrote:
> After attempting to home X, the GCONF value changes, but I get not movement on either stepper.
>
> Recv: !! Failed to home x: Timeout during endstop homing
> Recv: ok
> [...]
> Send: DUMP_TMC [...]

This definitely looks like an SPI problem.  My first guess is that
there is an issue in the Due spi code, but I don't see any difference
in that code with the code on the cruwaller branch.  What version of
cruwaller's code were you previously using?

One thing you could try - modify the 4000000 to 100000 in the
following line of code in klippy/extras/tmc2130.py:

        self.mcu.add_config_cmd(
            "config_spi oid=%d bus=%d pin=%s mode=%d rate=%d shutdown_msg=" % (
                self.oid, 0, cs_pin_params['pin'], 3, 4000000))

then run "sudo service klipper restart".

If that doesn't work, another thing you could try is commenting out
your fan section.  (The ar10 pin on the DUE is, oddly, also connected
to the NPCS0 pin..)

-Kevin

jakep82, 2018-07-02:
I was running his OULWare specific fork at the link.  I'll try the 2 things you mentioned tonight and see how it goes.

[https://github.com/OULWare/klipper](url)

jakep82, 2018-07-03:
I tried both of these things and neither worked.  I'm going to boot back into the OULWare/Cruwaller SD card I have and make sure the drivers are still working, but last time I checked they still worked on his fork.  Not sure why they aren't working here.

jakep82, 2018-07-03:
I switched back to the OULWare fork, and the drivers are working perfectly there.  Not sure why they work there, but not on the main Klipper branch.  The pin definitions are the same, and I think I have everything else set up correctly in my printer.cfg file for the main branch.

KevinOConnor, 2018-07-04:
On Mon, Jul 02, 2018 at 06:38:14PM -0700, jakep82 wrote:
> I switched back to the OULWare fork, and the drivers are working perfectly there.  Not sure why they work there, but not on the main Klipper branch.  The pin definitions are the same, and I think I have everything else set up correctly in my printer.cfg file for the main branch.

I'm not sure why.

What branch revision of OULWare are you using?  (git describe --tags)

What does DUMP_TMC report when using the slower spi rate?

Are there other SPI devices on the board (like thermocouples)?  If so,
try adding output_pin sections to force their cs pins high.

I can't see any significant changes on the OULWare branch, so I can't
explain why you would see a significant difference.  That said,
there's tons of "cosmetic" changes on that branch, so anything could
be hiding in there.

-Kevin

jakep82, 2018-07-04:
After changing the SPI rate to 100000 and restarting Klipper I get the info below.  I don't have a thermocouple.

> Send: dump_tmc stepper=stepper_x
Recv: // GCONF: 4924efcc
Recv: // GSTAT: 80000000
Recv: // IOIN: 40000000
Recv: // TSTEP:  4000000
Recv: // XDIRECT: 12000000
Recv: // MSCNT: 2d000000
Recv: // MSCURACT: 6a000000
Recv: // CHOPCONF: 6b000000
Recv: // DRV_STATUS: 6c000000
Recv: // PWM_SCALE: 6f000000
Recv: // LOST_STEPS: 71000000
Recv: ok
[...]
Send: dump_tmc stepper=stepper_y
Recv: // GCONF: 249277e6
Recv: // GSTAT:        0
Recv: // IOIN:  1000000
Recv: // TSTEP:  4000000
Recv: // XDIRECT: 12000000
Recv: // MSCNT: 2d000000
Recv: // MSCURACT: 6a000000
Recv: // CHOPCONF: 6b000000
Recv: // DRV_STATUS: 6c000000
Recv: // PWM_SCALE: 6f000000
Recv: // LOST_STEPS: 71000000

On the OULWare branch the output of git describe --tags is below.

> pi@octopi:~/klipper $ git describe --tags
v0.5.0-301-g2925618

KevinOConnor, 2018-07-05:
If you pull the latest code and reflash the micro-controller, does it help?

If it does not help, could you try disabling the thermocouple in your oulware branch and see if that has any impact on the oulware operation?

-Kevin

jakep82, 2018-07-06:
This fixed it!  Thanks for looking into this issue.  I've got my drivers running with sensorless homing now.  I'm about to do a test print, but everything seems 100% functional so I'm going to close this issue.

closed 2018-07-06 by jakep82