Klipper3d/klipper #436: Problem with Bltouch retraction

vsusanto, 2018-06-28:
Thank you for creating Klipper.
Klipper has been working ok with my printer, until recently.
The problem is the bltouch push-pin does not retract after the second homing (G28). While the pin is still deply, the tool head moves to X=0 and Y=0 position. 
The pin retracts after the tool head reach (0,0).
During bed leveling process, the push-pin is not deploy properly at the first mesh points.
Somehow, the alarm on the sensor does not reset itself when it receives 90 degree angle.

I have 2 bend push-pins resulted from this error.

Could the tool head move up, let say 5 mm, then retract the push-pin before it goes to 0,0 position?

Thank you.

oderwat, 2018-06-28:
You may try to adopt this for your printer.

```
[probe]
pin: PC4
z_offset: 1.11
speed: 2.0
activate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=10
    G4 P100
deactivate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=90
    G4 P100

[homing_override]
axes: z
set_position_z: 5
gcode:
    SET_SERVO SERVO=bltouch ANGLE=160
    G4 P200
    G90
    G1 Z10 F600
    G28 X0 Y0
    G1 X150 Y115 F7200
#    G1 X50 Y30 F7200
    G28 Z0
    G1 Z5
    SET_SERVO SERVO=bltouch ANGLE=90
    G4 P100
    G1 X0 Y0 F7200
```

I think that what you describe there is actually just one of the effects because Klipper does not implement BLTouch correctly (yet). See https://github.com/KevinOConnor/klipper/issues/387

You may want to check out https://github.com/KevinOConnor/klipper/issues/391 which is now closed, but contains a large deal of what I experienced before.

vsusanto, 2018-06-28:
Haas,
Thank you.
It looks like the line for activating bltouch got shorter. Was that line
caused the problem?
I will try your configuration.

Sincerely,

Victor

vsusanto, 2018-07-01:
Problem Fixed. 
I made the change to the homing override section.

Thank you, Haas.

closed 2018-07-01 by vsusanto