Klipper3d/klipper #440: Extruder 1 and Extruder 2 have different z_offsets

w1ebr, 2018-06-29:
I am sorry if this was answered already; I couldn't find it. Is there a way to handle the offset difference between two different hotends in m,y dual carriage setup. The hotends are not perfectly aligned with each other. The z_offset for the probe is the offset between the probe trigger point and the T0 hotend. Is there a way to compensate for T1 having a slightly different z_offset?

Gene

hg42, 2018-06-29:
so one of your hotends constantly collides with the printed part?

KevinOConnor, 2018-06-30:
Use the SET_GCODE_OFFSET command in the extruder's activate/deactivate_gcode scripts to apply an offset when switching extruders.

-Kevin

w1ebr, 2018-06-30:
Thank you, Kevin.

closed 2018-06-30 by w1ebr

mamoit, 2018-07-18:
Modifying the G-Code makes it extremely difficult to use the same one on 2 printers that are the same model, but not exactly the same.

KevinOConnor, 2018-07-19:
On Wed, Jul 18, 2018 at 01:18:48PM +0000, Miguel Almeida wrote:
> Modifying the G-Code makes it extremely difficult to use the same one on 2 printers that are the same model, but not exactly the same.

The Klipper config file lets one define commands to run on each
extruder activation (see config/example-extras.cfg).  The
recommendation would be to use SET_GCODE_OFFSET there.

-Kevin

mamoit, 2018-07-19:
That's a way better option :+1:
By the way, is there a way to update the config file with a GCODE or extended command?
In order to save a change to an offset for eg.

bruce356, 2018-07-31:
Hi,  @mamoit  I have a Dual X carriage printer with tool 1 & 2 but I need to set a small offset not only in Z but also a small offset in Y axis.

Hoping someone could be a little more descriptive that above because I am not sure how this would be added. Does the offset need to be applied to both tools or only one.
Thanks and regards - bruce

 [extruder1]
step_pin: ar26
dir_pin: ar25
enable_pin: !ar27
step_distance: .0020704
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar3
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250
deactivate_gcode = 
#SET_GCODE_OFFSET
	G90
	G1 X250 Y250 F5000
	G1 X250 Y259 F5000
	G1 X210 Y259 F5000
	G1 X210 Y250 F5000
activate_gcode = 
#SET_GCODE_OFFSET
	G90
	G1 X210 Y250 F5000
	G1 X210 Y259 F5000
	G1 X250 Y259 F5000
	G1 X250 Y200 F5000

mamoit, 2018-07-31:
if you calibrate T1 relatively to T0, you can set T0 as having no offset (`SET_GCODE_OFFSET X=0 Y=0 Z=0`) and then apply the offset to T1.
You will need the `activate_gcode` on both, so that when T0 comes into action it the offset is removed, and when T1 is activated, it is applied.

bruce356, 2018-08-01:
@mamoit thank you for the reply, I have configured the extruders as follows, hoping this is what you advised.
[extruder]
step_pin: ar23
dir_pin: ar22
enable_pin: !ar24
step_distance: .0020704
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar2
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog15
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250
deactivate_gcode = 
	SET_GCODE_OFFSET X=0 Y=0 Z=0
activate_gcode = 
	SET_GCODE_OFFSET X=0 Y=0 Z=0

[extruder1]
step_pin: ar26
dir_pin: ar25
enable_pin: !ar27
step_distance: .0020704
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar3
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250
deactivate_gcode = 
	SET_GCODE_OFFSET X=291 Y=-0.25 Z=0.02
activate_gcode = 
	SET_GCODE_OFFSET X=291.4 Y=-0.25 Z=0.02
Regards - bruce

mamoit, 2018-08-01:
you don't actually need the `deactivate_gcode`, only the `activate_gcode`, since when one tool is deactivated, you do not want to set the offsets to the values of it, but instead to the offsets of the one that is about to be activated.

bruce356, 2018-08-02:
@mamoit , in order for the X axis tools to line up in Marlin when printing (dual X carriage) this is how I had to set the offset because this is the distance between T0 & T1  when  tools are parked.

I thought it would simply translate into Klipper but obviously not, thanks for the guidance.
I have removed the "deactivate_gcode"
I will measure the difference in position of the tools @ say X125 which is the centre of my bed and place that figure into "activate_gcode" for T1.
Thanks very much - bruce

mamoit, 2018-08-02:
@bruce356 for reference, I have something like this for my 2nd extruder:
```
activate_gcode:
  SET_DUAL_CARRIAGE CARRIAGE=1
  SET_GCODE_OFFSET Z=-0.12 X=-.5
deactivate_gcode:
  G0 X348 F18000
```

The X offset can be defined by defining for the X axis `position_min` and `position_max`, or for the `dual_carriage`, depending if it is the 1st or the second head.

bruce356, 2018-08-02:
@mamoit , thanks that was very helpful,  I am also confused about the X carriage configs not sure if they are correct.
[stepper_x]
step_pin: ar17
dir_pin: !ar16		
enable_pin: !ar48
step_distance: .0125
endstop_pin: ^ar37
#endstop_pin: ^ar36
position_endstop: 0
position_max: 250
homing_speed: 60

[dual_carriage]
axis: x
step_pin: ar29
dir_pin: !ar28	
enable_pin: !ar39
step_distance: .0125
endstop_pin: ^ar36
position_endstop: 291
position_min: 0
position_max: 291
homing_speed = 60

My Bed is 250mm wide and each "tool carriage" moves approx. 41mm off the edge of the bed.
----41mm----===========250mm============----41mm----
         T0                                      BED                                           T1
Your advice would be appreciated.
Regards - bruce

mamoit, 2018-08-02:
`stepper_x:position_min` should be -41, as should the `stepper_x:position_endstop`, since when it homes it is in -41.
Don't forget to put the right coordintates on the `deactivate_gcode` of the extruders to park the head.

torbenw1996, 2018-08-12:
Is it possible to probe the length (e.g. with a piezo disc) off both tools at the begin of the print and adjust it with the measured values?