Klipper3d/klipper #443: Last version. Fire in the hole! Problem firmware for Arduino Due.

khseal, 2018-06-30:
Updated to the latest version of the Klipper.
The extruder itself began to heat up.
Klipper is not connected to the microcontroller.
Add out folder and log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2151679/klippy.log)

[out.zip](https://github.com/KevinOConnor/klipper/files/2151678/out.zip)

khseal, 2018-06-30:
Returned to a stable version of Arduino Due does not respond =(

khseal, 2018-06-30:
The problem is solved. The Klipper is compiled without errors on Ubuntu 18.04 but does not work. Attention, the extruder is heated after the firmware!

StefanMeGit, 2018-07-01:
That’s normal... Exchange the fan and extruder heater output. In this Case next Time just tue fan goes on instead of the heater.

khseal, 2018-07-01:
Nothing normal. Normally everything is compiled under Ubuntu 16.04.

KevinOConnor, 2018-07-01:
Just so I understand, you're saying the micro-controller code compiles okay on Ubuntu 16.04, but is being mis-compiled on Ubuntu 18.04?  If so, I'll take a look, but it will be some time before I can setup everything to test that.  I do regularly test with newer versions of the toolchain (specifically the gcc versions that ship with Fedora) and I have not seen any problems.

-Kevin

khseal, 2018-07-01:
Yes, you understood me correctly. Compile in Ubuntu 18.04 passed without errors and warnings, but the firmware does not work correctly. I attached my folder 'out'.

PreownedFIN, 2018-09-07:
Just going through the issues and noticed this.  

What expansion shield did you use?  
It is a known problem with all mass manufactured RAMPS-FD shields that if the DUE doesn't have power, the heaters get turned on.  
None of the sub-2.0 versions should be used anymore.  
Due to the V1.0, 1.1 and 1.2 having the fatal problem, the 2.0 revision was never implemented and pretty much all RAMPS-FDs on the market have the problem.

https://github.com/bobc/bobc_hardware/issues/19

khseal, 2018-09-08:
I modified ramps 1.4. There is a heating problem in the absence of power supply at the Arduino DUE.

KevinOConnor, 2018-09-09:
So, should this be closed then?  (I haven't had a chance to test Ubuntu 18.04 myself.)

-Kevin

khseal, 2018-09-09:
I will check the firmware on Ubuntu 18.04 in the near future. Initially, the problem was still because of the problem firmware. Heating the printer is a consequence of the configuration hardware and problem of firmware.

khseal, 2018-09-11:
The problem with the firmware in version 18.04 is still present. I tried to flash on two laptops, as a result, after firmware, the microcontroller does not respond to requests.

mcarpenterjr, 2018-10-09:
Any fix for this yet? Just attempted to compile on a RPI3 and Ubuntu 16.04 with no luck. getting the same same errors;

```
Extruder max_extrude_ratio=0.149671
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
```

Everything looks OK state goes to operational, except temperature is never logged and when I attempt to issue a command like "home" I get timeout on serial connect.

KevinOConnor, 2018-10-15:
I got a chance to install Ubuntu 18.04 and I can confirm that Klipper does not build arm binaries correctly on that OS.  It appears the Ubuntu 18.04 gcc-arm-none-eabi and/or libnewlib-arm-none-eabi packages are linking in the wrong version of the libc functions.  Instead of getting the thumb armv7-m C library we're getting a library for some other arm instruction set.

This does not appear to be a Klipper issue.  It's not clear to me how to prevent Ubuntu from miscompiling Klipper.  I recommend not compiling Klipper on Ubuntu 18.04.

Separately, I was able to build regular AVR binaries without issue.  So, this appears limited to the arm toolset.

-Kevin

khseal, 2018-10-18:
This is very bad. I installed almost in all computers lubuntu 18.04 =)

KevinOConnor, 2018-12-22:
As of commit 50e9c673, Klipper will attempt to detect the broken compiler in Ubuntu 18.04 and produce an error message.  It looks like there is an Ubuntu issue for it at: https://bugs.launchpad.net/ubuntu/+source/newlib/+bug/1767223

-Kevin

KevinOConnor, 2019-01-03:
I'm going to close this issue as I think the recently added build time error checking addresses this (at least as best as we can).

-Kevin

closed 2019-01-03 by KevinOConnor