Klipper3d/klipper #446: Too less support for this great Projekt!

StefanMeGit, 2018-07-01:
Hi guys! I am using this now for a couple of months and I am very satisfied with the software and the great work from Kevin. This project has earned more support! I see this as an real issue we have to solve!

https://www.patreon.com/koconnor/posts

oderwat, 2018-07-01:
I am patron since I use Klipper every day. I also tried to get people to the project and partially succeed. Actually my first post in a FB group about Klipper was that I managed to get the Ender 3 profile added to Klipper and that I think people should become Patreons for this cool project.

Then I wrote about how to use Klipper without using OctoPrint and got "put down" by the dev and some people. Their reasoning was, that using a hardware UART to USB would introduce errors. But this is even the norm on many printer mainboards. So I got it to work and myself is use this for many weeks now, every day. No errors. I can control and do print with Klipper directly from Simplify3D, Cura and Slic3r and every other application which uses the serial port. No OctoPrint needed. Klipper became a "black box" hardware, which sits between PC and printer. But the interest for this from the dev was zero, so I stopped promoting it. It felt like telling people to do something "illegal" with Klipper. I never got an answer, why the UART 2 USB works for all the Melzi boards but is not to be used with Klipper.

Here I got already disillusioned and my plan to increase my Patron and time contribution substantially got a bit damper. But I wanted to use Klipper anyway.

Next I wrote a software which lets you adjust the pressure advance value very quick and without wasting a lot of filament. Much quicker than printing multiple of the squares the Klipper tutorial proposes. Even with people validating my results it got put down by the dev. So I stopped developing this for anything more than my own usage.

Next I wrote about the BLTouch implementation being not correct. I got some code which helps with probing into the main branch. For the real problem I basically got put down again. I myself use the BLTouch for homing and currently write repeatable probe code and think about a way to use standard derivation measurement for better results. Not sure if I am going to contribute that code. I don't feel it is "of interest" for anybody. I also want to try a capacitive probe soon.

Lately I added the Anycubic Kossel (Linear) Plus printer config... So I can promote Klipper in the other FB group I am usually writing in. Got put down, because the "name" is confusing.

I also made Klipper work on my Mac without using the RasPi (while getting the Anycubic Kossel to work). This now got some interest but it is kinda worthless if you can't also print on it without adding OctoPrint (running on the Mac) into the equation. While using OctoPrint works I started to find a way to run it directly through a hardware serial port, so that the Slicers pic it up. This only worked by using two USB 2 TTL devices as null-device with some redirection. I think that could be perfected but I don't even dare to ask the dev about changing Klipper so it can work on a real hardware serial. He stated that he only wants his virtual device because of "errors".

You may see that I am interested in Klipper and spent some time with it already. It would feel like a big waste of time, if I didn't just print on two of my printers with Klipper in the way I want to.

Back to Patreon:

There was no update since Mai. I think that there is a lot to tell too the people. Getting a nice update there, would be a reason to repost the link everywhere. Seems not to happen. I would expect at least a monthly update.

One of the questions I usually get asked first is: "Does it have mesh bed leveling?", which was also the highest priority voted by the patrons. Afaik, nobody works on that. And to be honest, I think this is a gimmick feature which is not needed. I prefer having a plain surface and also tilt level my bed straight. The only reason I use a BLTouch is because I switch build surfaces a lot and this makes it a no brainer. Still there is the question: Will that be an upcoming feature?

So yeah! I still like Klipper but regret wasting so much time in discussing pointlessly with the dev and other people which seem to have much different views on what is important and what make Klipper a success. Currently I have the feeling that it is still meant to be an "elite only" project. Which is OK for the time. It will get more attention when features complete and harden.

epoxi, 2018-07-01:
@oderwat, I find very interesting your work. I not a user of clipper yet, but I'm looking to adapt some of my machines to use it. I found it very interesting and convenient as all my machines are CORE-XY.

But I would love to keep direct serial connectivity to the machine, so I can still use it connected to a PC (with Simplify3D, Pronterface, etc). I understand that the serial device would be already taken by the klipper firmware. So here is my actual stopper for using Klipper. Maybe a serial proxy (?) kind of implementation would help, but don't know if this is possible.

This comment is kind of OT, but @oderwat comments reminded my issue with the actual implementation.

hg42, 2018-07-01:
@epoxi there sems to be some misunderstanding about how Klipper works.
The mcus are flashed to a different firmwate (called klipper), which can only communicatebwith the python part (klippy).
If you want to use it with another host, you first need to stop klippy and re-flash the mcu with some usual firmware. So what's the point of this?

hg42, 2018-07-01:
however, you can control klippy via /tmp/printer, I do it myself using pronterface.

oderwat, 2018-07-01:
@hg42 Why should this be pointless? It is very convenient and easy to do. I did this a lot in the last days while trying things with the delta. I have the delta printer directly connected to my iMac and can with a simple script call either flash Klipper and use klippy or flash one of my Marlin versions and use that.

I already planned to enhance my RasPi based Klipper with a GCode command which switches firmware (Klipper, Stock Firmware, TH3D, my own version of Marlin and maybe Repetier variants) by re-flashing and changing how the serial is forwarded to the computer. I had to write my own scripts for stuff and to allow switching firmware with my "Black Box" was one of the things I included into the concept. I wanted to back this up with compiled firmware images for different supported printers which the user simply can select and flash without knowledge on how stuff actually works.

epoxi, 2018-07-01:
@hg42 I do understand how klipper works (hosts controlling an mcu wich interfaces with hardware).
The thing is that sometimes I need to send gcode from a computer rather than from octoprint GUI, either by serial (usb-serial), or maybe telnet. 

Somethig like this:

[image]

oderwat, 2018-07-01:
@epoxi well that is what I do since weeks using the USB2TTL serial hardware. You need to re-configure the RasPi to have a UART available, add actual USB2TTL hardware and run something like socat on the RasPi to connect the real serial to Klipper. If somebody would re-write Klipper to actually be able to attach to the UART serial directly it would be most natural and easy to use it that way. That is why I am disappointed by the dev, as he refused to do so, because "it is unsafe" to use an UART device instead of the virtual port.

oderwat, 2018-07-01:
@hg42 I don't know what you try to explain to me, what I didn't already mentioned? I said it is both possible and that I do it since weeks.

hg42, 2018-07-01:
I only want to say, that two hosts in parallel may be problematic or not as easy as he thinks.

I agree that klipper should open any file.
Btw. did you try that lately? At some point in time I had the impression there were some changes that could allow this now (though I didn't test it yet, using -I /dev/... or may be in the config).

oderwat, 2018-07-01:
I didn't suppose to use more than one host at once.

What do you mean with klipper should open any file? Using a real serial device instead of "/tmp/printer" or opening a virtual serial instead of the hardware `/dev/USB..` ports?

What I would like is that Klipper can be set onto a hardware serial device instead of `/tmp/printer`. This would save me some ugly "socat" juggling.

For using Klipper more easily inside of an USB isolated docker it would also be good if it would work in virtual serial ports for accessing the printer. I could not test if this works, because my printers are running. I am also personally not really interested in it, because it is not part of anything I would consider to use in my setups.

grigorig, 2018-07-01:
FWIW, I don't think mesh bed levelling is a gimmick. I have yet to see a perfectly plane build surface. In practice, all surfaces are twisted and/or bowed to some degree. And if you are lucky enough to have a nearly plane surface, repeated heating cycles and/or printed part removal can easily break that. And nothing is more frustrating than an inconsistent first layer that makes your prints fail.

oderwat, 2018-07-02:
@grigorig Of course you can believe what you want.

I just don't see the point: Even if if would work 100%, you end up with a non plain bottom layer and if not using some z fading this will also lead to a non uniform top surface. When you use z-fade it will result in uneven measurements and a non uniform bottom of the part.

So it seems it is only practical for small differences in the surface. The effect is also noticeable for the first layer. It also very much depends on your layer height. If you use 0,2 mm there is +/- 0,05 mm not really a problem for your first layer with a decent build surface.

I also actually don't know any probe/hardware setup which is exact enough to measure differences on a mirror-glass, borosilicate-glass or rolled C100S steel plate. They are just plain enough or bad and need replacement. From my long term experience: They stay plain. Even when the C100S is pretty heavily bend to remove parts it snaps back to be plain enough for the next print.

I also use all of this. I currently print on a foil based magnetic flexible PEI surface which even has slight bubbles and bends because it got damaged on the transport. I still don't have a problem with first layers. And I print a lot of small and big things all day and use the surface material and physics which matches the task.

But lets going for a bend aluminum plate with some flexible top or say just some blue tape:

How many probe points do you want to take for the mesh? If you use say 3x3 points it already takes some time but how to interpolate between those points. Bends are usually not linear but what curve are they? So lets say you have a 200x200 build plate and want really fine results. You may need a 10x10 grid of measure points which are 100 points and would still need to interpolate pieces of 20x20 mm. Those areas will still be bended and linear interpolation of a curved surface will make it even worse in some areas. You also can't be sure to have gotten the highest or lowest point in the samples. So you just put some zick-zack over round bends.

For all of those reasons I am not convinced that mesh bed leveling should be considered as more than a gimmick for real world applications up to a size where you could use a glas or high precision spring steel or other metal bed. So I would rather invest in some decent build plate instead of having a fake correction in software.

Having much bigger build areas may profit from mesh bed leveling. I still don't think that it will be interpolate in a good way if the differences go above a certain amount. It also needs a lot of sample points to actually work.

So what's left is tilt-bed leveling. This has a real usage for printers where you can't adjust the bed to nozzle distance. This works well if the adjustment is a real 3D adjustment besides that the nozzle will still not be plain regarding the surfaces.

Grimpster, 2018-07-02:
I need to kick in here.
tldr: (bold statement incoming agree with it or not) mesh bed if for lazy people and printers that aren't built correctly!

I came from ABL via a zortrax m200, moved to marlin and got my mesh running on a rather wonky discount printer "x5s that I've been rebuilding" then to klipper.

When I started using klipper I was on the edge of loosing it. My bed was impossible to print on, blaming it on the bed and not having a mesh. I was seconds away from quitting klipper cause of this.

Luckily for me, my stubborn brain took over. And it had me thinking.
-Using glass I have a 99.9% plane surface it being stiff and well, flat.
--Any artifact that may occur is due to a printer not build correctly.

I had concave areas on my plate, blaming it on my glass and my bed was tilting like a drunk squirrel.
I changed all the rollers on all my axis, you could easily tell they were grade D

Then I fixed my tilting by getting propper bearings from the Z rods.

It took me a while to get there, but not only did I get a near perfectly level buildplate but also my print quality increased tenfold.

For the sake of laziness I'd like to see meshbed dont get me wrong. But its a nice-to-have not a need-to!

What this project really needs in my opinion is a solid community. I'm on both the discord fb page. And the level of activity is well, lack cluster on both of them.

yschroeder, 2018-07-02:
Mesh bed leveling is a huge selling point for a 3D printer firmware. Especially for people who just want to print, and not fiddle (too much) with their setup. If tinkering with your 3D printer is your hobby, it is a part of your hobby to build a printer that has a level build surface. If your hoby is something else and you only want to use your printer as a tool, then you want to have mesh bed leveling.

We have a Prusa i3 MK2.5 at work. Upgraded from an MK2S. Printing directly on the heated PCB with a sheet of spring steel ontop is a great improvement over glass surfaces when removing finished prints. But to make such a setup work you cannot go without mesh bed leveling. I see the mesh leveling as an enabling technology for other improvements in 3D printing.

AxMod3DPrint, 2018-07-02:
Going to chip in here with my thoughts. @KevinOConnor has done a whole lot to push things forward in the thinking of how a 3D printer firmware can be implemented. There is always going to be resistance in how this is perceived by the masses as it's the new kid on the block, relatively speaking. Klipper is ultimately still beta firmware as far as I see it, so more features and tweaks will be added over time, by Kevin or the contributors. You have to remember though, the Klipper Dev team is essentially just Kevin, and as such there is only so much one person can do as such contributing to the firmware helps push it forward.

I know it's frustrating when a firmware you like doesn't do what you need or want it to do. Honestly, I think key features like mesh levelling will come in time, whether you think it's needed or not. It is down to the contributors and community as well. Of the issues mentioned here, there's a few edge use cases which may or may not benefit the whole firmware, I can't say. I certainly have an edge use in that 2 of my 5 printers, of which 3 run Klipper, are connected to a ZBox Ubuntu Server which powers 2 Octoprint and Klipper instances. The thing is the whole picture needs to be looked at and not just how an individual uses it. So yes you may have a good idea, but it may not be a high priority over something like getting the menu structure for LCDs in place, Mesh Levelling or making sure we have generic pin outs for common printers or boards. 

It's just a case of even if a PR is rejected you've done something to help push.it forward

oderwat, 2018-07-02:
@yschroeder "sprint steeel ... cannot go without mesh bed leveling" This is my usual setup since 1 1/2 years and I never needed mesh bed leveling on any printer to have it work. Maybe your spring steel is not actually quality spring steel?

But yes, I understand that it is a selling point. I also see a big market for people which "think" it solves their problems so they buy probe extensions for their printer and then just switch to mesh bed leveling because they think it is the best feature available. So they actually don't understand the differences in mesh bed leveling and bed tilt leveling and use the first one instead of the later one, even with a plain enough bed. That will usually work and the resulting mesh consist mostly of probe and interpolation errors.

Anyway. I don't think that Klipper can leave out having mesh bed leveling in the long run.

hg42, 2018-07-02:
I also don't like tilt leveling for cartesian (including corexy etc.) printers.
Who wants to move the Z-axis while moving in X/Y?
I prefer leveling mechanically.
The only alternative would be using multiple Z motors to level automatically.

But, the best solution is a strong construction that ensures the leveling doesn't change over a long time and a lot of prints. I usually only have to level after crashes or when I change something with my bed.

Most leveling is necessary because of mechanical weaknesses or wrong designs. That's why you need it with cheap printers. It's obvious that a 100 EUR printer cannot use an 8 mm aluminum plate for 80 EUR as a heated bed. The plate is also much to heavy to be moved fast (usually in Y). So something like a corexy helps a lot.

KevinOConnor, 2018-07-17:
Thanks everyone for their support.  For those critical of the project, I'll try my best to interpret it as constructive criticism.  As for mesh support - I'm not currently working on it.  My understanding is that a group of developers with "Prusa MK3" printers have gotten pretty far in their implementation.  I look forward to seeing the results of their work.

Thanks again,
-Kevin

closed 2018-07-17 by KevinOConnor