Klipper3d/klipper #449: bltouch faults

mvturnho, 2018-07-02:
Hi,

I have the bltouch running now, but it is very unreliable.
Sometimes its oke, and the printer starts printing, but other time the touch puls is not recognized by the controller and the head runs into the build plate. then it continues to probe although the sensor flases in an error.
Since I have also the 4 Z steppers for z_tilt after the probing it starts adjusting the z steppers with fautly values since klipper did not detect the bltouch sensor fault.

does the probe sensor get monitored by the MCU where it can miss a pulse or is it interrupt controlled?

I attached my log. You can see where I have fiddled with my config in the log.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2155480/klippy.zip)

oderwat, 2018-07-02:
I would suggest you add this problem to #387. It seems to consist of two things: First there is no error handling with Klipper yet. Second your BLTouch is maybe not adjusted correctly or works in different specs than mine. You may need to use the 90° / 60° way of probing to be on the safe side.

mvturnho, 2018-07-02:
Ok seems like you address the issues. What would be the best config for now. I use de bltouch addition from the config examples

oderwat, 2018-07-02:
Does your BLTouch works "endlessly" running 120° mode? This should be the first thing to get working. Adjusting the maggot screw or replacing it with some other screw sometimes helps. Others did re-magnetise the pin.

P.S.: I posted my config in #436

mvturnho, 2018-07-02:
Yes the 120 works.
Also put a scope on de probe and lowered two pin. At every push of the pin I see a signal.
When probing on the printer the pulse signal is sometimes missed.
I have the probe on my second MCU where the Z steppers are controlled. Should I move the probe to the primary MCU?

mvturnho, 2018-07-02:
That does not work. The probe is configured as an endstop and must be on the same MCU

mvturnho, 2018-07-02:
I used the 90/60 way of probing and things improved. Also made separate 5v supply. 
It still runs into the bed sometimes that might not be a Klipper issue.
Thanks for your help

KevinOConnor, 2018-07-07:
Okay.  I'm going to close this issue in favor of the existing issue #387.

-Kevin

closed 2018-07-07 by KevinOConnor