Klipper3d/klipper #452: Z-axis changes direction mid-print?

chrzr, 2018-07-02:
Hello!

I've been using Klipper since the early 2018 and I'm very happy with it. However, in the last two months I had a few problems during printing. For example today, I wanted to print a part, which seemed fine in the Simplify3D preview. After around 30%, I went to the printer and discovered that the nozzle was scraping on the buildplate and that the part had been ripped off. I stopped the print and the z-axis still wanted to move against the plate instead of moving to z = 10mm. I've never encountered such a behavior with my previous printers running Marlin. 
I have already looked at the klippy.log, but I could not find anything suspicious. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2156377/klippy.log)

hg42, 2018-07-02:
sounds like skipping steps?
The acceleration profile is different with klipper. I am using lower acceleration but higher speeds.
Also junction_deviation behaves different to "jerk" setting of Marlin. I tend to adjust it near to zero.

StefanMeGit, 2018-07-03:
What drivers are z using on Z?

hg42, 2018-07-03:
Z axies with spindles often need a very low acceleration...I am not sure why, because they usually have a high gear ratio. I guess it has to do with static friction, which may be very high for spindles.
I found this can change over time because of dirt or oxidation and decreased lubrication (oil drifting away etc.)

hg42, 2018-07-03:
my corexy uses 25-50 mm/s^2

KevinOConnor, 2018-07-04:
On Mon, Jul 02, 2018 at 06:29:53PM +0000, chrzr wrote:
> Hello!
>
> I've been using Klipper since the early 2018 and I'm very happy with it. However, in the last two months I had a few problems during printing. For example today, I wanted to print a part, [...]

I don't see anything unusual in the log.

This doesn't seem like a software issue.  It could be hardware.
Things you could check - if the Z actually changed direction, then it
may be a bad z driver (or it is overheating), or some debris on the
board could be shorting some pins on the driver or mcu.  You might
want to carefully inspect the printer board.

-Kevin

chrzr, 2018-07-08:
Sorry for not repsonding earlier. Had some busy days this week. 

I tried a few things in the meantime:
 - Adjusting the acceleration, speed and jerk settings -> This did not help
 - Checking the Smelzi board and drivers for debris -> All ok 
 - Checking the driver voltages. Z driver voltage was a bit low (0.65V), so I adjusted it do match the voltage of others (0.95V) -> This also did not help
 - Using a different slicer again (Slic3r Prusa Edition this time with the same settings/profile as before) -> Printed without problems. I have no idea what made the difference ¯\_(ツ)_/ ¯ 

As I had already ripped the control box appart, I went ahead and installed the MKS Gen 1.4 with TMC2130s. This did cost me some hours, as I had to configure the pins for the SPI connection. Aftetwards I used the momentum and installed a E3D V6 hotend which also waited to be installed.
So, currently I'm running Marlin again and everything is fine, but I plan on returning to Klipper as soon as possible! :-)

Thanks for all the responses and tips!

closed 2018-07-08 by chrzr