Klipper3d/klipper #455: Due/RADDS: why UART1 (pins 18 and 19 on the Due) is not usable?

micheleamerica, 2018-07-03:
I'm using Klipper on a Due/RADDS configuration and very happy with everything so far, except the fact I can only use the serial ports connected to the USB connectors, and not the UARTs that breakout on other Due pins, like Serial1 on pins 18 and 19.
Is there a reason for this? Or does it require some specific configuration?
It would be very practical if all available UARTs could be used.
Thanks.

jakep82, 2018-07-03:
I've successfully used Serial0 (Pins 0 and 1) with Klipper on a Due/OULWare configuration.  I had to disable serial console to make it work with my RPi.

StefanMeGit, 2018-07-03:
@micheleamerica  Can u please post your printer.cfg ? I have a lot of trouble using Radds. Thx.

micheleamerica, 2018-07-03:
Here it goes... sorry this is not a file but .cfg files are not allowed.
Take notice that this is for a delta... home made... 

# This file serves as documentation for config parameters of delta
# style printers. One may copy and edit this file to configure a new
# delta printer. Only parameters unique to delta printers are
# described here - see the "example.cfg" file for description of
# common config parameters.

# DO NOT COPY THIS FILE WITHOUT CAREFULLY READING AND UPDATING IT
# FIRST. Incorrectly configured parameters may cause damage.

# The stepper_a section describes the stepper controlling the front
# left tower (at 210 degrees). This section also controls the homing
# parameters (homing_speed, homing_retract_dist) for all towers.
[stepper_a]
step_pin: ar24
dir_pin: ar23
enable_pin: !ar26
step_distance: 0.00625
endstop_pin: ^ar34
homing_speed: 50
homing_retract_dist: 10
position_endstop: 337.396399
#position_endstop: 337.319
#position_endstop: 336.669
#   Distance (in mm) between the nozzle and the bed when the nozzle is
#   in the center of the build area and the endstop triggers. This
#   parameter must be provided for stepper_a; for stepper_b and
#   stepper_c this parameter defaults to the value specified for
#   stepper_a.
arm_length: 281.907867
#   Length (in mm) of the diagonal rod that connects this tower to the
#   print head. This parameter must be provided for stepper_a; for
#   stepper_b and stepper_c this parameter defaults to the value
#   specified for stepper_a.
angle: 210.087642
#angle:
#   This option specifies the angle (in degrees) that the tower is
#   at. The default is 210 for stepper_a, 330 for stepper_b, and 90
#   for stepper_c.

# The stepper_b section describes the stepper controlling the front
# right tower (at 330 degrees).
[stepper_b]
step_pin: ar17
dir_pin: ar16
enable_pin: !ar22
step_distance: 0.00625
endstop_pin: ^ar36
position_endstop: 337.687068
angle: 330.145970

# The stepper_c section describes the stepper controlling the rear
# tower (at 90 degrees).
[stepper_c]
step_pin: ar2
dir_pin: ar3
enable_pin: !ar15
step_distance: 0.00625
endstop_pin: ^ar38
position_endstop: 337.091567
angle: 90.000000

[extruder]
step_pin: ar61
#analog7
dir_pin: ar60
#analog6
enable_pin: !ar62
[... 280 lines cut ...]
items: menu31, menu32, menu33

[menu menu31]
type: command
name: Printing: {1:>3s}
parameter: toolhead.is_printing:b['NO','YES']

[menu menu32]
type: command
name: Pause Print
gcode: @//pause

[menu menu33]
type: command
name: Resume Print
gcode: @//resume

#[menu menu4]
#type: input
#name: Choose: {1:s}
#parameter: 0:i('----', 'cola','pepsi','fanta','sprite')
#input_min: 0
#input_max: 4
#input_step: 1
#gcode: M117 My favorite: {1}

KevinOConnor, 2018-07-04:
On Tue, Jul 03, 2018 at 04:05:05PM +0000, micheleamerica wrote:
> I'm using Klipper on a Due/RADDS configuration and very happy with everything so far, except the fact I can only use the serial ports connected to the USB connectors, and not the UARTs that breakout [...]

The only reason is that no one has done the development work to
support other configurations.

-Kevin

KevinOConnor, 2018-07-12:
Sounds good.  Feel free to open a pull request when ready.

closed 2018-07-12 by KevinOConnor