Klipper3d/klipper #457: If a drop down GUI was made for klipper configuration it would take off like a rocket. marlin who?

Crunch69, 2018-07-04:
If a drop down GUI was made for klipper configuration it would take off like a rocket. even one better if the GUI was made as a plugin for octoprint

AxMod3DPrint, 2018-07-04:
I did look at plugin creation for Octoprint but it fried my brain as I'm no  programmer and don't know JS or Python. I am however, creating a octopi/klipper image at the moment, so klipper is available as soon as you image the Pi. You're right in saying that it would help massively for people who don't know Linux at all. 

The main things that I think would be needed are in no particular order:
* Config Upload
* Access to firmware_restart & restart (although I can add this to my Klipper_Octoi image via the Octoprint Config using the power)
* Firmware Flashing, linking into KConfig
* Automatic linking into /tmp/printer (which is something I can do via the Octoprint config on my RasPi image)

Crunch69, 2018-07-04:
AxMod3dprint u may want to build on mmone/OctoprintKlipperPlugin. What  I am speaking of a pulldown where you select your hotend or select your probe from a pull down

AxMod3DPrint, 2018-07-04:
Technically that could be built off the plugin as well.

outlookhazy, 2018-07-09:
I'm not necessarily volunteering, but I did this for myself with a .NET app that uses a remote .ssh session to edit the .cfg file, and then sending printer commands via the octoprint raw command API. This allowed me to create a single button PID autotune that pushes the values directly into the config file when it's done. My planned use case for this was for making things like delta_autocalibrate automagic and limit the potential for me to transpose a digit and send my printer into orbit.

The actual editing/generation of the config file is pretty easy, the hard part would be making a GUI that's generic enough to cover all valid printer options and configurations per-section...

KevinOConnor, 2018-07-17:
FYI, I know that @mmone has put together an OctoPrint plugin (https://github.com/mmone/OctoprintKlipperPlugin).  I'm also open to others contributing frontend improvements.  It's not something I personally plan to work on.

-Kevin

closed 2018-07-17 by KevinOConnor