Klipper3d/klipper #460: MKS Sbase temperature problem

davejowett, 2018-07-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2166290/klippy.log)
Using latest code on MKS Sbase keep getting Extrude below minimum temperature.

dieideeistgut, 2018-07-05:
Confirmative. Same here - preheating to a near temp (c.a. 50deg less) works fine. Could be due to my Start G-Code though. Will recheck later.

jakep82, 2018-07-05:
The default minimum extrusion temperature is 170.  If you want to print at or near that temperature you should drop the minimum extrusion temperature.  Here's a snippet from the example.cfg file.

> #min_extrude_temp: 170
#   The minimum temperature (in Celsius) at which extruder move
#   commands may be issued. The default is 170 Celsius.

noahwilliamsson, 2018-07-06:
@davejowett Running `awk -F extruder '/extruder/{print FS,$2}'` over your log file suggests the heater suddenly failed despite PWM being at 100%.
```
extruder : target=170 temp=171.3 pwm=0.297  # Line starting with: Stats 11482.3
extruder : target=170 temp=170.8 pwm=0.440
extruder : target=170 temp=170.4 pwm=0.440
extruder : target=170 temp=170.0 pwm=0.507
extruder : target=170 temp=169.3 pwm=0.620
extruder : target=170 temp=168.8 pwm=0.687
extruder : target=170 temp=168.2 pwm=0.744
extruder : target=170 temp=167.4 pwm=0.809
extruder : target=170 temp=166.8 pwm=0.873
extruder : target=170 temp=166.2 pwm=0.927
extruder : target=170 temp=165.3 pwm=0.994
extruder : target=170 temp=164.6 pwm=0.994
extruder : target=170 temp=163.8 pwm=0.994
extruder : target=170 temp=162.9 pwm=0.994
extruder : target=170 temp=162.1 pwm=1.000
extruder : target=170 temp=161.3 pwm=1.000
extruder : target=170 temp=160.3 pwm=1.000
extruder : target=170 temp=159.5 pwm=1.000
extruder : target=170 temp=158.5 pwm=1.000
extruder : target=170 temp=157.7 pwm=1.000
extruder : target=170 temp=156.9 pwm=1.000
extruder : target=170 temp=155.9 pwm=1.000
extruder : target=170 temp=155.1 pwm=1.000
extruder : target=170 temp=154.3 pwm=1.000
extruder : target=170 temp=153.2 pwm=1.000
extruder : target=170 temp=152.4 pwm=1.000
extruder : target=170 temp=151.6 pwm=1.000
extruder : target=170 temp=150.6 pwm=1.000
extruder  not heating at expected rate
```

noahwilliamsson, 2018-07-06:
Using a [simple moving median](https://en.wikipedia.org/wiki/Moving_average#Moving_median) over the last three readings for the `self.can_extrude` test would likely solve the issue with random bad readings popping up, assuming the temperature callback is updated frequently (once every few seconds).

```
samples.append(temp)
samples = samples[-3:]
temp_smm =  samples[0] if len(samples) == 1 else sorted(samples)[1]
```

When ramping up (or down) the temperature you would essentially make a judgement based on the previous reading, and while printing you would ignore any random outliers.

I don't know if these bad readings are expected or if they suggest that there is a problem on the LPC1768 side.  FWIW,  I haven't experienced this before when running Smoothieware.

davejowett, 2018-07-06:
repeated print at 200c with same error message. The m112 command is sent by me to stop logging. The error happens with the temperature at 202 before and 203 after.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2169447/klippy.log)

KevinOConnor, 2018-07-06:
I added some basic temperature smoothing (commit b0ee323e) that will hopefully address this.

-Kevin

davejowett, 2018-07-07:
updated this morning. Print now completes fine. Also checked PID tune and that ran fine. Thanks for your prompt help. Am happy to close this issue.

closed 2018-07-07 by davejowett

AxMod3DPrint, 2018-07-19:
Just as a note here for information. I'd suggest moving the Thermistors off the RAMPS and on to the Re-Arm. http://panucattdevices.freshdesk.com/support/solutions/articles/1000256380-using-on-board-thermistor-analog-ports This will also help keep your temps more stable due to less noise from the RAMPS. I had horrible swings with Smoothie before I did this. 

Also after commit b0ee323e2e01ba2084bea8de733f16474f1167eb I have had to run a PID tune on that printer to stabilise the temps due to the PID smoothing. I'd not used it in a week or so and the temps were all over the place.

KevinOConnor, 2018-07-19:
On Thu, Jul 19, 2018 at 09:39:12AM -0700, noahwilliamsson wrote:
> Interesting.
> 
> I can also add that it seems like ADC spikes are a thing on the LPC176x: https://os.mbed.com/users/chris/notebook/Getting-best-ADC-performance/

It looks to me like the ADC on the LPC176x is really bad.  It's noisy
on Smoothieware and Marlin as well - they implement a median filter so
it is less noticeable, but the underlying problem is still there.

On Klipper, an 8x oversample is done in the micro-controller on every
measurement (8 samples are taken 1ms apart and then the temperature is
calculated from the average of those samples).  Given the wide swings
in reported temperatures, some of the individual samples must have
been horribly inaccurate.  The fact that many people are seeing
temperature fluctuations across different boards indicates to me that
it is likely in the LPC176x itself.  We definitely don't see this
problem on the cheap AVR chips.

> A median filtering approach is better suited for rejecting random anomalies than the cumulative moving average(?) approach that b0ee323 implemented.  While CMA works great for low amplitude noise, it will merely take the edge off major spikes.
> 
> In the klippy.log attached to the first comment there is a temperature "drop" of 22 degrees (13% error).  Myself I've seen readings that are off by upwards of 20% (on a Re-ARM+RAMPS board).
> 
> In the case of a low temperature spike, I'm guessing that Klipper will set the heater to 100% until the next temperature reading, despite the new CMA smoothing.

The additional cumulative moving average over the already oversampled
temperatures (b0ee323) that Klipper implements in the host is applied
to the values reported to OctoPrint (M105), the verify_heater check,
the min_extrude_temp check, and to the M109/M190 wait for temp checks.

The PID algorithm and the log get the raw values.  This was
intentional as the PID is generally robust with noise and adding a
filter before the PID would delay its reaction time.

It does mean that a spurious low temp reading would likely cause the
PID to go full power, but that would only be for 300ms until the next
reading.  So, I think it's largely harmless.

> I made a small [script](https://gist.github.com/noahwilliamsson/cde6b9ea46ae075079f07576c10f510e) to graph the temperature readings from klippy.log on https://plot.ly.  It's evident that the default `smooth_time: 2.0` (as implemented in b0ee323) is sometimes not enough to smooth these major spikes, meaning that Klipper will sometimes interpret the heater temperature as being below the default `min_extrude_temp: 170`, and prevent extrusion.

It's always possible to increase the smooth time or to reduce
min_extrude_temp.  The min_extrude_temp check is just a "sanity check"
so tuning it isn't a big problem.

It is also possible to implement a median filter in the host.  I
wasn't sure if that was worth the effort as I don't think the PID
needs pre-filtering and the remaining users generally don't care about
a little noise.

> I didn't study the LPC176x firmware closely but if it's not already doing it, I think it would be great if it sampled the ADCs multiple times and returned a value after doing median filtering.  That would likely reduce the chances of Klippy seeing these major spikes.

As above, the micro-controller code does do oversampling.  I would
avoid adding a median filter to the micro-controller as that adds
complexity to the micro-controller code - if a median filter is
desired, the host should be able to implement it with less complexity.

-Kevin